espressif/esp-dl #120: 运行/tools/quantization_tool/examples/example.py 报错MatMul算子在esp-dl上不支持

junnan-xjtu, 2023-04-03:
报错信息：
Generating the quantization table:
MatMul is not supported on esp-dl yet
Traceback (most recent call last):
  File "example.py", line 38, in <module>
    calib.generate_quantization_table(model_proto, calib_dataset, pickle_file_path)
  File "calibrator.py", line 344, in calibrator.Calibrator.generate_quantization_table
  File "calibrator.py", line 231, in calibrator.Calibrator.generate_output_model
  File "calibrator.py", line 226, in calibrator.Calibrator.check_model
ValueError: current model is not supported by esp-dl

junnan-xjtu, 2023-04-03:
请问这是什么原因造成的，代码没有改动过

Raven888888, 2023-04-06:
Did you face this problem during setup? #121

Auroragan, 2023-04-10:
Hi, example.py 中会调用 optimized_model_path = optimize_fp_model(model_path) 这个函数来将 matmul 转为 gemm，函数定义在 https://github.com/espressif/esp-dl/blob/master/tools/quantization_tool/optimizer.py，检查一下路径之类的有没有问题吧

zhouayi, 2023-11-22:
我也出现了同样的问题，然后我原本的环境是python=3.7，onnx=1.13.1。然后我将onnx降低版本为1.9.0后，问题解决了
[image]

closed 2024-10-21 by sun-xiangyu