espressif/esp-dl #121: Unable to import calibrator

Raven888888, 2023-04-05:
Here is my env
Python 3.8
Windows
```
absl-py                       0.12.0
aiohttp                       3.6.3
anyio                         3.1.0
argon2-cffi                   20.1.0
astunparse                    1.6.3
async-generator               1.10
async-timeout                 3.0.1
attrs                         21.2.0
Babel                         2.9.1
backcall                      0.2.0
backports.functools-lru-cache 1.6.4
bleach                        3.3.0
blinker                       1.4
brotlipy                      0.7.0
cachetools                    4.2.2
certifi                       2020.6.20
cffi                          1.14.5
chardet                       3.0.4
click                         7.1.2
colorama                      0.4.4
coloredlogs                   15.0.1
cryptography                  3.4.7
decorator                     5.0.9
defusedxml                    0.7.1
entrypoints                   0.3
filelock                      3.10.7
flatbuffers                   1.12
gast                          0.4.0
google-auth                   1.30.2
google-auth-oauthlib          0.4.4
google-pasta                  0.2.0
grpcio                        1.34.1
h5py                          3.1.0
huggingface-hub               0.13.3
humanfriendly                 10.0
idna                          2.10
importlib-metadata            4.5.0
ipykernel                     5.5.5
ipython                       7.24.1
ipython-genutils              0.2.0
jedi                          0.18.0
Jinja2                        3.0.1
joblib                        1.0.1
json5                         0.9.5
jsonschema                    3.2.0
jupyter-client                6.1.12
jupyter-core                  4.7.1
jupyter-server                1.8.0
jupyterlab                    3.0.16
jupyterlab-pygments           0.1.2
jupyterlab-server             2.6.0
Keras-Applications            1.0.8
keras-nightly                 2.5.0.dev2021032900
Keras-Preprocessing           1.1.2
llvmlite                      0.39.1
Markdown                      3.3.4
MarkupSafe                    2.0.1
matplotlib-inline             0.1.2
mistune                       0.8.4
mkl-fft                       1.3.0
mkl-random                    1.2.1
mkl-service                   2.3.0
mpmath                        1.3.0
multidict                     4.7.6
nbclassic                     0.3.1
nbclient                      0.5.3
nbconvert                     6.0.7
nbformat                      5.1.3
nest-asyncio                  1.5.1
networkx                      3.1
notebook                      6.4.0
numba                         0.56.4
numpy                         1.19.2
oauthlib                      3.1.1
onnx                          1.13.1
onnxoptimizer                 0.3.10
onnxruntime                   1.7.0
opt-einsum                    3.3.0
packaging                     20.9
pandocfilters                 1.4.2
parso                         0.8.2
pickleshare                   0.7.5
Pillow                        8.2.0
pip                           20.2.4
prometheus-client             0.11.0
prompt-toolkit                3.0.18
protobuf                      3.20.3
pyasn1                        0.4.8
pyasn1-modules                0.2.8
pycparser                     2.20
Pygments                      2.9.0
PyJWT                         1.7.1
pyOpenSSL                     20.0.1
pyparsing                     2.4.7
pyreadline                    2.1
pyreadline3                   3.4.1
pyrsistent                    0.17.3
PySocks                       1.7.1
python-dateutil               2.8.1
pytz                          2021.1
pywin32                       300
pywinpty                      0.5.7
PyYAML                        6.0
pyzmq                         22.1.0
regex                         2023.3.23
requests                      2.25.1
requests-oauthlib             1.3.0
rsa                           4.7.2
scikit-learn                  0.24.2
scipy                         1.6.3
Send2Trash                    1.5.0
setuptools                    50.3.0.post20201006
six                           1.15.0
sniffio                       1.2.0
sympy                         1.11.1
tensorboard                   2.5.0
tensorboard-data-server       0.6.1
tensorboard-plugin-wit        1.8.0
tensorflow                    2.5.3
tensorflow-estimator          2.5.0
termcolor                     1.1.0
terminado                     0.9.1
testpath                      0.5.0
tf2onnx                       1.14.0
threadpoolctl                 2.1.0
tokenizers                    0.13.3
torch                         2.0.0
tornado                       6.1
tqdm                          4.65.0
traitlets                     5.0.5
transformers                  4.27.4
typing-extensions             3.7.4.3
urllib3                       1.26.5
wcwidth                       0.2.5
webencodings                  0.5.1
websocket-client              0.57.0
Werkzeug                      2.0.1
wheel                         0.35.1
win-inet-pton                 1.1.0
wincertstore                  0.2
wrapt                         1.12.1
yarl                          1.6.2
zipp                          3.4.1
```

When run example to use quantization tool
```
$ python example.py
Traceback (most recent call last):
  File "example.py", line 16, in <module>
    from calibrator import *
ImportError: DLL load failed while importing calibrator: The specified module could not be found.
```

Not sure which library is causing the problem. The code is closed source, and `import *` is not helping.

Raven888888, 2023-04-06:
@NatashaLi-ESPRESSIF do u know?

Auroragan, 2023-04-10:
Hi, the compiled library for windows is here: https://github.com/espressif/esp-dl/tree/master/tools/quantization_tool/windows. They are compiled using python 3.7, please try 3.7 first.

DSSanjaya5, 2024-05-16:
> Hi, the compiled library for windows is here: https://github.com/espressif/esp-dl/tree/master/tools/quantization_tool/windows. They are compiled using python 3.7, please try 3.7 first.

As per the documentation I tried with Python 3.7 (Windows 11) but the mentioned packages give error - Could not find a version that satisfies the requirement onnxruntime==1.7.0 . What's the problem here, is the documentation wrong or is there any other mistake.

DSSanjaya5, 2024-05-19:
Initially I tried to install the packages as per the given env and requirements, ie. Py-3.7 and the other things. But I got errors when I installed those packages with the specified versions. So I went with my Py3.7 also with Py-3.11.0 and installed the latest versions of the required packages. Now there is no problem with the packages but I'm getting this error :
from calibrator import *
ImportError: DLL load failed while importing calibrator: The specified module could not be found.

MyFolder
- calibrator_acc.pyd
- calibrator.pyd
- evaluator.pyd
- main.py
- opotimizer.py
- requirements.txt

main.py
from optimizer import *
from calibrator import *
from evaluator import *

So please look into this issue and share your views. Thank you

BlueSkyB, 2024-05-21:
Hi, The relevant environment requirements are as follows:

Python == 3.7
Numba == 0.53.1
ONNX == 1.9.0
ONNX Runtime == 1.7.0
ONNX Optimizer == 0.2.6

You can find these requirements in esp-dl/tools/quantization_tool/requirements.txt. It is not necessary to use the latest versions, as they may cause issues.

The issue you encountered, "Could not find a version that satisfies the requirement onnxruntime==1.7.0," was a message from your computer environment when installing the packages, not a message from esp-dl. It could be due to other dependencies restricting the installation of onnxruntime 1.7.0, or your download source might not have that version. You can use a search engine to learn more about managing Python environments or Anaconda environments.

HavingaThijs, 2024-07-05:
It seems the compiled libraries need to be added to your `PYTHON_PATH`. On Linux, it worked for me when I added `export PYTHONPATH="${PYTHONPATH}:<your-path-to-esp-dl>/tools/quantization_tool/linux/"`. 

Related: #93, #101.

sun-xiangyu, 2024-10-21:
please try new esp-dl

closed 2024-10-21 by sun-xiangyu