espressif/esp-dl #126: idf5.0版本不保存人脸特征数据

mootterry, 2023-06-24:
idf5.0版本不保存人脸特征数据，调用enroll的接口后数据没有保存，重启后没有已注册的人脸。
怀疑是写入前没有进行擦除，于是在调用encroll前先执行fr分区的擦除，于是问题得到解决。
请处理一下这个问题，谢谢

xccsmart, 2023-08-14:
我也遇到这个问题了，但是即便擦除还是不行
    if (recognizer.get_enrolled_id_num() == 0)
    {
        const esp_partition_t *partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_ANY, "fr");
        cout << "esp_partition_erase_range  " << esp_partition_erase_range(partition, 0, partition->size) << "\n";
        id316 = recognizer.enroll_id(image_rgb888_316_1, landmarks_316_1, name_316, true);
        cout << "name: " << name_316 << ", id: " << id316 << "\n";

        id319 = recognizer.enroll_id(image_rgb888_319_1, landmarks_319_1, name_319, true);
        cout << "name: " << name_319 << ", id: " << id319 << "\n";
    }

xccsmart, 2023-08-14:
遇到相同问题，但是擦除fr分区也没有解决问题
    if (recognizer.get_enrolled_id_num() == 0)
    {
        const esp_partition_t *partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_ANY, "fr");
        cout << partition->label << ",size: " << partition->size << "\n";
        cout << "esp_partition_erase_range  " << esp_partition_erase_range(partition, 0, partition->size) << "\n";
        id316 = recognizer.enroll_id(image_rgb888_316_1, landmarks_316_1, name_316, true);
        cout << "name: " << name_316 << ", id: " << id316 << "\n";
        id319 = recognizer.enroll_id(image_rgb888_319_1, landmarks_319_1, name_319, true);
        cout << "name: " << name_319 << ", id: " << id319 << "\n";
    }

Mr-Techtron, 2023-09-02:
@mootterry 
你能使用idf5.0检测人脸吗？从我这边看，我没有看到任何面部检测。我使用的是 esp-idf 的“release/v5.0”和 esp-who 的 idfv5.0。

Translated : Are you able to detect human face using idf5.0? From my side, I don't see any face detection. I am using the "release/v5.0" of esp-idf and idfv5.0 of esp-who.

xccsmart, 2023-09-02:
可以检测到，只是特征值保存不下来

Mr-Techtron, 2023-09-02:
奇怪的！在我这边，我无法检测到面部标志。您是否对示例代码进行了一些更改以使其正常工作？
Translated : Strange! On my side, I am unable to detect the facial landmark. Did you make some changes in the example code to make it work?

vahidrezaee, 2023-10-19:
I  face the same problem, 
the enrolled face ids with update_flash = true,  doesn't save in flash,
after restarting  the  recognizer-> set_ids_from_flash() returns  0/
Did anybody fix this problem?

mootterry, 2023-10-19:
> I face the same problem, the enrolled face ids with update_flash = true, doesn't save in flash, after restarting the recognizer-> set_ids_from_flash() returns 0/ Did anybody fix this problem?

same problem!!!

vahidrezaee, 2023-10-19:
dear @mootterry 
how do we contact the authors of esp-dl library, to get some technical support?
I have an ESP32-S3-eye evaluation board.
when I flash the default framework .bin file into the board, the enrolled face IDs will be saved into flash.
but when I build the es32-s3-eye source code and flash the output into the board.
the enrolled face IDs will be erased from flash memory after restarting the board.

Sandra-lol, 2023-10-25:
Hi guys, the master and idf5.0 branches have been updated, pls try again to see if the issue still exists. Thanks!

YIYUEG, 2025-06-12:
我也碰到了这个问题。就是说只要想保存到flash，必须要将分区擦除才行。recognizer->enroll_id((uint16_t *)frame->buf, {(int)frame->height, (int)frame->width, 3}, detect_results.front().keypoint, username, true);这个函数第一次能保存但是第二次会报错。我是用recognizer->enroll_id((uint16_t *)frame->buf, {(int)frame->height, (int)frame->width, 3}, detect_results.front().keypoint, username);
                            {
                                const esp_partition_t *partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_ANY, "storage");
                                esp_partition_erase_range(partition, 0, partition->size);
                            }
                            recognizer->write_ids_to_flash();这个方法保存数据的，就是每次要把分区"storage"（当然得改成你的分区名）擦除，才能写进去，希望对小伙伴们有帮助。