espressif/esp-dl #131: deploy my model using TVM got erro: NameError: name 'GenerateESPConstants' is not defined

gjhave, 2023-08-20:
Hi, everyboy:
     i try to deploy my model.onnx to esp32s3, using the official common TVM with below:
'python esp_quantize_onnx.py --input_model model_opt.onnx --output_model model_quant.onnx --calibrate_dataset calib_img.npy'
but got an erro:"**NameError: name 'GenerateESPConstants' is not defined**"
all the operation before this command are normal!

the erro message shows it come from "tvm/python/tvm/relay/op/contrib/esp.py", it seems a esp custorm code in tvm.
does anyone got the same issue with mine? and how to fix it?

gjhave, 2023-08-21:
i tried again to quantization my model, with esp-dl-release-1.1, there is no tvm tool in folder.
so i infer that my GenerateESPConstants problem was caused by unreleased master branch.

yes, the example.py in esp-dl-release-1.1/tools/quantization_tool/examples is normally worked, but i got an other problem,
that is, cureent esp-dl-release is only support by python 3.7, also the calibrator.pyd, evaluator.py are supported only by python3.7, that result in the env can only support torch 1.13
but my model was trainning by torch 2.0,  such a bad news to me.
hope the esp-dl working with torch2.0 coming ASAP

TFST, 2023-08-22:
我也遇到了同样的问题。我开始用的自己的python解释器，报错NameError: name 'GenerateESPConstants' is not defined，发现是有两个函数没定义，就猜测是要用idf中自带的python解释器，且需要版本是3.7或3.8。但是下载时默认5.0时python3.11，我不知道怎么改idf-tools自带的python版本。

lijunru-hub, 2023-08-23:
Training environment with python3.11 + torch 2.0, then model transformed to onnx. The quantization environment is python3.7. there is no conflict!

closed 2023-08-27 by gjhave