espressif/esp-dl #134: 请问Model Zoo中的human_face_detect, cat_face_detect在识别过程中最大的内存占用是多少？内部与外部内存是如何分配的

MichaelDu9226, 2023-08-31:
将Model Zoo中的人脸识别等算法内置到了Micropython中，可以通过micropython指令方便的调用这些模型，识别速度非常满意。

在开启摄像头及图传后，我们内部内存剩余在50-80KB左右，一旦开启人脸检测后，很容易报wifi内存错误，我们分析可能是内存不足引起的。因为这些模型是不开源的，所以想了解一下这些模型的内存占用情况，尤其是峰值情况。

我在使用esp who\examples\human_face_detection\lcd例程时，开启了另外一个task，间隔10ms打印内存信息，偶尔可以看到内存占用了70K左右内部内存的情况。
如一下截图
[image]

使用的内存函数：printf("m: %d, inter: %d, dram: %d\r\n", esp_get_free_heap_size(), heap_caps_get_free_size(MALLOC_CAP_INTERNAL), heap_caps_get_free_size(MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT));

mootterry, 2023-09-02:
我也是遇到同样的问题，不能与wifi和ble稳定的共存。

sun-xiangyu, 2024-10-21:
please try new esp-dl

closed 2024-10-21 by sun-xiangyu