espressif/esp-dl #139: autotvm:Cannot find tuning records for: target=c -keys=cpu -model=esp32

ShawnHymel, 2023-09-29:
I am following the guide here: [https://docs.espressif.com/projects/esp-dl/en/latest/esp32s3/tutorials/deploying-models-through-tvm.html](https://docs.espressif.com/projects/esp-dl/en/latest/esp32s3/tutorials/deploying-models-through-tvm.html).I trained a basic 2-layer DNN on MNIST using torch and exported the model in ONNX format. I saved a representative dataset *train_set.npy* and a single sample as *sample.npy*. I optimized and quantized the model with the following:

```
python3.7 -m onnxruntime.quantization.preprocess --input model.onnx --output model_opt.onnx
python3.7 ./esp-dl/tools/tvm/esp_quantize_onnx.py --input_model model_opt.onnx --output_model model_quant.onnx --calibrate_dataset train_set.npy
```

I then try to convert the quantized model to ESP-DL format with the following:

```
python3.7 esp-dl/tools/tvm/export_onnx_model.py --target_chip esp32s3 --model_path model_quant.onnx --img_path sample.npy --template_path ./esp-dl/tools/tvm/template_project_for_model --out_path esp32-inference-project
```

When I enable logging in the *export_onnx_model.py* script, I get the following:

```
Model Information:
------------------
Input Name: onnx::Gemm_0
Input Shape: (1, 784)
Input DType: float
Output Name: 7
Output Shape: (1, 10)
Output DType: float
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for multiply based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for multiply based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for expand_dims based on highest priority (10)
INFO:te_compiler:Using injective.cpu for expand_dims based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for subtract based on highest priority (10)
INFO:te_compiler:Using injective.cpu for full based on highest priority (10)
INFO:te_compiler:Using injective.cpu for full based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for subtract based on highest priority (10)
INFO:te_compiler:Using injective.cpu for full based on highest priority (10)
INFO:te_compiler:Using injective.cpu for full based on highest priority (10)
WARNING:autotvm:One or more operators have not been tuned. Please tune your model for better performance. Use DEBUG logging level to see more details.
DEBUG:autotvm:Cannot find tuning records for:
    target=c -keys=cpu -model=esp32
    key=('dense_pack.x86', ('TENSOR', (1, 784), 'int16'), ('TENSOR', (256, 784), 'int16'), None, 'int32')
TVM will apply a default schedule which may negatively impact performance.
INFO:te_compiler:Using dense_pack.x86 for nn.dense based on highest priority (10)
DEBUG:autotvm:Cannot find tuning records for:
    target=c -keys=cpu -model=esp32
    key=('dense_pack.x86', ('TENSOR', (1, 256), 'int16'), ('TENSOR', (10, 256), 'int16'), None, 'int32')
TVM will apply a default schedule which may negatively impact performance.
INFO:te_compiler:Using dense_pack.x86 for nn.dense based on highest priority (10)
INFO:te_compiler:Using layout_transform.generic for layout_transform based on highest priority (10)
INFO:te_compiler:Using layout_transform.generic for layout_transform based on highest priority (10)
INFO:te_compiler:Using injective.cpu for divide based on highest priority (10)
INFO:te_compiler:Using injective.cpu for round based on highest priority (10)
INFO:te_compiler:Using injective.cpu for clip based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
DEBUG:autotvm:Cannot find tuning records for:
    target=c -keys=cpu -model=esp32
    key=('dense_pack.x86', ('TENSOR', (1, 784), 'int16'), ('TENSOR', (32, 784, 8), 'int16'), None, 'int32')
TVM will apply a default schedule which may negatively impact performance.
INFO:te_compiler:Using dense_pack.x86 for nn.contrib_dense_pack based on highest priority (10)
INFO:te_compiler:Using injective.cpu for add based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for multiply based on highest priority (10)
INFO:te_compiler:Using injective.cpu for zeros based on highest priority (10)
INFO:te_compiler:Using injective.cpu for greater_equal based on highest priority (10)
INFO:te_compiler:Using injective.cpu for where based on highest priority (10)
INFO:te_compiler:Using injective.cpu for add based on highest priority (10)
INFO:te_compiler:Using injective.cpu for right_shift based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for clip based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for nn.relu based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
DEBUG:autotvm:Cannot find tuning records for:
    target=c -keys=cpu -model=esp32
    key=('dense_pack.x86', ('TENSOR', (1, 256), 'int16'), ('TENSOR', (2, 256, 5), 'int16'), None, 'int32')
TVM will apply a default schedule which may negatively impact performance.
INFO:te_compiler:Using dense_pack.x86 for nn.contrib_dense_pack based on highest priority (10)
INFO:te_compiler:Using injective.cpu for add based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for multiply based on highest priority (10)
INFO:te_compiler:Using injective.cpu for zeros based on highest priority (10)
INFO:te_compiler:Using injective.cpu for greater_equal based on highest priority (10)
INFO:te_compiler:Using injective.cpu for where based on highest priority (10)
INFO:te_compiler:Using injective.cpu for add based on highest priority (10)
INFO:te_compiler:Using injective.cpu for right_shift based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for clip based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for multiply based on highest priority (10)
esp_dl_library_path: /content/esp-dl
generated project in: ./esp32-inference-project/new_project
```

It looks like the model is converted. However, *autotvm* complains about not finding tuning records:

```
WARNING:autotvm:One or more operators have not been tuned. Please tune your model for better performance. Use DEBUG logging level to see more details.
DEBUG:autotvm:Cannot find tuning records for:
    target=c -keys=cpu -model=esp32
    key=('dense_pack.x86', ('TENSOR', (1, 784), 'int16'), ('TENSOR', (256, 784), 'int16'), None, 'int32')
TVM will apply a default schedule which may negatively impact performance.
```

This does not appear to be the intended behavior of the script. Any suggestions on how to fix this issue?

ShawnHymel, 2023-09-29:
If it helps, here is my Colab script that trains a model in torch, converts it to ONNX, and attempts to convert it to ESP-DL with a template project: https://colab.research.google.com/gist/ShawnHymel/82d1a11278da45831f0d943d44ea2cc1/pytorch-mnist-onnx-quantization.ipynb

ShawnHymel, 2023-09-29:
Not sure if it's related, but if I change the `__main__` part in *export_onnx_model.py* to the following:

```
if __name__ == '__main__':
    import argparse
    
    import logging
    logging.basicConfig(level=logging.DEBUG)

    parser = argparse.ArgumentParser(description='Test Assistant')
    parser.add_argument('--target_chip', help='esp32s3')
    parser.add_argument('--model_path', help='path of onnx model')
    parser.add_argument('--img_path', help='path of npy file which stores an input image of this model')
    parser.add_argument('--template_path', help='path of template project')
    parser.add_argument('--out_path', help='path of generated project')
    args = parser.parse_args()
    
    dump_tvm_onnx_project(target_chip=args.target_chip, model_path=args.model_path, img_path=args.img_path,  template_path=args.template_path, generated_project_path=args.out_path)
    debug_onnx_model(args.target_chip, args.model_path, args.img_path)
```

I get the following output:

```
Model Information:
------------------
Input Name: onnx::Gemm_0
Input Shape: (1, 784)
Input DType: float
Output Name: 7
Output Shape: (1, 10)
Output DType: float
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for multiply based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for multiply based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for expand_dims based on highest priority (10)
INFO:te_compiler:Using injective.cpu for expand_dims based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for subtract based on highest priority (10)
INFO:te_compiler:Using injective.cpu for full based on highest priority (10)
INFO:te_compiler:Using injective.cpu for full based on highest priority (10)
INFO:te_compiler:Using injective.cpu for cast based on highest priority (10)
INFO:te_compiler:Using injective.cpu for subtract based on highest priority (10)
INFO:te_compiler:Using injective.cpu for full based on highest priority (10)
INFO:te_compiler:Using injective.cpu for full based on highest priority (10)
WARNING:autotvm:One or more operators have not been tuned. Please tune your model for better performance. Use DEBUG logging level to see more details.
DEBUG:autotvm:Cannot find tuning records for:
    target=c -keys=cpu -model=esp32
    key=('dense_pack.x86', ('TENSOR', (1, 784), 'int16'), ('TENSOR', (256, 784), 'int16'), None, 'int32')
TVM will apply a default schedule which may negatively impact performance.
INFO:te_compiler:Using dense_pack.x86 for nn.dense based on highest priority (10)
DEBUG:autotvm:Cannot find tuning records for:
    target=c -keys=cpu -model=esp32
    key=('dense_pack.x86', ('TENSOR', (1, 256), 'int16'), ('TENSOR', (10, 256), 'int16'), None, 'int32')
TVM will apply a default schedule which may negatively impact performance.
INFO:te_compiler:Using dense_pack.x86 for nn.dense based on highest priority (10)
INFO:te_compiler:Using layout_transform.generic for layout_transform based on highest priority (10)
[... 305 lines cut ...]
  15: _PyObject_FastCallKeywords
  14: 0x00000000005abdaa
  13: _PyObject_Call_Prepend
  12: _PyFunction_FastCallDict
  11: _PyEval_EvalCodeWithName
  10: _PyEval_EvalFrameDefault
  9: 0x00000000004de44c
  8: _PyObject_FastCallKeywords
  7: 0x00007978893430d6
  6: 0x0000797889339eae
  5: 0x0000797888977492
  4: 0x000079788897ae2d
  3: operator()
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/runtime/graph_executor/debug/graph_executor_debug.cc:203
  2: tvm::runtime::GraphExecutorDebug::DebugGetNodeOutput(int, DLTensor*)
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/runtime/graph_executor/debug/graph_executor_debug.cc:325
  1: tvm::runtime::NDArray::CopyTo(DLTensor*) const
        at /home/gansichen/Workspace/projects/local/framework/tvm/include/tvm/runtime/ndarray.h:392
  0: tvm::runtime::NDArray::CopyFromTo(DLTensor const*, DLTensor*, void*)
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/runtime/ndarray.cc:293
  File "/home/gansichen/Workspace/projects/local/framework/tvm/src/runtime/ndarray.cc", line 293
InternalError: Check failed: from_size == to_size (256 vs. 16384) : TVMArrayCopyFromTo: The size must exactly match
```

It looks like it's failing the debug step, and I've been unable to track down exactly where.

noorhaq, 2023-10-06:
Not sure why it was failing for you.
But here's my notebook working https://colab.research.google.com/gist/noorhaq/926bbbacecf0473f598cdd386668f763/pytorch-mnist-onnx-quantization.ipynb#scrollTo=q5NfoEMMqx0R
If you require any assistance in using the generated project let me know and I will share it with you

ShawnHymel, 2023-10-06:
Hi @noorhaq,

Thanks for checking it out! The notebook does indeed run, but the model fails the debugging check step. You have to uncomment the following line in *esp-dl/tools/tvm/export-onnx-model.py* to see the failure:

```
debug_onnx_model(args.target_chip, args.model_path, args.img_path)
```

Auroragan, 2023-10-09:
Hi,
'generated project in: ./esp32-inference-project/new_project' shows that your conversion is succeed.
autotvm is not used in this script and is not supported for esp boards yet. I think 'Cannot find tuning records' is just an internal debug info suggesting default strategy will be used not the ones that autotvm found.

ShawnHymel, 2023-10-11:
Hi @Auroragan,

Good to know, thank you! I'll close out the issue.

MichelBerg, 2024-01-26:
Currently, I am working with ESP-DL TVM and I got the same issue while uncommenting the debug function.

The “error” message:
`InternalError: Check failed: from_size == to_size (256 vs. 16384) : TVMArrayCopyFromTo: The size must exactly match`

is created because of the debug function.

It looks like the example is based on the following model from ESP-DL tutorial https://github.com/espressif/esp-dl/blob/master/tutorial/tvm_example/model.onnx
I was curious about it, so I looked up the model architecture by using Netron:
[image]

Due to that @ShawnHymel, you got that error with the debug output. If you use the example model, you won't get that error. You could also change the output of any layer you want to read, that is specified by your own model. Of course you have to adjust the `tvm_out` shape as well as the name of the layer.

I suggest removing this code or put it in another example. It only leads to confusion in my opinion.
https://github.com/espressif/esp-dl/blob/19d377eb47230bc101fc159086b0cabbdf6f17b0/tools/tvm/export_onnx_model.py#L219C4-L222C43