espressif/esp-dl #152: ESP-DL Quant Tool | TVM conversion & deployment on ESP32S3-EYE

MichelBerg, 2024-01-26:
Recently I am working with ESP-DL Quantization Tool and TVM. I am trying to build different models and try to deploy them to the ESP32S3-EYE.

Creating several simple models works in most cases with the quantization tool, but rarely with TVM.

While comparing simple model like this:
[image]

My tests are using a static dummy image with the correct shape.
Using the Quantization Tool: ~113122 us
Using TVM: ~680491 us

Overall the performance of TVM, seams to be very inefficient for me right now. In most of these szenarios with "large" Models, TVM needs a large amount of memory. To fix the overflow in the **.dram0.bss** segment, I already moved the data to the **PSRAM**. (using **EXT_RAM_BSS_ATTR** and the necessary menuconfig ind the esp-idf)

What I want to do is Transfer learning with MobileNetV2. Building and train it with TF and converting it to ONNX isn't a problem. But somehow the conversions given with the ESP-DL Tools (Quant and TVM) is one. I understand that using the quantization tool won't be able to convert MobileNetV2 due to the missing layers, that are not implemented yet ([ESP-DL Layers](https://github.com/espressif/esp-dl/tree/master/include/layer))
But using the TVM method leads to errors by using `esp-dl/tools/tvm/export_onnx_model.py`.

My Transfer learning dummy model looks like this:

```
base_model = tf.keras.applications.MobileNetV2(input_shape=IMAGE_SHAPE,
                                               alpha=0.35,
                                               include_top=False,
                                               weights='imagenet')
model = tf.keras.Sequential([
    base_model,
    layers.GlobalAveragePooling2D(),
    layers.Dense(len(class_names), activation='softmax')
])
```
Exporting it to ONNX looks like this:

`python -m tf2onnx.convert --saved-model tmp_model --output {model_name} --opset {opset}`

For building a model i use a seperate google colab notebook. 
The versions I use:
- Python: 3.10
- ONNX: 1.15.0
- TF2ONNX: 1.16.1 (primary using opset=13)

As a test I tried the TVM export by using different opsets (10-15). But the error looks for every version the same.

My colab notebook for the TVM export is based on https://github.com/espressif/esp-dl/issues/139.

The resulting error looks like this:
```
Using '/content/onnx/onnx_from_tensorflow_models/96x96x3_mobilenetV2_0_35_opset_14_model.onnx' for ESP-DL TVM conversion!
WARNING:root:Please use QuantFormat.QDQ for activation type QInt8 and weight type QInt8. Or it will lead to bad performance on x64.
Collecting tensor data and making histogram ...
Finding optimal threshold for each tensor using entropy algorithm ...
Number of tensors : 107
Number of histogram bins : 128 (The number may increase depends on the data it collects)
Number of quantized bins : 128

[13:24:51] /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/transforms/convert_layout.cc:99: Warning: Desired layout(s) not specified for op: nn.global_avg_pool2d
Model Information:
------------------
Input Name: input_4
Input Shape: (1, 96, 96, 3)
Input DType: float
Output Name: dense_4
Output Shape: (1, 10)
Output DType: float
Traceback (most recent call last):
  File "esp-dl/tools/tvm/export_onnx_model.py", line 237, in <module>
    dump_tvm_onnx_project(target_chip=args.target_chip, model_path=args.model_path, img_path=args.img_path,  template_path=args.template_path, generated_project_path=args.out_path)
  File "esp-dl/tools/tvm/export_onnx_model.py", line 167, in dump_tvm_onnx_project
    mod = preprocess_onnx_for_esp(mod, params)
  File "/content/esp-dl/tools/tvm/python/tvm/relay/op/contrib/esp.py", line 61, in preprocess_onnx_for_esp
    return seq(mod)
  File "/content/esp-dl/tools/tvm/python/tvm/ir/transform.py", line 160, in __call__
[... 280 lines cut ...]
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/transforms/fold_constant.cc:270
  8: tvm::relay::Eval(tvm::RelayExpr, tvm::runtime::Map<tvm::GlobalTypeVar, tvm::TypeData, void, void>, std::unordered_set<tvm::runtime::String, std::hash<tvm::runtime::String>, std::equal_to<tvm::runtime::String>, std::allocator<tvm::runtime::String> >, DLDevice, tvm::Target, tvm::runtime::Map<tvm::runtime::String, tvm::runtime::ObjectRef, void, void>)
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/backend/interpreter.cc:1108
  7: tvm::relay::Prepare(tvm::IRModule, tvm::CompilationConfig const&)
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/backend/interpreter.cc:961
  6: tvm::runtime::TVMRetValue tvm::runtime::PackedFunc::operator()<tvm::IRModule, tvm::transform::PassContext>(tvm::IRModule&&, tvm::transform::PassContext&&) const
  5: operator()
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/transforms/type_infer.cc:981
  4: tvm::relay::TypeInferencer::Infer(tvm::GlobalVar, tvm::relay::Function)
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/transforms/type_infer.cc:787
  3: tvm::relay::TypeInferencer::Solve()
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/transforms/type_infer.cc:623
  2: tvm::relay::TypeSolver::Solve()
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/analysis/type_solver.cc:632
  1: tvm::TypedEnvFunc<bool (tvm::runtime::Array<tvm::Type, void> const&, int, tvm::Attrs const&, tvm::TypeReporter const&)>::operator()(tvm::runtime::Array<tvm::Type, void> const&, int, tvm::Attrs const&, tvm::TypeReporter const&) const
        at /home/gansichen/Workspace/projects/local/framework/tvm/include/tvm/ir/env_func.h:142
  0: tvm::relay::SplitRel(tvm::runtime::Array<tvm::Type, void> const&, int, tvm::Attrs const&, tvm::TypeReporter const&)
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/op/tensor/transform.cc:3070
  File "/home/gansichen/Workspace/projects/local/framework/tvm/src/relay/analysis/type_solver.cc", line 643
InternalError: Check failed: (false) is false: [13:28:38] /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/op/tensor/transform.cc:3070: InternalError: Check failed: (reporter->Assert(indices[i] > begin)) is false: indices_or_sections need to be a sorted ascending list
```
My goal is to compare these tools from **Espressif** directly with **TF Lite micro** or with models that are build with [**EdgeImpulse**](https://edgeimpulse.com/).
I wonder if somebody tried the same and faced similar issues.

I would be very happy about any kind of advice and support.

MichelBerg, 2024-01-30:
I got an other version of MobilenetV2 actually converting using the TVM tool. But after flashing my ESP32S3-EYE i get the following error: 

`
ELF file SHA256: f5fcc5d1cf125178

Rebooting...

ESP-ROM:esp32s3-20210327

Build:Mar 27 2021

rst:0x3 (RTC_SW_SYS_RST),boot:0x2a (SPI_FAST_FLASH_BOOT)

Saved PC:0x40375945

0x40375945: esp_restart_noos_dig at /home/bergm/esp_v5.0/esp-idf/components/esp_system/esp_system.c:64 (discriminator 1)

SPIWP:0xee

mode:DIO, clock div:1

load:0x3fce3810,len:0x1860

load:0x403c9700,len:0xd7c

load:0x403cc700,len:0x2e5c

entry 0x403c9934

I (25) boot: ESP-IDF v5.0.4-384-ge42bcf2e88 2nd stage bootloader

I (26) boot: compile time 19:13:40

I (27) boot: Multicore bootloader

I (29) boot: chip revision: v0.1

I (33) qio_mode: Enabling default flash chip QIO

I (38) boot.esp32s3: Boot SPI Speed : 80MHz

I (43) boot.esp32s3: SPI Mode       : QIO

I (48) boot.esp32s3: SPI Flash Size : 8MB

I (52) boot: Enabling RNG early entropy source...

I (58) boot: Partition Table:

I (61) boot: ## Label            Usage          Type ST Offset   Length

I (69) boot:  0 factory          factory app      00 00 00010000 003c0000

I (76) boot:  1 nvs              WiFi data        01 02 003d0000 00004000

I (83) boot:  2 fr               Unknown data     01 06 003e0000 00020000

I (91) boot: End of partition table

I (95) esp_image: segment 0: paddr=00010020 vaddr=3c070020 size=d9864h (890980) map

I (239) esp_image: segment 1: paddr=000e988c vaddr=3fc92e00 size=02bc4h ( 11204) load

I (241) esp_image: segment 2: paddr=000ec458 vaddr=40374000 size=03bc0h ( 15296) load

I (247) esp_image: segment 3: paddr=000f0020 vaddr=42000020 size=61b54h (400212) map

I (313) esp_image: segment 4: paddr=00151b7c vaddr=40377bc0 size=0b1f8h ( 45560) load

I (329) boot: Loaded app from partition at offset 0x10000
[... 112 lines cut ...]

0x4201fd39: tie728_s8_conv2d_per_channel_hwcn_bias_relu_loop at /home/gansichen/Workspace/delete/aigit/master/dl/components/dl/base/isa/tie728/dl_tie728_s8_conv2d.S:2028

0x4200bd3f: void dl::base::conv_operation_shell<signed char, long>(dl::base::ArgsType<signed char>&, void (*)(signed char*, signed char*, void*), void (*)(signed char*, signed char*, void*), void (*)(long*, signed char*, dl::base::ArgsType<signed char> const&), void (*)(long*, signed char*, dl::base::ArgsType<signed char> const&), void (*)(signed char*, long*, dl::base::ArgsType<signed char> const&)) at /home/gansichen/Workspace/delete/aigit/master/dl/components/dl/base/dl_base.hpp:471

0x4200c84b: void dl::base::conv2d<signed char, signed char, signed char, signed char, long>(void*) at /home/gansichen/Workspace/delete/aigit/master/dl/components/dl/base/dl_base_conv2d.cpp:1220

0x4200bb7d: dl::nn::conv2d(dl::Tensor<signed char>&, dl::Tensor<signed char>&, std::vector<int, std::allocator<int> >&, dl::Filter<signed char> const&, int, int, dl::Bias<signed char> const*, dl::Activation<signed char> const*, std::vector<int, std::allocator<int> > const&) at /home/gansichen/Workspace/delete/aigit/master/dl/components/dl/nn/src/dl_nn_conv2d.cpp:72

0x4200b090: esp_dl_nn_conv2d_s8_wrapper at /home/gansichen/Workspace/delete/aigit/master/dl/components/dl/tvm/src/dl_nn_kernel.cpp:39

0x42008dea: tvmgen_default_esp_main_1 at /home/bergm/Desktop/SeedStudioModels/tvm_96x96x3_mobilenetv2_0.35/local_dummy_tvm_inference_project/new_project/components/tvm_model/model/codegen/host/src/default_lib6.c:14

0x420086ba: tvmgen_default___tvm_main__ at /home/bergm/Desktop/SeedStudioModels/tvm_96x96x3_mobilenetv2_0.35/local_dummy_tvm_inference_project/new_project/components/tvm_model/model/codegen/host/src/default_lib1.c:329

0x42007e4d: tvmgen_default_run at /home/bergm/Desktop/SeedStudioModels/tvm_96x96x3_mobilenetv2_0.35/local_dummy_tvm_inference_project/new_project/components/tvm_model/model/codegen/host/src/default_lib0.c:83007

0x42007dca: inference_task at /home/bergm/Desktop/SeedStudioModels/tvm_96x96x3_mobilenetv2_0.35/local_dummy_tvm_inference_project/new_project/main/app_main.c:30

0x4037a8e5: vPortTaskWrapper at /home/bergm/esp_v5.0/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:149
`

The model is based on: https://github.com/Seeed-Studio/sscma-model-zoo/blob/main/docs/en/Person_Classification_MobileNetV2_0.35_Rep_96.md where i downloaded the ONNX file and exported it with ESP-DL TVM.

I wonder why this happens even if the TVM export worked.

BlueSkyB, 2024-02-05:
The final error in the first version of MobilenetV2 is the native interface of TVM, which may be due to the model structure not being adapted.

The issue of second version of MobilenetV2 is strange. You can try this commit first to see if it can solve your problem: 97e1c52 in https://github.com/espressif/esp-dl.git.

In addition, the support for TVM in ESP-DL is not very mature, and there may be exceptions in the deployment of some custom models. Moreover, the main branch of TVM has undergone significant updates, so this implementation needs to be refactored. However, due to our current manpower constraints, we won't start the refactoring work in the short term; we need to complete other tasks first. So if any exceptions occur, you may need to handle them on your own in the short term. Sorry for the inconvenience.

hbellafkir, 2024-05-07:
@MichelBerg, can you share how exactly you moved the `.bbs` file to psram? I tried setting `CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY` in the sdkconfig file, but it still gives memory overflow for MobileNetV2. Here is the error: `esp32-template-project.elf section .dram0.bss will not fit in region dram0_0_seg`

closed 2024-10-21 by sun-xiangyu