espressif/esp-dl #154: ESP-DL tutorial/tvm_example No module named 'tvm.relay.op.contrib.esp' (IDFGH-12113)

MarkMendelsohn, 2024-02-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

successful build

### What is the actual behavior?

esp-dl/tutorial/tvm_example$ . ./script.sh 
Collecting tensor data and making histogram ...
Finding optimal threshold for each tensor using entropy algorithm ...
Number of tensors : 25
Number of histogram bins : 128 (The number may increase depends on the data it collects)
Number of quantized bins : 128
WARNING:root:Please use QuantFormat.QDQ for activation type QInt8 and weight type QInt8. Or it will lead to bad performance on x64.
Traceback (most recent call last):
  File "../../tools/tvm/export_onnx_model.py", line 11, in <module>
    from tvm.relay.op.contrib.esp import preprocess_onnx_for_esp, evaluate_onnx_for_esp
ModuleNotFoundError: No module named 'tvm.relay.op.contrib.esp'

### Steps to reproduce.

cd esp-dl/tutorial/tvm_example

. ./script.sh

### Build or installation Logs.

```plain
esp-dl/tutorial/tvm_example$ . ./script.sh 
Collecting tensor data and making histogram ...
Finding optimal threshold for each tensor using entropy algorithm ...
Number of tensors : 25
Number of histogram bins : 128 (The number may increase depends on the data it collects)
Number of quantized bins : 128
WARNING:root:Please use QuantFormat.QDQ for activation type QInt8 and weight type QInt8. Or it will lead to bad performance on x64.
Traceback (most recent call last):
  File "../../tools/tvm/export_onnx_model.py", line 11, in <module>
    from tvm.relay.op.contrib.esp import preprocess_onnx_for_esp, evaluate_onnx_for_esp
ModuleNotFoundError: No module named 'tvm.relay.op.contrib.esp'
```

### More Information.

Also tried on Mac Intel, M1 and M2 and Windows 11

BlueSkyB, 2024-02-19:
You can refer to the following link for details: https://docs.espressif.com/projects/esp-dl/en/latest/esp32s3/tutorials/deploying-models-through-tvm.html#set-tvm-package-path

Check if the PYTHONPATH environment variable is set to the correct value, ensuring that the path matches the local tvm/python folder path (export PYTHONPATH=$PYTHONPATH:/path-to-esp-dl/esp-dl/tools/tvm/python).

MarkMendelsohn, 2024-02-19:
On Ubuntu, the following output is produced:

esp-dl/tutorial/tvm_example$ . ./script.sh
Collecting tensor data and making histogram ...
Finding optimal threshold for each tensor using entropy algorithm ...
Number of tensors : 25
Number of histogram bins : 128 (The number may increase depends on the data it collects)
Number of quantized bins : 128
WARNING:root:Please use QuantFormat.QDQ for activation type QInt8 and weight type QInt8. Or it will lead to bad performance on x64.
Model Information:
------------------
Input Name: input_1
Input Shape: (1, 32, 32, 3)
Input DType: float
Output Name: Identity
Output Shape: (1, 10)
Output DType: float
One or more operators have not been tuned. Please tune your model for better performance. Use DEBUG logging level to see more details.
esp_dl_library_path: /media/mpm/FiveTDrive/esp-dl
generated project in: ./new_project

The environment variable is set:
$ env | grep PY
PYTHONPATH=/media/mpm/FiveTDrive/esp-dl/tools/tvm/python

Also:

esp-dl/tools/tvm/python/tvm$ file *.so
libtvm_runtime.so: ELF 64-bit LSB shared object, x86-64, version 1 (GNU/Linux), dynamically linked, BuildID[sha1]=b282dcf2103ca41ed42da59b5cab959277d8cac6, with debug_info, not stripped
libtvm.so:         ELF 64-bit LSB shared object, x86-64, version 1 (GNU/Linux), dynamically linked, BuildID[sha1]=c635d10c85e10f342ea6b2d48de8b8056125536b, with debug_info, not stripped

On the MAC:

BlueSkyB, 2024-02-20:
Your previous error was: “ModuleNotFoundError: No module named 'tvm.relay.op.contrib.esp'.”
So, has your problem been resolved on the Ubuntu platform? Currently, it is not supported on macOS platform.

closed 2024-10-21 by sun-xiangyu