espressif/esp-dl #155: ESP32 Model Deployment Project using TVM

YigitKaanBingol, 2024-03-11:
Hello guys,
I am trying to deploy my custom  Yolov5 model to Esp32s3, but i am facing that issue while i try to "export_onnx_model.py", I am getting this error.

`
Model Information:
------------------
Input Name: images
Input Shape: (1, 3, 192, 192)
Input DType: float
Output Name: output0
Output Shape: (1, 2268, 6)
Output DType: float
[13:06:06] /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/transforms/convert_layout.cc:99: Warning: Desired layout(s) not specified for op: nn.max_pool2d
[13:06:06] /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/transforms/convert_layout.cc:99: Warning: Desired layout(s) not specified for op: nn.max_pool2d
[13:06:06] /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/transforms/convert_layout.cc:99: Warning: Desired layout(s) not specified for op: nn.max_pool2d
[13:06:06] /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/transforms/convert_layout.cc:99: Warning: Desired layout(s) not specified for op: image.resize2d
[13:06:06] /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/transforms/convert_layout.cc:99: Warning: Desired layout(s) not specified for op: image.resize2d
Traceback (most recent call last):
  File "../../tools/tvm/export_onnx_model.py", line 237, in <module>
    dump_tvm_onnx_project(target_chip=args.target_chip, model_path=args.model_path, img_path=args.img_path,  template_path=args.template_path, generated_project_path=args.out_path)
  File "../../tools/tvm/export_onnx_model.py", line 167, in dump_tvm_onnx_project
    mod = preprocess_onnx_for_esp(mod, params)
  File "/home/train/esp-dl/tools/tvm/python/tvm/relay/op/contrib/esp.py", line 61, in preprocess_onnx_for_esp
    return seq(mod)
  File "/home/train/esp-dl/tools/tvm/python/tvm/ir/transform.py", line 160, in __call__
    return _ffi_transform_api.RunPass(self, mod)
  File "/home/train/esp-dl/tools/tvm/python/tvm/_ffi/_ctypes/packed_func.py", line 238, in __call__
    raise get_last_ffi_error()
tvm.error.InternalError: Traceback (most recent call last):
  84: 0xffffffffffffffff
  83: _start
  82: Py_BytesMain
  81: Py_RunMain
  80: PyRun_SimpleFileExFlags
  79: 0x0000000000673b60
  78: 0x0000000000673aba
  77: 0x0000000000673a40
  76: PyEval_EvalCode
  75: _PyEval_EvalCodeWithName
  74: _PyEval_EvalFrameDefault
  73: _PyFunction_Vectorcall
  72: _PyEval_EvalCodeWithName
  71: _PyEval_EvalFrameDefault
  70: _PyFunction_Vectorcall
  69: _PyEval_EvalCodeWithName
  68: _PyEval_EvalFrameDefault
  67: _PyObject_MakeTpCall
  66: 0x000000000057a42d
  65: _PyFunction_Vectorcall
  64: _PyEval_EvalFrameDefault
  63: _PyObject_MakeTpCall
  62: 0x000000000057a42d
  61: _PyFunction_Vectorcall
  60: _PyEval_EvalCodeWithName
  59: _PyEval_EvalFrameDefault
  58: _PyObject_MakeTpCall
  57: 0x00007f834ca199db
  56: _ctypes_callproc
  55: 0x00007f83422be409
  54: 0x00007f83422beff4
  53: tvm::runtime::TVMRetValue tvm::runtime::PackedFunc::operator()<tvm::IRModule, tvm::transform::PassContext>(tvm::IRModule&&, tvm::transform::PassContext&&) const
  52: operator()
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/backend/contrib/esp/generate_constants.cc:325
  51: tvm::relay::contrib::esp::GenerateConstants(tvm::IRModule const&)
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/backend/contrib/esp/generate_constants.cc:312
  50: tvm::relay::MixedModeMutator::VisitExpr(tvm::RelayExpr const&)
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/ir/expr_functor.cc:102
  49: ExpandDataflow<tvm::relay::MixedModeMutator::VisitExpr(const Expr&)::<lambda(const Expr&)>, tvm::relay::MixedModeMutator::VisitExpr(const Expr&)::<lambda(const Expr&)> >
        at /home/gansichen/Workspace/projects/local/framework/tvm/include/tvm/relay/expr_functor.h:509
  48: ExpandDataflow<tvm::relay::MixedModeMutator::VisitExpr(const Expr&)::<lambda(const Expr&)>, tvm::relay::MixedModeMutator::VisitExpr(const Expr&)::<lambda(const Expr&)>, tvm::relay::ExpandDataflow(tvm::relay::Expr, FCheckVisited, FVisitLeaf) [with FCheckVisited = tvm::relay::MixedModeMutator::VisitExpr(const Expr&)::<lambda(const Expr&)>; FVisitLeaf = tvm::relay::MixedModeMutator::VisitExpr(const Expr&)::<lambda(const Expr&)>]::<lambda(const Expr&)> >
[... 217 lines cut ...]
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/transforms/fold_constant.cc:202
  9: ConstEvaluate
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/transforms/fold_constant.cc:270
  8: tvm::relay::Eval(tvm::RelayExpr, tvm::runtime::Map<tvm::GlobalTypeVar, tvm::TypeData, void, void>, std::unordered_set<tvm::runtime::String, std::hash<tvm::runtime::String>, std::equal_to<tvm::runtime::String>, std::allocator<tvm::runtime::String> >, DLDevice, tvm::Target, tvm::runtime::Map<tvm::runtime::String, tvm::runtime::ObjectRef, void, void>)
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/backend/interpreter.cc:1108
  7: tvm::relay::Prepare(tvm::IRModule, tvm::CompilationConfig const&)
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/backend/interpreter.cc:961
  6: tvm::runtime::TVMRetValue tvm::runtime::PackedFunc::operator()<tvm::IRModule, tvm::transform::PassContext>(tvm::IRModule&&, tvm::transform::PassContext&&) const
  5: operator()
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/transforms/type_infer.cc:981
  4: tvm::relay::TypeInferencer::Infer(tvm::GlobalVar, tvm::relay::Function)
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/transforms/type_infer.cc:787
  3: tvm::relay::TypeInferencer::Solve()
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/transforms/type_infer.cc:623
  2: tvm::relay::TypeSolver::Solve()
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/analysis/type_solver.cc:632
  1: tvm::TypedEnvFunc<bool (tvm::runtime::Array<tvm::Type, void> const&, int, tvm::Attrs const&, tvm::TypeReporter const&)>::operator()(tvm::runtime::Array<tvm::Type, void> const&, int, tvm::Attrs const&, tvm::TypeReporter const&) const
        at /home/gansichen/Workspace/projects/local/framework/tvm/include/tvm/ir/env_func.h:142
  0: tvm::relay::SplitRel(tvm::runtime::Array<tvm::Type, void> const&, int, tvm::Attrs const&, tvm::TypeReporter const&)
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/op/tensor/transform.cc:3070
  File "/home/gansichen/Workspace/projects/local/framework/tvm/src/relay/analysis/type_solver.cc", line 643
InternalError: Check failed: (false) is false: [13:09:31] /home/gansichen/Workspace/projects/local/framework/tvm/src/relay/op/tensor/transform.cc:3070: InternalError: Check failed: (reporter->Assert(indices[i] > begin)) is false: indices_or_sections need to be a sorted ascending list
`

Thank you for any response.

AllenMao, 2024-03-12:
you can try non quantitative models，example：`python ../../tools/tvm/export_onnx_model.py --model_path ${onnx_model}_opt.onnx --img_path ${input_file}.npy --target_chip esp32s3 --out_path "." --template_path "../../tools/tvm/template_project_for_model/"`

YigitKaanBingol, 2024-03-12:
Ok, i'll try and share the results asap!
thank you.

AllenMao, 2024-03-13:
> Ok, i'll try and share the results asap! thank you.

upgrade onnx==1.13.0, It works!!!

YigitKaanBingol, 2024-03-19:
> > hare the results asap! thank you.
> 
> upgrade onnx==1.13.0, It works!!!

Non quantized solution works.
Thanks

Cilouche, 2024-03-21:
Hi,

I have the same problem, i was try to updrade onnx==1.13.0  but show me same problem
and when i use a Non quantized model another problem appear 
`nst, tvm::Range> > > const&, unsigned long, bool, std::unordered_set<tvm::tir::IterVar, std::hash<tvm::tir::IterVar>, std::equal_to<tvm::tir::IterVar>, std::allocator<tvm::tir::IterVar> > const&, std::unordered_map<tvm::tir::IterVar, tvm::PrimExpr, std::hash<tvm::tir::IterVar>, std::equal_to<tvm::tir::IterVar>, std::allocator<std::pair<tvm::tir::IterVar const, tvm::PrimExpr> > >*, bool)
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/te/operation/op_utils.cc:206
  3: tvm::te::PassUpIndex(tvm::te::Stage const&, tvm::runtime::Map<tvm::tir::IterVar, tvm::Range, void, void> const&, std::unordered_map<tvm::tir::IterVar, tvm::PrimExpr, std::hash<tvm::tir::IterVar>, std::equal_to<tvm::tir::IterVar>, std::allocator<std::pair<tvm::tir::IterVar const, tvm::PrimExpr> > >*, bool)
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/te/schedule/message_passing.cc:244
  2: tvm::indexdiv(tvm::PrimExpr, tvm::PrimExpr, tvm::Span)
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/tir/op/op.cc:399
  1: tvm::floordiv(tvm::PrimExpr, tvm::PrimExpr, tvm::Span)
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/tir/op/op.cc:409
  0: tvm::runtime::Optional<tvm::PrimExpr> tvm::arith::TryConstFold<tvm::tir::FloorDiv>(tvm::PrimExpr, tvm::PrimExpr)
        at /home/gansichen/Workspace/projects/local/framework/tvm/src/tir/transforms/../../arith/const_fold.h:273
  File "/home/gansichen/Workspace/projects/local/framework/tvm/src/arith/const_fold.h", line 302
InternalError: Check failed: pb->value != 0 (0 vs. 0) : Divide by zero`

Any suggestion please
thanks

MichelBerg, 2024-04-03:
Do you guys got running models with TVM ? Where you able to export the yolo model? I used TVM as well but it was inconsistent and buggy.

Cilouche, 2024-04-03:
> Do you guys got running models with TVM ? Where you able to export the yolo model? I used TVM as well but it was inconsistent and buggy.

Not yet, I'm still waiting for the return of @YigitKaanBingol

YigitKaanBingol, 2024-04-04:
Sorry guys, I've just able to deploy it and run the example, unfortunately i had no time to test accuracy of model.

Cilouche, 2024-04-04:
> Output Shape: (1, 2268, 6)

Output Shape: (1, 2268, **6**) the **6** indicates the number of classes, in fact you only trained on these 6 classes, right? if I understand correctly

Thanks

YigitKaanBingol, 2024-04-04:
I dont think so, i trained yolov5n model for one class, and export to .onnx format.
The reason for Output Shape: (1, 2268, 6), 6 is for x,y,w,h,confidence,class,
 if you have 80 classes,  your shape will be -> (1,2268,85)
2268 is based on your exportted model input shape.
I hope i made myself clear.

Cilouche, 2024-04-04:
> I dont think so, i trained yolov5n model for one class, and export to .onnx format. The reason for Output Shape: (1, 2268, 6), 6 is for x,y,w,h,confidence,class, if you have 80 classes, your shape will be -> (1,2268,85) 2268 is based on your exportted model input shape. I hope i made myself clear.

thank you for these answers

Labman4, 2024-07-12:
anyone solve the memory overflow with non quantitative models during complie

closed 2024-10-21 by sun-xiangyu