espressif/esp-dl #162: 参照官方提供的使用 ESP-DL 深度学习库基于 ESP32-S3 实现手势识别的这篇文章中在2.3评估里使用了里面的代码但是报错（前面完全按照示例走）

emoof1, 2024-04-24:
错误信息
[outputs, _] = eva.evalaute_quantized_model(test_images[i * batch_size:(i + 1) * batch_size], False)
AttributeError: 'Evaluator' object has no attribute 'evalaute_quantized_model'

Sandra-lol, 2024-04-24:
请问是什么文章？

emoof1, 2024-04-24:
> 请问是什么文章？

[image]

emoof1, 2024-04-24:
> 请问是什么文章？

https://zhuanlan.zhihu.com/p/590495368

emoof1, 2024-04-24:
以及在3.2.3初始化层里的代码中直接运行会显示
error: 'get_statefulpartitionedcall_sequential_1_conv2d_3_biasadd_filter' was 
not declared in this scope
我并没有在文章中找到get_statefulpartitionedcall_sequential_1_conv2d_3_biasadd_filter()相关的定义

emoof1, 2024-04-24:
以及按照3.3.3，3.3.4，3.3.5整合起来后程序跑完却没有输出
[main.zip](https://github.com/espressif/esp-dl/files/15094201/main.zip)
This is my project it based on ESP-DL

BlueSkyB, 2024-04-26:
> 错误信息 [outputs, _] = eva.evalaute_quantized_model(test_images[i * batch_size:(i + 1) * batch_size], False) AttributeError: 'Evaluator' object has no attribute 'evalaute_quantized_model'

DL 里面这个接口名应该是'evalute_quantized_model'

emoof1, 2024-04-26:
> > 错误信息 [outputs, _] = eva.evalaute_quantized_model(test_images[i * batch_size:(i + 1) * batch_size], False) AttributeError: 'Evaluator' object has no attribute 'evalaute_quantized_model'
> 
> DL 里面这个接口名应该是'evalute_quantized_model'

Traceback (most recent call last):
  File "H:\opencvprojiect\ESP-test\change\评估.py", line 33, in <module>
    [outputs, _] = eva.evalute_quantized_model(test_images[i * batch_size:(i + 1) * batch_size], True)
  File "evaluator.py", line 1173, in evaluator.Evaluator.evalute_quantized_model
  File "D:\python\lib\site-packages\onnxruntime\capi\onnxruntime_inference_collection.py", line 280, in __init__
    self._create_inference_session(providers, provider_options)
  File "D:\python\lib\site-packages\onnxruntime\capi\onnxruntime_inference_collection.py", line 309, in _create_inference_session
    sess = C.InferenceSession(session_options, self._model_bytes, False, self._read_config_from_model)
onnxruntime.capi.onnxruntime_pybind11_state.InvalidGraph: [ONNXRuntimeError] : 10 : INVALID_GRAPH : This is an invalid model. Error in Node:StatefulPartitionedCall/sequential/conv2d/BiasAdd/QACC_clip0 : Node (StatefulPartitionedCall/sequential/conv2d/BiasAdd/QACC_clip0) has input size 3 not in range [min=1, max=1].
但是出现了新的错误

BlueSkyB, 2024-04-26:
> > > 错误信息 [outputs, _] = eva.evalaute_quantized_model(test_images[i * batch_size:(i + 1) * batch_size], False) AttributeError: 'Evaluator' object has no attribute 'evalaute_quantized_model'
> > 
> > 
> > DL 里面这个接口名应该是'evalute_quantized_model'
> 
> Traceback (most recent call last): File "H:\opencvprojiect\ESP-test\change\评估.py", line 33, in [outputs, _] = eva.evalute_quantized_model(test_images[i * batch_size:(i + 1) * batch_size], True) File "evaluator.py", line 1173, in evaluator.Evaluator.evalute_quantized_model File "D:\python\lib\site-packages\onnxruntime\capi\onnxruntime_inference_collection.py", line 280, in **init** self._create_inference_session(providers, provider_options) File "D:\python\lib\site-packages\onnxruntime\capi\onnxruntime_inference_collection.py", line 309, in _create_inference_session sess = C.InferenceSession(session_options, self._model_bytes, False, self._read_config_from_model) onnxruntime.capi.onnxruntime_pybind11_state.InvalidGraph: [ONNXRuntimeError] : 10 : INVALID_GRAPH : This is an invalid model. Error in Node:StatefulPartitionedCall/sequential/conv2d/BiasAdd/QACC_clip0 : Node (StatefulPartitionedCall/sequential/conv2d/BiasAdd/QACC_clip0) has input size 3 not in range [min=1, max=1]. 但是出现了新的错误

这个提示是说 clip 算子，有三个输入，不符合 一个输入的要求(DL 中构建Clip算子是按三个输入构建ONNX Node的)。可以检查下你导出ONNX时，使用的ONNX版本，老的ONNX版本Clip算子是只有一个算入的。

emoof1, 2024-04-26:
> > > > 错误信息 [outputs, _] = eva.evalaute_quantized_model(test_images[i * batch_size:(i + 1) * batch_size], False) AttributeError: 'Evaluator' object has no attribute 'evalaute_quantized_model'
> > > 
> > > 
> > > DL 里面这个接口名应该是'evalute_quantized_model'
> > 
> > 
> > Traceback (most recent call last): File "H:\opencvprojiect\ESP-test\change\评估.py", line 33, in [outputs, _] = eva.evalute_quantized_model(test_images[i * batch_size:(i + 1) * batch_size], True) File "evaluator.py", line 1173, in evaluator.Evaluator.evalute_quantized_model File "D:\python\lib\site-packages\onnxruntime\capi\onnxruntime_inference_collection.py", line 280, in **init** self._create_inference_session(providers, provider_options) File "D:\python\lib\site-packages\onnxruntime\capi\onnxruntime_inference_collection.py", line 309, in _create_inference_session sess = C.InferenceSession(session_options, self._model_bytes, False, self._read_config_from_model) onnxruntime.capi.onnxruntime_pybind11_state.InvalidGraph: [ONNXRuntimeError] : 10 : INVALID_GRAPH : This is an invalid model. Error in Node:StatefulPartitionedCall/sequential/conv2d/BiasAdd/QACC_clip0 : Node (StatefulPartitionedCall/sequential/conv2d/BiasAdd/QACC_clip0) has input size 3 not in range [min=1, max=1]. 但是出现了新的错误
> 
> 这个提示是说 clip 算子，有三个输入，不符合 一个输入的要求(DL 中构建Clip算子是按三个输入构建ONNX Node的)。可以检查下你的ONNX以及ONNXRUNTIME版本，新的ONNX Clip算子定义是按三个输入来的。

[image]
看过版本号了和官方的一样 但是int8的那个就是跑不起来也不知道怎么改

[image]

BlueSkyB, 2024-04-29:
我重新创建新的环境，下载依赖库，使用你参考的知乎博文使用的代码和onnx，是没有问题的。(博文里有给链接：https://github.com/alibukharai/Blogs/tree/main/ESP-DL/esp_dl_formate_conversion)
所以库没有问题，你再检查下自己的环境和代码。

emoof1, 2024-04-29:
> 我重新创建新的环境，下载依赖库，使用你参考的知乎博文使用的代码和onnx，是没有问题的。(博文里有给链接：https://github.com/alibukharai/Blogs/tree/main/ESP-DL/esp_dl_formate_conversion) 所以库没有问题，你再检查下自己的环境和代码。

okk 感谢感谢 我再看看 我还有一个问题 就是如果要处理RGB彩图的我该怎么修改已知我的数据集是RGB24位的 但是设备里的彩图是RGB565的 有什么合理的解决方案吗

closed 2024-05-04 by emoof1