espressif/esp-dl #163: I need help about “assert failed: dl::Tensor<T>& dl::Tensor<T>::reshape(std::vector<int>) [with T = short int] dl_variable.cpp:77 (size == size_gt)“

emoof1, 2024-05-04:
---------This is my camera config
    .pixel_format = PIXFORMAT_RGB565, 
    .frame_size = FRAMESIZE_96X96,     
---------This is my model on python
print(tf.__version__)

model = Sequential()
model.add(Conv2D(32, (5, 5), activation='relu', input_shape=(96, 96, 3)))
model.add(MaxPooling2D((2, 2)))
model.add(Dropout(0.2))
model.add(Conv2D(64, (3, 3), activation='relu'))
model.add(MaxPooling2D((2, 2)))
model.add(Dropout(0.2))
model.add(Conv2D(64, (3, 3), activation='relu'))
model.add(MaxPooling2D((2, 2)))
model.add(Dropout(0.2))
model.add(Conv2D(64, (3, 3), activation='relu'))
model.add(MaxPooling2D((2, 2)))
model.add(Dropout(0.2))
model.add(Conv2D(64, (3, 3), activation='relu'))
model.add(MaxPooling2D((2, 2)))
model.add(Flatten())
model.add(Dense(128, activation='relu'))
model.add(Dense(64, activation='relu'))
model.add(Dense(32, activation='relu'))
model.add(Dense(2, activation='softmax'))

model.compile(optimizer='adam',loss='sparse_categorical_crossentropy',metrics=['accuracy'])

---------This is my model on C produced by python

#pragma once
#include <stdint.h>
#include "dl_layer_model.hpp"
#include "dl_layer_base.hpp"
#include "dl_layer_max_pool2d.hpp"
#include "dl_layer_conv2d.hpp"
#include "dl_layer_reshape.hpp"
#include "dl_layer_softmax.hpp"
#include "handrecognition_coefficient.hpp"
#include "handrecognition_coefficient.cpp"
using namespace dl;
using namespace layer;
using namespace handrecognition_coefficient;
class HANDRECOGNITION : public Model<int16_t> 
{
private:
    Conv2D<int16_t> l1;
    MaxPool2D<int16_t> l2;
    Conv2D<int16_t> l3;
    MaxPool2D<int16_t> l4;
    Conv2D<int16_t> l5;
    MaxPool2D<int16_t> l6;
	Conv2D<int16_t> l7;
    MaxPool2D<int16_t> l8;
	Conv2D<int16_t> l9;
    MaxPool2D<int16_t> l10;
    Reshape<int16_t> l11;
    Conv2D<int16_t> l12;
    Conv2D<int16_t> l13;
    Conv2D<int16_t> l14;
    Conv2D<int16_t> l15;	
public:
    Softmax<int16_t> l16; // output layer
HANDRECOGNITION () : 
                     l1(Conv2D<int16_t>(-8, get_statefulpartitionedcall_sequential_conv2d_biasadd_filter(), get_statefulpartitionedcall_sequential_conv2d_biasadd_bias(), get_statefulpartitionedcall_sequential_conv2d_biasadd_activation(), PADDING_VALID, {}, 1,1, "l1")),
                     l2(MaxPool2D<int16_t>({2,2},PADDING_VALID, {}, 2, 2, "l2")),                      
                     l3(Conv2D<int16_t>(-9, get_statefulpartitionedcall_sequential_conv2d_1_biasadd_filter(), get_statefulpartitionedcall_sequential_conv2d_1_biasadd_bias(), get_statefulpartitionedcall_sequential_conv2d_1_biasadd_activation(), PADDING_VALID,{}, 1,1, "l3")),                       
                     l4(MaxPool2D<int16_t>({2,2},PADDING_VALID,{}, 2, 2, "l4")),                       
                     l5(Conv2D<int16_t>(-9, get_statefulpartitionedcall_sequential_conv2d_2_biasadd_filter(), get_statefulpartitionedcall_sequential_conv2d_2_biasadd_bias(), get_statefulpartitionedcall_sequential_conv2d_2_biasadd_activation(), PADDING_VALID,{}, 1,1, "l5")),                    
[... 146 lines cut ...]
				case 0:
				ESP_LOGI(TAG,"Phone:0");
				break;
				case 1:
				ESP_LOGI(TAG,"No_Phone:1");
				break;
				default:
				ESP_LOGI(TAG,"No result");
			}
			ESP_LOGI(TAG,"\n");
			esp_camera_fb_return(fbd);
			xSemaphoreGive(xMutexInventory);
			fbd=NULL;
			vTaskDelay(5000 / portTICK_PERIOD_MS);
		}else{
			vTaskDelay(1000 / portTICK_PERIOD_MS);
		}
	}
}
but there is a error :
assert failed: dl::Tensor<T>& dl::Tensor<T>::reshape(std::vector<int>) [with T = short int] dl_variable.cpp:77 (size == size_gt)
Backtrace: 0x403760c6:0x3fceb880 0x4037c691:0x3fceb8a0 0x40382795:0x3fceb8c0 0x4201746b:0x3fceb9e0 0x4200cea6:0x3fceba10 0x42017db2:0x3fcebab0 0x4200b899:0x3fcebad0 0x4037f325:0x3fcebf20
ELF file SHA256: 768f2a452bcd0c8f
I don't know how to solve it , if you kown please tell me .
Thanks !

100312dog, 2024-05-06:
@emoof1 
┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━┓
┃ Layer (type)                    ┃ Output Shape           ┃       Param # ┃
┡━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━┩
│ conv2d (Conv2D)                 │ (None, 92, 92, 32)     │         2,432 │
├─────────────────────────────────┼────────────────────────┼───────────────┤
│ max_pooling2d (MaxPooling2D)    │ (None, 46, 46, 32)     │             0 │
├─────────────────────────────────┼────────────────────────┼───────────────┤
│ dropout (Dropout)               │ (None, 46, 46, 32)     │             0 │
├─────────────────────────────────┼────────────────────────┼───────────────┤
│ conv2d_1 (Conv2D)               │ (None, 44, 44, 64)     │        18,496 │
├─────────────────────────────────┼────────────────────────┼───────────────┤
│ max_pooling2d_1 (MaxPooling2D)  │ (None, 22, 22, 64)     │             0 │
├─────────────────────────────────┼────────────────────────┼───────────────┤
│ dropout_1 (Dropout)             │ (None, 22, 22, 64)     │             0 │
├─────────────────────────────────┼────────────────────────┼───────────────┤
│ conv2d_2 (Conv2D)               │ (None, 20, 20, 64)     │        36,928 │
├─────────────────────────────────┼────────────────────────┼───────────────┤
│ max_pooling2d_2 (MaxPooling2D)  │ (None, 10, 10, 64)     │             0 │
├─────────────────────────────────┼────────────────────────┼───────────────┤
│ dropout_2 (Dropout)             │ (None, 10, 10, 64)     │             0 │
├─────────────────────────────────┼────────────────────────┼───────────────┤
│ conv2d_3 (Conv2D)               │ (None, 8, 8, 64)       │        36,928 │
├─────────────────────────────────┼────────────────────────┼───────────────┤
│ max_pooling2d_3 (MaxPooling2D)  │ (None, 4, 4, 64)       │             0 │
├─────────────────────────────────┼────────────────────────┼───────────────┤
│ dropout_3 (Dropout)             │ (None, 4, 4, 64)       │             0 │
├─────────────────────────────────┼────────────────────────┼───────────────┤
│ conv2d_4 (Conv2D)               │ (None, 2, 2, 64)       │        36,928 │
├─────────────────────────────────┼────────────────────────┼───────────────┤
│ max_pooling2d_4 (MaxPooling2D)  │ (None, 1, 1, 64)       │             0 │
├─────────────────────────────────┼────────────────────────┼───────────────┤
│ flatten (Flatten)               │ (None, 64)             │             0 │
├─────────────────────────────────┼────────────────────────┼───────────────┤
│ dense (Dense)                   │ (None, 128)            │         8,320 │
├─────────────────────────────────┼────────────────────────┼───────────────┤
│ dense_1 (Dense)                 │ (None, 64)             │         8,256 │
├─────────────────────────────────┼────────────────────────┼───────────────┤
│ dense_2 (Dense)                 │ (None, 32)             │         2,080 │
├─────────────────────────────────┼────────────────────────┼───────────────┤
│ dense_3 (Dense)                 │ (None, 2)              │            66 │
└─────────────────────────────────┴────────────────────────┴───────────────┘
use model.summary() to see output shape for each layer.
the reason why you encounter the assert failed error here is that you try to reshape a (1, 1, 64) shape tensor to (1, 1, 6400)
just use ```Flatten``` here.
btw, dense layers should use ```FullyConnected``` .

```
#include "dl_layer_fullyconnected.hpp"
#include "dl_layer_flatten.hpp"
Flatten<int16_t> l11;
FullyConnected<int16_t> l12;
FullyConnected<int16_t> l13;
FullyConnected<int16_t> l14;
FullyConnected<int16_t> l15;
```

emoof1, 2024-05-07:
@100312dog  Thank you very much but I don‘t know how to define Flatten and FullyConnected in HANDRECOGNITION () : 。
I was hoping you could share something about the usage of them like l1(Conv2D<int16_t>(-8, get_statefulpartitionedcall_sequential_conv2d_biasadd_filter(), get_statefulpartitionedcall_sequential_conv2d_biasadd_bias(), get_statefulpartitionedcall_sequential_conv2d_biasadd_activation(), PADDING_VALID, {}, 1,1, "l1"))。I used is ESP32-DL storehouse。Extremely grateful

emoof1, 2024-05-07:
@100312dog  By the way, I would like to know how to achieve targeting using ESP-DL do you have any good suggestions or references to articles?

100312dog, 2024-05-07:
@emoof1 
see constructor of them in 
https://github.com/espressif/esp-dl/blob/master/include/layer/dl_layer_fullyconnected.hpp
https://github.com/espressif/esp-dl/blob/master/include/layer/dl_layer_flatten.hpp

```
            /**
             * @brief Construct a new Flatten object
             * 
             * @param name name of layer
             * @param inplace true: the output will store to input0
             *                false: the output will store to a separate memory
             */
            Flatten(const char *name = "Flatten", bool inplace = false) : Layer(name), output(NULL), inplace(inplace), output_shape({})
            {}

```

```
            /**
             * @brief Construct a new FullyConnected object.
             * 
             * @param output_exponent exponent of output
             * @param filter          filter of FullyConnected
             * @param bias            bias of FullyConnected, if you don't specify anything, no bias is added
             * @param activation      activation of FullyConnected, if you don't specify anything, no activation is applied
             * @param flatten         true: input shape is [x1, x2, ..., xn], filter shape is [1, 1, x1 * x2 * ... * xn, output_dim], output shape is [output_dim]
                                      false: input shape is [x1, x2, ..., xn, input_dim], filter shape is [1, 1, input_dim, output_dim], output shape is [x1, x2, ...., xn, output_dim]
             * @param name            name of layer
             */
            FullyConnected(const int output_exponent,
                           const Filter<feature_t> *filter,
                           const Bias<bias_t> *bias = NULL,
                           const Activation<feature_t> *activation = NULL,
                           const bool flatten = true,
                           const char *name = "FullyConnected") : Layer(name),
                                                                  output_exponent(output_exponent),
                                                                  flatten(flatten),
                                                                  filter(filter),
                                                                  bias(bias),
                                                                  activation(activation),
                                                                  output_shape({})
```
according your code provided, the usage should be
```
l11(Flatten<int16_t>("l11", false)),
l12(FullyConnected<int16_t>(-11, get_fused_gemm_0_filter(), get_fused_gemm_0_bias(), get_fused_gemm_0_activation(), true, "l12")),
l13(FullyConnected<int16_t>(-11, get_fused_gemm_1_filter(), get_fused_gemm_1_bias(), get_fused_gemm_1_activation(), true, "l13")),
l14(FullyConnected<int16_t>(-11, get_fused_gemm_2_filter(), get_fused_gemm_2_bias(), get_fused_gemm_2_activation(), true, "l14")),
l15(FullyConnected<int16_t>(-13, get_fused_gemm_3_filter(), get_fused_gemm_3_bias(), true, "l15")),
```

closed 2024-10-14 by sun-xiangyu