espressif/esp-dl #166: Accessing dl::Tensor Class Data and Resizing Image

hifzhil, 2024-07-14:
Hello,

I am currently working on a project where I need to resize a camera_fb_t frame data using the dl::image::resize_image function. Here is the function signature for reference:
```
/**
 * @brief Resize an image to the target shape.
 * 
 * @param image The input image Tensor.
 * @param target_shape The target shape of the resized image.
 * @param resize_type One of IMAGE_RESIZE_BILINEAR, IMAGE_RESIZE_MEAN, or IMAGE_RESIZE_NEAREST.
 * @return Tensor<uint8_t>* Pointer to the resized image Tensor.
 */
Tensor<uint8_t> *resize_image(Tensor<uint8_t> &image, std::vector<int> target_shape, resize_type_t resize_type);

```

To achieve this, I need to create containers for storing the input and output tensors. Here's the code snippet I am using:
```
dl::Tensor<uint8_t> inputImage;
inputImage.data = frame->buf;
inputImage.shape = {frame->width, frame->height, 3};

std::vector<int> targetShape = {28, 28, 1};
dl::image::resize_type_t resizeType = dl::image::IMAGE_RESIZE_BILINEAR;

dl::Tensor<uint8_t>* resizedImage = dl::image::resize_image(inputImage, targetShape, resizeType);

```

However, I encountered an error indicating that the dl::Tensor<unsigned char> class has no member named data. Here is the error message for reference:
`class dl::Tensor<unsigned char>' has no member named 'data'`

I have searched for documentation on the dl::Tensor class but could not find any comprehensive information. Could you please assist me in understanding the correct way to access and manipulate the data within the dl::Tensor class? Specifically, I need to know:

How to correctly assign the frame->buf data to the inputImage tensor.
How to correctly create and handle the output tensor returned by the resize_image function.
Any guidance or examples would be greatly appreciated.

Thank you for your help.

Best regards,
Hifzhil

100312dog, 2024-07-15:
@hifzhil 
you can find dl::Tensor interface in [https://github.com/espressif/esp-dl/blob/master/include/typedef/dl_variable.hpp](url)
```
dl::Tensor<uint8_t> inputImage;
inputImage.set_shape({frame->width, frame->height, 3}).set_element((uint8_t *)frame->buf).set_auto_free(false);

std::vector<int> targetShape = {28, 28, 1};
dl::image::resize_type_t resizeType = dl::image::IMAGE_RESIZE_BILINEAR;

dl::Tensor<uint8_t>* resizedImage = dl::image::resize_image(inputImage, targetShape, resizeType);
```
Here you should set_auto_free(false). 
Becuase your input data memory is not mangaged by dl::Tensor. However, the default value of auto_free in dl::Tensor is true.

hifzhil, 2024-07-15:
Thanks a lot for your help. I finally understand it now!

Really appreciate it.

closed 2024-07-15 by hifzhil