espressif/esp-dl #167: Update the prebuilt library to the more recent version of python

MR-VENOM, 2024-07-22:
All the pre-built library that are used in are compiled using the python version 3,7 and the modules are also quite outdated  will there be an updated version with more recent version of python and and the onnx modules.

BlueSkyB, 2024-07-25:
Due to some inherent issues with the ESP-DL architecture, we are currently refactoring ESP-DL. 
During the refactoring process, we will correct some issues, add some convenient and practical new features, support our new P4 AI chip, and support the latest IDF V5.3. These are currently under development and will be released once completed. Subsequently, the focus will be on maintaining the new ESP-DL, while the old ESP-DL will be maintained as is.

MR-VENOM, 2024-07-25:
Hi @BlueSkyB  thank you fro your response, Will the esp32s3 be still supported on the new ? if so can you provide when will be this be available for testing ?

closed 2024-10-21 by sun-xiangyu