espressif/esp-dl #168: Quantization Layer Not supported: Softmax and constant

raprakashvi, 2024-07-25:
Hi, 
I am following the tutorial https://blog.espressif.com/hand-gesture-recognition-on-esp32-s3-with-esp-deep-learning-176d7e13fd37 and have been getting issue on quantization step. 

My python version is 3.7 and whenever I get to quantization , I get the error:

raprakashvi, 2024-07-25:
@robinvanemden @kedars any thoughts?

BlueSkyB, 2024-07-30:
For the constant operator, you can add 'extract_constant_to_initializer' in the onnxoptimizer.optimize() within the optimize_fp_model function. Refer to this commit for the modification: https://github.com/espressif/esp-dl/commit/020231ea04ef27e8cf23471a15365513a46eeeeb.

For the LogSoftmax operator, esp-dl does not currently support it; you can check your ONNX file again.

closed 2024-10-14 by sun-xiangyu