espressif/esp-dl #169: When will the ESP-DL library support ESP32-P4

mootterry, 2024-07-31:


sun-xiangyu, 2024-08-01:
In the new version, we've done significant refactoring and plan to release the source code for optimized operators, automatic computation graph loading, and memory planning.

The beta version, supporting ESP32-P4, will be out before August 30th. 
The official release, scheduled for September 30th, will support both ESP32-S3 and ESP32-P4, with more documentation.

sun-xiangyu, 2024-09-02:
please refer to [release/esp32p4](https://github.com/espressif/esp-dl/tree/release/esp32p4). It is not compatible with the master branch, and currently supports only the ESP32-P4.
We will be adding more documentation and support ESP32-S3 in the next month.

nicklasb, 2024-09-11:
However, the release/esp32p4 does not currently compile, can we fix at least that, we can test it out:
```
managed_components\espressif__esp-dl\dl\typedef\src\dl_tensor_base.cpp: In function 'int dl::quantize(float, float, float, float)':
managed_components\espressif__esp-dl\dl\typedef\src\dl_tensor_base.cpp:6:18: error: 'dl_esp32p4_round_half_even' was not declared in this scope
    6 |     int output = dl_esp32p4_round_half_even(input * scale);
```

(also, it is after August 30th. :-) )

sun-xiangyu, 2024-09-12:
@nicklasb,  Thank you for your feedback.
I have updated some code and added CI to ensure that all examples are compilable. Please try again.

sun-xiangyu, 2024-09-12:
The new ESP-DL has a different usage from the previous one. The previous ESP-DL has been moved to esp-dl/esp-dl. This folder is a separate component that includes all APIs and operator implementations. The esp-dl/models directory is used to store specific model files and running code separately. The examples directory is for demonstrating how to use ESP-DL and models.

nicklasb, 2024-09-12:
> @nicklasb, Thank you for your feedback. I have updated some code and added CI to ensure that all examples are compilable. Please try again.

I am afraid I am getting the exact same error:
```
esp-dl/esp-dl/dl/typedef/src/dl_tensor_base.cpp:6:18: error: 'dl_esp32p4_round_half_even' was not declared in this scope  
    6 |     int output = dl_esp32p4_round_half_even(input * scale);
```

dl_esp32p4_round_half_even() doesn't seem to be *anywhere*.

sun-xiangyu, 2024-09-13:
dl_esp32p4_round_half_even: https://github.com/espressif/esp-dl/blob/release/esp32p4/esp-dl/dl/tool/isa/esp32p4/dl_esp32p4_round.S

nicklasb, 2024-09-15:
@sun-xiangyu  

Ok, now it worked, however the 
```
    delete detect;
    delete pedestrian;
```
..causes errors, seems there is some issue with the cleanup:
```

I (33) boot: ESP-IDF v5.3 2nd stage bootloader
I (33) boot: compile time Sep 14 2024 13:03:22
I (34) boot: Multicore bootloader
I (37) boot: chip revision: v0.1
I (40) qio_mode: Enabling default flash chip QIO
I (45) boot.esp32p4: SPI Speed      : 80MHz
I (50) boot.esp32p4: SPI Mode       : QIO
I (55) boot.esp32p4: SPI Flash Size : 16MB
I (59) boot: Enabling RNG early entropy source...
I (65) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (76) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  2 factory          factory app      00 00 00010000 007d0000
I (99) boot: End of partition table
I (102) esp_image: segment 0: paddr=00010020 vaddr=480c0020 size=4f05ch (323676) map
I (160) esp_image: segment 1: paddr=0005f084 vaddr=30100000 size=00020h (    32) load
I (161) esp_image: segment 2: paddr=0005f0ac vaddr=30100020 size=0003ch (    60) load
I (166) esp_image: segment 3: paddr=0005f0f0 vaddr=4ff00000 size=00f28h (  3880) load
I (176) esp_image: segment 4: paddr=00060020 vaddr=48000020 size=b3474h (734324) map
I (292) esp_image: segment 5: paddr=0011349c vaddr=4ff00f28 size=104c0h ( 66752) load
I (305) esp_image: segment 6: paddr=00123964 vaddr=4ff11400 size=02d34h ( 11572) load
I (312) boot: Loaded app from partition at offset 0x10000
I (312) boot: Disabling RNG early entropy source...
I (326) hex_psram: vendor id    : 0x0d (AP)
I (326) hex_psram: Latency      : 0x01 (Fixed)
I (326) hex_psram: DriveStr.    : 0x00 (25 Ohm)
I (329) hex_psram: dev id       : 0x03 (generation 4)
I (335) hex_psram: density      : 0x07 (256 Mbit)
I (341) hex_psram: good-die     : 0x06 (Pass)
I (346) hex_psram: SRF          : 0x02 (Slow Refresh)
I (351) hex_psram: BurstType    : 0x00 ( Wrap)
I (356) hex_psram: BurstLen     : 0x03 (2048 Byte)
I (362) hex_psram: BitMode      : 0x01 (X16 Mode)
I (367) hex_psram: Readlatency  : 0x04 (14 cycles@Fixed)
I (373) hex_psram: DriveStrength: 0x00 (1/1)
I (379) MSPI DQS: tuning success, best phase id is 2
I (571) MSPI DQS: tuning success, best delayline id is 11
I esp_psram: Found 32MB PSRAM device
I esp_psram: Speed: 200MHz
I (572) mmu_psram: flash_drom_paddr_start: 0x10000
I (606) mmu_psram: flash_irom_paddr_start: 0x60000
I (675) hex_psram: psram CS IO is dedicated
I (675) cpu_start: Multicore app
I (1584) esp_psram: SPI SRAM memory test OK
W (1593) clk: esp_perip_clk_init() has not been implemented yet
I (1600) cpu_start: Pro cpu start user code
I (1600) cpu_start: cpu freq: 360000000 Hz
I (1601) app_init: Application information:
I (1603) app_init: Project name:     human_face_detect
I (1609) app_init: App version:      v2.0-36-g9a0d2a6
I (1615) app_init: Compile time:     Sep 14 2024 13:02:48
I (1621) app_init: ELF file SHA256:  6f4af5b36...
0x4ff034b0: cpu_utility_ll_reset_cpu at C:/Espressif/frameworks/esp-idf-v5.3/components/hal/esp32p4/include/hal/cpu_utility_ll.h:23
 (inlined by) esp_cpu_reset at C:/Espressif/frameworks/esp-idf-v5.3/components/esp_hw_support/cpu.c:49

0x4ff03620: esp_cpu_wait_for_intr at C:/Espressif/frameworks/esp-idf-v5.3/components/esp_hw_support/cpu.c:57 (discriminator 1)

I (1626) app_init: ESP-IDF:          v5.3
I (1631) efuse_init: Min chip rev:     v0.1
[... 134 lines cut ...]
4ff19630: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff19650: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000007 0x4ff18688 0x4ff186a0 0x00000000
4ff19670: 0x4ff18688 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x4ff18f90
4ff19690: 0x00000000 0x4ff14364 0x4ff14364 0x4ff1968c 0x4ff1435c 0x00000018 0x4ff2beb4 0x4ff2beb4
4ff196b0: 0x4ff1968c 0x00000000 0x00000001 0x4ff186a0 0x6e69616d 0x00000000 0x00000000 0x00000000
4ff196d0: 0x00000000 0x4ff19680 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000 0x4ff17730
4ff196f0: 0x4ff17798 0x4ff17800 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x4ff298a0
4ff19710: 0x4fc0aa82 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff19730: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff19750: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff19770: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
0x00000000 0x00000000 0x00000000
4ff197b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff197d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000600 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff197f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff19810: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff19830: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff19850: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff19870: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff19890: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff198b0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff198d0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff198f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

```

closed 2024-11-01 by sun-xiangyu