espressif/esp-dl #171: 无法链接 lib/mkmodel_windows

Criminal-9527, 2024-09-23:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

我在使用新版本的构建系统来构建pose_estimation这个项目，但是无法链接 lib/mkmodel_windows，我的cmakefile.txt如下：
```
idf_component_register(
    SRCS
        "main.c"
        "pe_forward.c"
        "image_util.c"
    INCLUDE_DIRS
        "."
        "../include"
)

target_link_libraries(${COMPONENT_LIB} INTERFACE "D:/Espressif/frameworks/esp-idf-v5.2.2/examples/dl-camera/lib/libdetection.a")
target_link_libraries(${COMPONENT_LIB} INTERFACE "D:/Espressif/frameworks/esp-idf-v5.2.2/examples/dl-camera/lib/libdl.a")
target_link_libraries(${COMPONENT_LIB} INTERFACE "D:/Espressif/frameworks/esp-idf-v5.2.2/examples/dl-camera/lib/libfd.a")
target_link_libraries(${COMPONENT_LIB} INTERFACE "D:/Espressif/frameworks/esp-idf-v5.2.2/examples/dl-camera/lib/libfr.a")
target_link_libraries(${COMPONENT_LIB} INTERFACE "D:/Espressif/frameworks/esp-idf-v5.2.2/examples/dl-camera/lib/libod.a")
target_link_libraries(${COMPONENT_LIB} INTERFACE "D:/Espressif/frameworks/esp-idf-v5.2.2/examples/dl-camera/lib/libpe.a")
target_link_libraries(${COMPONENT_LIB} INTERFACE "D:/Espressif/frameworks/esp-idf-v5.2.2/examples/dl-camera/lib/mkmodel_windows")
```
在链接lib/mkmodel_windows时报错说无法识别这个文件格式：
/lib/mkmodel_windows: file not recognized: file format not recognized
在原本的构建系统中我也没找到怎么链接这个文件的（请原谅我的构建系统学的稀烂，上面直接全部使用绝对路径了，但是原本项目中一些只有声明没定义的函数，在使用上面方法链接之后能找到了，我觉得这种链接办法还是能生效的，只是对于mkmodel_windows这个文件有问题）

BlueSkyB, 2024-09-23:
esp-dl 也没有 mkmodel_windows 这个文件，是你自己的文件吗？
然后目前 esp-dl master 分支，使用idf v5.2可能会有兼容性问题，推荐用v5.1

Criminal-9527, 2024-09-23:
在下面这个链接里：
https://github.com/espressif/esp-dl/tree/420fc7e219ba98e40a5493b9d4be270db2f2d724
应该是之前版本里的一个例子吧，最新版里面好像没这个例子了，就是手掌检测和手部特征点检测的这个例子

sun-xiangyu, 2024-10-14:
后续会添加相关模型

closed 2024-10-14 by sun-xiangyu