espressif/esp-dl #173: Linkin undefined reference on esp-dl example (IDF 5.3.0)

simoberny, 2024-10-01:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I'm trying to compile new ESP-DL example for my ESP32-S3. 

### Actual behavior (suspected bug)

Compilation goes clean, but when linking I got undefined reference on multiple fbs component

### Error logs or terminal output

```shell
[1/6] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\Simone.Bernabe\workspace\esp32\esp-dl\examples\pedestrian_detect\build\bootloader\esp-idf\esptool_py && C:\Users\Simone.Bernabe\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe C:/Users/Simone.Bernabe/esp/v5.3.1/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 C:/Users/Simone.Bernabe/workspace/esp32/esp-dl/examples/pedestrian_detect/build/bootloader/bootloader.bin"
Bootloader binary size 0x5810 bytes. 0x27f0 bytes (31%) free.
[2/4] Linking CXX executable pedestrian_detect.elf
FAILED: pedestrian_detect.elf
cmd.exe /C "cd . && C:\Users\Simone.Bernabe\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=C:/Users/Simone.Bernabe/workspace/esp32/esp-dl/examples/pedestrian_detect/build/pedestrian_detect.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.wdt.ld -T esp32s3.rom.version.ld -T esp32s3.rom.newlib.ld -T memory.ld -T sections.ld @CMakeFiles\pedestrian_detect.elf.rsp -o pedestrian_detect.elf  && cd ."
C:/Users/Simone.Bernabe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/esp-dl/libesp-dl.a(dl_model_base.cpp.obj):(.literal._ZN2dl6module16RequantizeLinear11deserializeEPN3fbs8FbsModelENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN2dl6module16RequantizeLinear11deserializeEPN3fbs8FbsModelENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x8): undefined reference to `fbs::FbsModel::get_operation_attribute(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, dl::quant_type_t&)'
C:/Users/Simone.Bernabe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/esp-dl/libesp-dl.a(dl_model_base.cpp.obj):(.literal._ZN2dl5ModelD2Ev+0x0): undefined reference to `fbs::FbsModel::~FbsModel()'
C:/Users/Simone.Bernabe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/esp-dl/libesp-dl.a(dl_model_base.cpp.obj):(.literal._ZN2dl6module7Reshape11deserializeEPN3fbs8FbsModelENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN2dl6module7Reshape11deserializeEPN3fbs8FbsModelENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x0): undefined reference to `fbs::FbsModel::get_operation_parameter(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, bool, unsigned long)'
C:/Users/Simone.Bernabe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/esp-dl/libesp-dl.a(dl_model_base.cpp.obj):(.literal._ZN2dl6module7Flatten11deserializeEPN3fbs8FbsModelENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN2dl6module7Flatten11deserializeEPN3fbs8FbsModelENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x4): undefined reference to `fbs::FbsModel::get_operation_attribute(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int&)'   
C:/Users/Simone.Bernabe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/esp-dl/libesp-dl.a(dl_model_base.cpp.obj):(.literal._ZN2dl6module4Gemm11deserializeEPN3fbs8FbsModelENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN2dl6module4Gemm11deserializeEPN3fbs8FbsModelENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x18): undefined reference to `fbs::FbsModel::get_operation_attribute(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, dl::activation_type_t&)'
C:/Users/Simone.Bernabe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/esp-dl/libesp-dl.a(dl_model_base.cpp.obj):(.literal._ZN2dl6module6Concat11deserializeEPN3fbs8FbsModelENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN2dl6module6Concat11deserializeEPN3fbs8FbsModelENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x0): undefined reference to `fbs::FbsModel::get_operation_output_shape(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, std::vector<int, std::allocator<int> >&)'
C:/Users/Simone.Bernabe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/esp-dl/libesp-dl.a(dl_model_base.cpp.obj):(.literal._ZN2dl6module8Resize2D11deserializeEPN3fbs8FbsModelENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN2dl6module8Resize2D11deserializeEPN3fbs8FbsModelENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0xc): undefined reference to `fbs::FbsModel::get_operation_attribute(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, dl::resize_mode_t&)'
C:/Users/Simone.Bernabe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/esp-dl/libesp-dl.a(dl_model_base.cpp.obj):(.literal._ZN2dl6module9Transpose11deserializeEPN3fbs8FbsModelENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN2dl6module9Transpose11deserializeEPN3fbs8FbsModelENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x4): undefined reference to `fbs::FbsModel::get_operation_attribute(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::vector<int, std::allocator<int> >&)'
C:/Users/Simone.Bernabe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/esp-dl/libesp-dl.a(dl_model_base.cpp.obj):(.literal._ZN2dl5Model4loadEPN3fbs8FbsModelE+0x24): undefined reference to `fbs::FbsModel::load_map()'
C:/Users/Simone.Bernabe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/esp-dl/libesp-dl.a(dl_model_base.cpp.obj):(.literal._ZN2dl5Model4loadEPN3fbs8FbsModelE+0x28): undefined reference to `fbs::FbsModel::get_model_name[abi:cxx11]()'  
C:/Users/Simone.Bernabe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/esp-dl/libesp-dl.a(dl_model_base.cpp.obj):(.literal._ZN2dl5Model4loadEPN3fbs8FbsModelE+0x2c): undefined reference to `fbs::FbsModel::get_model_version()'
C:/Users/Simone.Bernabe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/esp-dl/libesp-dl.a(dl_model_base.cpp.obj):(.literal._ZN2dl5Model4loadEPN3fbs8FbsModelE+0x30): undefined reference to `fbs::FbsModel::get_model_doc_string[abi:cxx11]()'
[... cut ...] sic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, dl::quant_type_t&)'
[... 100 lines cut ...]
3. Compile Pedestrian or mobilenet example for ESP32-S3

### Project release version

5.3.0

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

PowerShell

### Additional context

Using IDF Visual Code extension

asadm, 2024-10-04:
hey, did you find an answer for this? I am also stuck on this part.

sun-xiangyu, 2024-10-08:
Hi, @simoberny  @asadm , thanks for your feedback. we have fixed it, please try again.

simoberny, 2024-10-08:
> Hi, @simoberny @asadm , thanks for your feedback. we have fixed it, please try again.

It works flawlessly,
Thanks 

Best regards

closed 2024-10-08 by simoberny