espressif/esp-dl #174: Missing libfbs_model.a for esp32s3 target

asadm, 2024-10-04:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Program compiles and links

### Actual behavior (suspected bug)

Linking fail

### Error logs or terminal output

```shell
[358/360] Linking CXX executable pedestrian_detect.elf
FAILED: pedestrian_detect.elf 
: && /Users/asad/esp/v5.3/esp-idf/tools/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++ -mlongcalls  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=/Users/asad/Downloads/esp-dl-master/examples/pedestrian_detect/build/pedestrian_detect.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.wdt.ld -T esp32s3.rom.version.ld -T esp32s3.rom.newlib.ld -T memory.ld -T sections.ld 

...

(.text._ZN2dl6memory19MemoryManagerGreedy24get_tensor_info_from_fbsEPN3fbs8FbsModelESt6vectorIPNS_6module6ModuleESaIS8_EERS5_IPNS0_10TensorInfoESaISC_EE+0xd66): undefined reference to `_ZN3fbs8FbsModel20get_value_info_dtypeENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE'
/Users/asad/esp/v5.3/esp-idf/tools/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/esp-dl/libesp-dl.a(dl_memory_manager_greedy.cpp.obj): in function `_ZNSt11char_traitsIcE4copyEPcPKcj':
/Users/asad/esp/v5.3/esp-idf/tools/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/char_traits.h:368:(.text._ZN2dl6memory19MemoryManagerGreedy24get_tensor_info_from_fbsEPN3fbs8FbsModelESt6vectorIPNS_6module6ModuleESaIS8_EERS5_IPNS0_10TensorInfoESaISC_EE+0xdce): undefined reference to `_ZN3fbs8FbsModel23get_value_info_exponentENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```
```

### Steps to reproduce the behavior

1. git clone esp-dl
2. open `examples/pedestrian_detect` in VSCode with ESP-IDF installed
3. set Device Target to esp32s3
4. Build

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

15.0

### Shell

ZSH

100312dog, 2024-10-08:
@asadm Thank you for your bug report. We've fixed this just now.

asadm, 2024-10-08:
works great, amazing! thank you all.

closed 2024-10-14 by sun-xiangyu