espressif/esp-dl #175: When will face recognition be available on the ESP32-P4?

mootterry, 2024-10-05:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

When will face recognition be available on the ESP32-P4, like it was on the ESP32-S3?

### Use cases

none

100312dog, 2024-10-08:
@mootterry We're working on this. It should be available at the end of this month. We'll first update the basic interface for face recognition in esp-dl, then we'll update the full example in esp-who.

sun-xiangyu, 2024-10-21:
@100312dog , now  face recognition is available on esp32p4, please try.

LeandroLuquetta, 2024-10-21:
Hello, can you provide a link to the esp32p4 face recognition?

sun-xiangyu, 2024-10-22:
example: https://github.com/espressif/esp-dl/tree/master/examples/human_face_recognition
model: https://github.com/espressif/esp-dl/tree/master/models/human_face_recognition

closed 2024-11-01 by sun-xiangyu