espressif/esp-dl #176: The example "human_face_detect" crashes.

TDA-2030, 2024-10-21:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

After running the example `human_face_detect`, memory is properly released.

### Actual behavior (suspected bug)

The program crashed and is continuously restarting indefinitely.

### Error logs or terminal output

```shell
ESP-ROM:esp32p4-eco1-20240205
Build:Feb  5 2024
rst:0x1 (POWERON),boot:0x30f (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x177c
load:0x4ff2abd0,len:0xf24
load:0x4ff2cbd0,len:0x33ec
entry 0x4ff2abda
I (25) boot: ESP-IDF v5.4-dev-3602-ga97a7b0962 2nd stage bootloader
I (26) boot: compile time Oct 20 2024 20:34:48
I (26) boot: Multicore bootloader
I (29) boot: chip revision: v0.1
I (30) boot: efuse block revision: v0.0
I (34) qio_mode: Enabling default flash chip QIO
I (38) boot.esp32p4: SPI Speed      : 80MHz
I (42) boot.esp32p4: SPI Mode       : QIO
I (46) boot.esp32p4: SPI Flash Size : 16MB
I (50) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 007d0000
I (84) boot: End of partition table
I (86) esp_image: segment 0: paddr=00010020 vaddr=480c0020 size=4ddf8h (318968) map
I (143) esp_image: segment 1: paddr=0005de20 vaddr=30100000 size=0002ch (    44) load
I (145) esp_image: segment 2: paddr=0005de54 vaddr=3010002c size=0003ch (    60) load
I (149) esp_image: segment 3: paddr=0005de98 vaddr=4ff00000 size=02180h (  8576) load
I (158) esp_image: segment 4: paddr=00060020 vaddr=48000020 size=bbac0h (768704) map
I (281) esp_image: segment 5: paddr=0011bae8 vaddr=4ff02180 size=10ae0h ( 68320) load
I (295) esp_image: segment 6: paddr=0012c5d0 vaddr=4ff12c80 size=02edch ( 11996) load
I (302) boot: Loaded app from partition at offset 0x10000
I (302) boot: Disabling RNG early entropy source...
I (312) hex_psram: vendor id    : 0x0d (AP)
I (313) hex_psram: Latency      : 0x01 (Fixed)
I (313) hex_psram: DriveStr.    : 0x00 (25 Ohm)
I (313) hex_psram: dev id       : 0x03 (generation 4)
I (318) hex_psram: density      : 0x07 (256 Mbit)
I (323) hex_psram: good-die     : 0x06 (Pass)
I (327) hex_psram: SRF          : 0x02 (Slow Refresh)
I (332) hex_psram: BurstType    : 0x00 ( Wrap)
I (336) hex_psram: BurstLen     : 0x03 (2048 Byte)
I (340) hex_psram: BitMode      : 0x01 (X16 Mode)
I (345) hex_psram: Readlatency  : 0x04 (14 cycles@Fixed)
I (350) hex_psram: DriveStrength: 0x00 (1/1)
I (354) MSPI DQS: tuning success, best phase id is 2
I (540) MSPI DQS: tuning success, best delayline id is 12
I esp_psram: Found 32MB PSRAM device
I esp_psram: Speed: 200MHz
[... 550 lines cut ...]
```

### Steps to reproduce the behavior

build example `human_face_detect` with Linux, then flash to esp32_p4_function_ev_board

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 20.04.6 LTS x86_64

### Shell

ZSH

100312dog, 2024-10-21:
@TDA-2030 Thank you for your feedback. This problem is related to the ppa(Pixel-Processing Accelerator) driver. 
Two ways to solve this.
1. shutdown ppa, use software crop and resize.
[https://github.com/espressif/esp-dl/blob/master/esp-dl/vision/image/dl_image_preprocessor.hpp#L58](url)
2. modify esp-idf code manually.
[https://github.com/espressif/esp-idf/blob/master/components/esp_driver_ppa/src/ppa_srm.c#L246](url)
modify this line to
uint32_t out_ext_window_len = config->out.pic_w * new_block_h * out_pixel_depth / 8;

100312dog, 2024-10-22:
@TDA-2030 Oh, that's true. I'll fix this.

closed 2024-11-11 by sun-xiangyu