espressif/esp-dl #177: Max Pool does not work

davidchongg, 2024-10-28:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

The model should run

### Actual behavior (suspected bug)

The model failed with this error:

### Error logs or terminal output

```shell
I (413) dl::Model: model:main_graph, version:0

I (423) dl::Model: /conv/conv1/Conv: Conv
I (423) dl::Model: /conv/conv2/Conv: Conv
I (433) dl::Model: /conv/pool1/MaxPool: MaxPool
E (433) dl::Model: Do not support MaxPool, please implement and register it first.
```

### Steps to reproduce the behavior

1. Copy example code for deploying model on ESP32 using espdl
2. Replace models with own pretrained pytorch model 

### Project release version

v1.4.0 latest version

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11 23H2

### Shell

PowerShell

sun-xiangyu, 2024-10-28:
Currently we do not support the MaxPool operator. If you are interested, you can refer to the [documentation](https://github.com/espressif/esp-dl/blob/master/tutorial/how_to_add_a_new_module(operator)_en.md) to add it.

sun-xiangyu, 2024-10-31:
@davidchongg , please update esp-dl and try again.

closed 2024-11-06 by davidchongg