espressif/esp-dl #178: Will face recognition on the ESP32-P4 support liveness detection

mootterry, 2024-11-01:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

Will face recognition on the ESP32-P4 support liveness detection?  To what extent will the liveness detection be effective, for instance, against photos or videos

### Use cases

face recognition

sun-xiangyu, 2024-11-01:
There are currently no plans to support liveness detection.
Do you have any recommended solutions? If the method and model size are suitable, we can consider adding it to esp-dl.

ganfeng45, 2024-12-18:
> There are currently no plans to support liveness detection. Do you have any recommended solutions? If the method and model size are suitable, we can consider adding it to esp-dl.

https://github.com/minivision-ai/Silent-Face-Anti-Spoofing

ganfeng45, 2025-01-15:
> 目前还没有计划支持活体检测。 您有什么推荐的解决方案吗？如果方法和模型大小合适，我们可以考虑将其添加到 esp-dl 中。

https://github.com/minivision-ai/Silent-Face-Anti-Spoofing.git