espressif/esp-dl #182: Can input tensor be converted directly from frame get from camera on ESP32-S3-EYE?

thorfinn0330, 2024-12-02:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Hi,
In example Mobilenet_v2, input is get from model file, I want to implement inference by input converted from RGB565 frame get from camera on ESP32-S3-EYE, I have done the conversion to correct shape with test input but got error
**dl::Model: The size of user_inputs(2) don't equal with the size of model inputs(1).**

Below is the implementation and results when checking the format of the test input and the input I converted.
```c++
TensorBase convertAndResizeRGB565ToTensorBase(camera_fb_t* frame, int width, int height) {
    // Validate input
    if (!frame || !frame->buf) {
        printf("Invalid frame buffer");
    }

    // Original dimensions
    int src_width = (int)frame->width;
    int src_height = (int)frame->height;
    int dst_width = width;
    int dst_height = height;

    // Allocate memory for converted and resized RGB888 data
    std::vector<uint8_t> rgb888_data(dst_width * dst_height * 3);

    // Precompute scaling factors
    float x_ratio = static_cast<float>(src_width - 1) / dst_width;
    float y_ratio = static_cast<float>(src_height - 1) / dst_height;

    // Convert and resize
    const uint16_t* src_ptr = reinterpret_cast<const uint16_t*>(frame->buf);
    uint8_t* dst_ptr = rgb888_data.data();

    for (int dy = 0; dy < dst_height; dy++) {
        for (int dx = 0; dx < dst_width; dx++) {
            // Calculate source pixel coordinates
            float src_x = dx * x_ratio;
            float src_y = dy * y_ratio;

            // Find surrounding source pixels
            int x0 = static_cast<int>(src_x);
            int y0 = static_cast<int>(src_y);
            int x1 = std::min(x0 + 1, src_width - 1);
            int y1 = std::min(y0 + 1, src_height - 1);

            // Interpolation weights
            float wx = src_x - x0;
            float wy = src_y - y0;

            // Bilinear interpolation for each color channel
            for (int channel = 0; channel < 3; channel++) {
                // Get surrounding pixel values
                uint16_t p00 = src_ptr[y0 * src_width + x0];
                uint16_t p10 = src_ptr[y0 * src_width + x1];
                uint16_t p01 = src_ptr[y1 * src_width + x0];
                uint16_t p11 = src_ptr[y1 * src_width + x1];

                // Extract color components
                uint8_t c00, c10, c01, c11;
                switch (channel) {
                    case 0: // Red
                        c00 = (p00 >> 11) & 0x1F;
                        c10 = (p10 >> 11) & 0x1F;
                        c01 = (p01 >> 11) & 0x1F;
                        c11 = (p11 >> 11) & 0x1F;
                        break;
                    case 1: // Green
                        c00 = (p00 >> 5) & 0x3F;
                        c10 = (p10 >> 5) & 0x3F;
                        c01 = (p01 >> 5) & 0x3F;
                        c11 = (p11 >> 5) & 0x3F;
                        break;
                    case 2: // Blue
                        c00 = p00 & 0x1F;
                        c10 = p10 & 0x1F;
                        c01 = p01 & 0x1F;
                        c11 = p11 & 0x1F;
                        break;
                }

                // Interpolate
                float interpolated = 
                    (c00 * (1 - wx) * (1 - wy)) +
                    (c10 * wx * (1 - wy)) +
                    (c01 * (1 - wx) * wy) +
                    (c11 * wx * wy);

                // Scale to 8-bit and store
                uint8_t scaled_value;
                switch (channel) {
                    case 0: // Red
                        scaled_value = (static_cast<uint8_t>(interpolated) << 3) | (static_cast<uint8_t>(interpolated) >> 2);
                        break;
                    case 1: // Green
                        scaled_value = (static_cast<uint8_t>(interpolated) << 2) | (static_cast<uint8_t>(interpolated) >> 4);
                        break;
                    case 2: // Blue
                        scaled_value = (static_cast<uint8_t>(interpolated) << 3) | (static_cast<uint8_t>(interpolated) >> 2);
                        break;
                }

                dst_ptr[(dy * dst_width + dx) * 3 + channel] = scaled_value;
            }
        }
    }

    // Create shape vector
    std::vector<int> shape = {1, dst_height, dst_width, 3};  // NHWC format

    // Create TensorBase object
    TensorBase tensor(
        shape,           // Shape
        rgb888_data.data(), // Data pointer
        -5,               // Exponent (default)
        DATA_TYPE_INT8, // Data type (8-bit unsigned integers)
        true,            // Deep copy
        MALLOC_CAP_8BIT  // Memory capabilities
    );

    return tensor;
}
``` 
In app_main

```c++
 int ret = camera_init(20000000, PIXFORMAT_RGB565, SENSOR_FRAME_SIZE, 1);
Model *model = new Model("model", fbs::MODEL_LOCATION_IN_FLASH_PARTITION);
    std::map<std::string, TensorBase *> graph_test_inputs = get_graph_test_inputs(model);
    TensorBase input_tensor = convertAndResizeRGB565ToTensorBase(frame1, 224, 224);
    graph_test_inputs["input_2"] = &input_tensor;
``` 
[image]

Thank you in advance for your help.
Best regards

100312dog, 2024-12-04:
@thorfinn0330 An update is coming soon. It should be available by the end of this week. I've refactored the dl_image, It provides easy way to transform the camera image to a model input. And an example of mobilenet inference with an image is also added.

100312dog, 2024-12-06:
@thorfinn0330 
Hi, you can refer to this example
https://github.com/espressif/esp-dl/tree/master/examples/imagenet_cls

if you want to transform an image to a model input, just create a ImagePreprocessor instance like this
https://github.com/espressif/esp-dl/blob/master/models/imagenet_cls/imagenet_cls.cpp#L21
and call its member function preprocess.
https://github.com/espressif/esp-dl/blob/master/esp-dl/vision/image/dl_image_preprocessor.hpp#L57

The error you encountered is because you try to create a new input besides the test input. Then you have two inputs, while the model mobilenetv2 only accepts one input.

We have three api to run the model
https://github.com/espressif/esp-dl/blob/master/esp-dl/dl/model/include/dl_model_base.hpp#L173
```
virtual void run(runtime_mode_t mode = RUNTIME_MODE_SINGLE_CORE);
virtual void run(TensorBase *input, runtime_mode_t mode = RUNTIME_MODE_SINGLE_CORE);
virtual void run(std::map<std::string, TensorBase *> &user_inputs,
                 runtime_mode_t mode = RUNTIME_MODE_SINGLE_CORE,
                 std::map<std::string, TensorBase *> user_outputs = {});
```

We recommend to use the fisrt one. The memory of the model input is allocated when it is created. So if you use the two apis below, a memory copy is required.
You can directly put the data to the the allocated model input, then run the first api.

In the code you provided, i don't quite understand this part.
```
                // Scale to 8-bit and store
                uint8_t scaled_value;
                switch (channel) {
                    case 0: // Red
                        scaled_value = (static_cast<uint8_t>(interpolated) << 3) | (static_cast<uint8_t>(interpolated) >> 2);
                        break;
                    case 1: // Green
                        scaled_value = (static_cast<uint8_t>(interpolated) << 2) | (static_cast<uint8_t>(interpolated) >> 4);
                        break;
                    case 2: // Blue
                        scaled_value = (static_cast<uint8_t>(interpolated) << 3) | (static_cast<uint8_t>(interpolated) >> 2);
                        break;
                }
```

In a usual preprocess pipeline, after resize, you may need to normalize the data and do the quantization.

thorfinn0330, 2024-12-10:
thanks for replying that was really helpful, when changing from **graph_test_inputs["input_2"] = &input_tensor;** to **graph_test_inputs["input.1"] = &input_tensor;** it worked as input to the model. however the example you provided is quite hardcoded and I can't integrate it into my current project.

I change the mobilenetv2 model to a self-defined model to enhance the dark image with the same input and output model shape after quantization as follows:

```
graph main_graph (
%input.1[INT8, 1x240x240x3], exponents: [-6]
) initializers (
%/Constant_output_0[INT8, 1]
%conv2.weight[INT8, 3x3x8x8]
%conv1.bias[INT32, 8]
%conv1.weight[INT8, 3x3x3x8]
%conv3.bias[INT32, 3]
%conv2.bias[INT32, 8]
%conv3.weight[INT8, 3x3x8x3]
%/Constant_1_output_0[INT8, 1]
) { 
%/relu/Relu_output_0 = Conv[activation = 'Relu', dilations = [1, 1], group = 1, kernel_shape = [3, 3], pads = [1, 1, 1, 1], quant_type = 'S8', strides = [1, 1]](%input.1, %conv1.weight, %conv1.bias)
 %/relu_1/Relu_output_0 = Conv[activation = 'Relu', dilations = [1, 1], group = 1, kernel_shape = [3, 3], pads = [1, 1, 1, 1], quant_type = 'S8', strides = [1, 1]](%/relu/Relu_output_0, %conv2.weight, %conv2.bias)
 %/conv3/Conv_output_0 = Conv[activation = 'Linear', dilations = [1, 1], group = 1, kernel_shape = [3, 3], pads = [1, 1, 1, 1], quant_type = 'S8', strides = [1, 1]](%/relu_1/Relu_output_0, %conv3.weight, %conv3.bias)
 %16 = Clip[quant_type = 'S8'](%/conv3/Conv_output_0, %/Constant_output_0, %/Constant_1_output_0)
 return %16
}
values ​​info:
%/conv3/Conv_output_0[INT8, 1x240x240x3], exponents: [-7]
%/relu/Relu_output_0[INT8, 1x240x240x8], exponents: [-5]
%16[INT8, 1x240x240x3], exponents: [-7]
%/relu_1/Relu_output_0[INT8, 1x240x240x8], exponents: [-4]
%input.1[INT8, 1x240x240x3], exponents: [-6]
``` 

I don't know how to put the input image into the espdl file in the test inputs value

The input is captured from the camera with PIXFORMAT_RGB565 and FRAMESIZE_240X240 (equal to the input model so no need to resize for now, I only care about converting it to TensorBase that can be used for the input model)
```c++
TensorBase convertRGB565ToTensorBase(camera_fb_t* frame, int8_t exponent) {
// Validate input
if (!frame || !frame->buf) {
 printf("Invalid frame buffer");
 }

 // Calculate dimensions
 int width = (int)frame->width;
 int height = (int)frame->height;
 size_t pixel_count = width * height;

 // Allocate memory for converting RGB888 data
 std::vector<uint8_t> rgb888_data(pixel_count * 3);

 // Convert RGB565 to RGB888
 const uint16_t* rgb565_ptr = reinterpret_cast<const uint16_t*>(frame->buf);
 uint8_t* rgb888_ptr = rgb888_data.data();

 for (size_t i = 0; i < pixel_count; ++i) {
 uint16_t pixel = rgb565_ptr[i];

 // Extract RGB components from RGB565
 uint8_t r = (pixel >> 11) & 0x1F;
 uint8_t g = (pixel >> 5) & 0x3F;
 uint8_t b = pixel & 0x1F;

 // Scale to 8-bit range
 float rf = static_cast<float>(r) / 31.0f; //* 255.0f;
 float gf = static_cast<float>(g) / 63.0f;// * 255.0f;
 float bf = static_cast<float>(b) / 31.0f;// * 255.0f;

 // Normalize and quantize to INT8 range
 rgb888_ptr[i * 3 + 0] = static_cast<int8_t>(std::round(rf / std::pow(2, exponent)));
 rgb888_ptr[i * 3 + 1] = static_cast<int8_t>(std::round(gf / std::pow(2, exponent)));
 rgb888_ptr[i * 3 + 2] = static_cast<int8_t>(std::round(bf / std::pow(2, exponent)));
[... 97 lines cut ...]
``` 
in main:
```c++
    Model *model = new Model("model", fbs::MODEL_LOCATION_IN_FLASH_PARTITION);
    std::map<std::string, TensorBase *> graph_test_inputs = get_graph_test_inputs(model);
    std::map<std::string, TensorBase *> graph_test_outputs;
    TensorBase input_tensor = convertRGB565ToTensorBase(frame1, -6);
    graph_test_inputs["input.1"] = &input_tensor;

    model->run(graph_test_inputs);
        // Safely access a tensor
    if (graph_test_inputs.count("input.1") > 0) {
        TensorBase input_tensor = deepCopyTensorBase((model->get_outputs())["16"]);
        camera_fb_t* frame2 = tensorBaseToRGB565FrameBuffer(input_tensor);
}
``` 

If there is any error in the way I convert, normalize from RGB565 to TensorBase and vice versa, please let me know how to fix it
Here is the result after inference by the model I defined and the conversion steps above:
[image]

[image]

[image]
[image]

100312dog, 2024-12-12:
@thorfinn0330 
The first thing you have to confirm is the byte order of your camera output.
If you're working with the official ESP32-S3-EYE camera. The output is in little endian order.
https://github.com/espressif/esp-dl/blob/master/esp-dl/vision/image/dl_image_define.hpp#L15
```
// gggbbbbb rrrrrggg
#define DL_IMAGE_LITTLE_ENDIAN_RGB565_BIT1(x) ((uint8_t)(((x) & 0x1F00) >> 5))
#define DL_IMAGE_LITTLE_ENDIAN_RGB565_BIT2(x) ((uint8_t)((((x) & 0x7) << 5) | (((x) & 0xE000) >> 11)))
#define DL_IMAGE_LITTLE_ENDIAN_RGB565_BIT3(x) ((uint8_t)((x) & 0xF8))
```
You can try to save the converted rgb888 image to see whether the conversion is right or not.

The second thing you have to confirm is the image process pipeline of your custom model. You have to make sure you're using the same image process pipeline during inference and training.
When training the model, is the input bgr or rgb?  Is normalization in the process pipeline? 
The values ​​{123.675, 116.28, 103.53}, {58.395, 57.12, 57.375} are mean, std of each channel of the ImageNet Dataset. Not a single image, but the statistics of the entire dataset.

In esp-dl, quant before feeding the input to model, and dequant after getting the output of the model is required.

closed 2024-12-30 by sun-xiangyu