espressif/esp-dl #183: Stuck at Calibration Progress in the quantize_onnx_model.py example

morenzoe, 2024-12-04:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Calibration Progress (Phase 1) percentage going up

### Actual behavior (suspected bug)

The calibration progress is terminated and ESP-PPQ restart (the logo shows up again in terminal) as a new process

### Error logs or terminal output

```shell
PS D:\moren\SNN\master\esp-dl> & d:/moren/SNN/master/esp-dl/.venv/Scripts/python.exe d:/moren/SNN/master/esp-dl/tools/quantization/quantize_onnx_model.py

    ___________ ____        ____  ____  ____
   / ____/ ___// __ \      / __ \/ __ \/ __ \
  / __/  \__ \/ /_/ /_____/ /_/ / /_/ / / / /
 / /___ ___/ / ____/_____/ ____/ ____/ /_/ /
/_____//____/_/         /_/   /_/    \___\_\

load imagenet calibration dataset from directory: ./imagenet\calib
[17:19:10] PPQ Quantization Fusion Pass Running ...       Finished.
[17:19:10] PPQ Quantize Simplify Pass Running ...         Finished.
[17:19:10] PPQ Parameter Quantization Pass Running ...    Finished.
Calibration Progress(Phase 1):   0%|                                                | 0/32 [00:00<?, ?it/s]
    ___________ ____        ____  ____  ____
   / ____/ ___// __ \      / __ \/ __ \/ __ \
  / __/  \__ \/ /_/ /_____/ /_/ / /_/ / / / /
 / /___ ___/ / ____/_____/ ____/ ____/ /_/ /
/_____//____/_/         /_/   /_/    \___\_\

load imagenet calibration dataset from directory: ./imagenet\calib
[17:19:17] PPQ Quantization Fusion Pass Running ...       Finished.
[17:19:17] PPQ Quantize Simplify Pass Running ...         Finished.
[17:19:17] PPQ Parameter Quantization Pass Running ...    Finished.
Calibration Progress(Phase 1):   0%|                                                | 0/32 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Python310\lib\multiprocessing\spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "C:\Python310\lib\multiprocessing\spawn.py", line 125, in _main
    prepare(preparation_data)
  File "C:\Python310\lib\multiprocessing\spawn.py", line 236, in prepare
    _fixup_main_from_path(data['init_main_from_path'])
  File "C:\Python310\lib\multiprocessing\spawn.py", line 287, in _fixup_main_from_path
    main_content = runpy.run_path(main_path,
  File "C:\Python310\lib\runpy.py", line 289, in run_path
    return _run_module_code(code, init_globals, run_name,
  File "C:\Python310\lib\runpy.py", line 96, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "C:\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "d:\moren\SNN\master\esp-dl\tools\quantization\quantize_onnx_model.py", line 123, in <module>        
    quant_ppq_graph = espdl_quantize_onnx(
  File "D:\moren\SNN\master\esp-dl\.venv\lib\site-packages\ppq\core\defs.py", line 54, in _wrapper
    return func(*args, **kwargs)
  File "D:\moren\SNN\master\esp-dl\.venv\lib\site-packages\ppq\api\espdl_interface.py", line 200, in espdl_quantize_onnx
    ppq_graph = quantize_onnx_model(
  File "D:\moren\SNN\master\esp-dl\.venv\lib\site-packages\ppq\core\defs.py", line 54, in _wrapper
    return func(*args, **kwargs)
  File "D:\moren\SNN\master\esp-dl\.venv\lib\site-packages\ppq\api\interface.py", line 263, in quantize_onnx_model
    quantizer.quantize(
  File "D:\moren\SNN\master\esp-dl\.venv\lib\site-packages\ppq\core\defs.py", line 54, in _wrapper
    return func(*args, **kwargs)
  File "D:\moren\SNN\master\esp-dl\.venv\lib\site-packages\ppq\quantization\quantizer\base.py", line 69, in 
quantize
    quant_pipeline.optimize(
  File "D:\moren\SNN\master\esp-dl\.venv\lib\site-packages\ppq\quantization\optim\base.py", line 81, in optimize
    optim_pass.optimize(graph=graph, **kwargs)
  File "D:\moren\SNN\master\esp-dl\.venv\lib\site-packages\ppq\core\defs.py", line 54, in _wrapper
    return func(*args, **kwargs)
  File "D:\moren\SNN\master\esp-dl\.venv\lib\site-packages\ppq\quantization\optim\calibration.py", line 176, in optimize
    self.calibrate(desc='Calibration Progress(Phase 1)', dataloader=dataloader,
  File "D:\moren\SNN\master\esp-dl\.venv\lib\site-packages\ppq\quantization\optim\calibration.py", line 114, in calibrate
    for data in dataloader:
  File "D:\moren\SNN\master\esp-dl\.venv\lib\site-packages\torch\utils\data\dataloader.py", line 484, in __iter__
    return self._get_iterator()
  File "D:\moren\SNN\master\esp-dl\.venv\lib\site-packages\torch\utils\data\dataloader.py", line 415, in _get_iterator
    return _MultiProcessingDataLoaderIter(self)
  File "D:\moren\SNN\master\esp-dl\.venv\lib\site-packages\torch\utils\data\dataloader.py", line 1138, in __init__
    w.start()
  File "C:\Python310\lib\multiprocessing\process.py", line 121, in start
    self._popen = self._Popen(self)
  File "C:\Python310\lib\multiprocessing\context.py", line 224, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "C:\Python310\lib\multiprocessing\context.py", line 336, in _Popen
    return Popen(process_obj)
  File "C:\Python310\lib\multiprocessing\popen_spawn_win32.py", line 45, in __init__
    prep_data = spawn.get_preparation_data(process_obj._name)
  File "C:\Python310\lib\multiprocessing\spawn.py", line 154, in get_preparation_data
    _check_not_importing_main()
  File "C:\Python310\lib\multiprocessing\spawn.py", line 134, in _check_not_importing_main
    raise RuntimeError('''
RuntimeError:
        An attempt has been made to start a new process before the
        current process has finished its bootstrapping phase.

        This probably means that you are not using fork to start your
        child processes and you have forgotten to use the proper idiom
        in the main module:

            if __name__ == '__main__':
                freeze_support()
                ...

        The "freeze_support()" line can be omitted if the program
        is not going to be frozen to produce an executable.
```

### Steps to reproduce the behavior

1. Clone esp-dl github
2. Make venv in VS Code using Python 3.10.6
3. Install esp-ppq, onnxruntime, pandas, and torchvision
4. Comment out os and subprocess code (Line 76-78. It keep making another error in VS Code, so I just did it manually)
5. Download imagenet_calib.zip from the [imagenet_url](https://dl.espressif.com/public/imagenet_calib.zip), unzip it in the CALIB_DIR
6. Download mobilenetv2-10.onnx from [model_url](https://github.com/onnx/models/blob/main/validated/vision/classification/mobilenet/model/mobilenetv2-10.onnx), rename and move it according to ONNX_PATH
7. Change quant_setting_mobilenet_v2 input 1 to None (line 119)
8. Select port to COM5 (actually connected to ESP32), select device target esp32p4 via built-in USB-JTAG
9. Run the python file

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 10

### Shell

PowerShell

### Additional context

Hi, I am pretty much a newbie with ESP-DL and ESP-IDF, but I have been using ESP32, ESP8266, and ESP32-CAM with Arduino IDE for several years. I am using VS Code with ESP-IDF extension v1.6.5 and ESP-IDF v5.3.1. I was able to build, flash, and monitor the blink example with ESP-IDF. However, quantize_onnx_model.py didn't work and quantize_torch_model.py also didn't work with similar error as stated above. I don't think the connected device matters because this is still just the model quantization step, not model deployment (changing the target to ESP32 WROVER-KIT 3.3V doesn't solve the error anyway). Any help or pointer would be much appreciated!

In the end, I would like to quantize and deploy my own PyTorch model, that has been converted to onnx, to ESP32. So I would like to also know how to use esp-dl for the ESP32 DEVKIT DOIT board. If possible, what are the model limitations? If not possible, what kind of ESP32 dev board should I use? Thank you in advance!

100312dog, 2024-12-05:
@morenzoe Python multiprocessing module has some restrictions on windows. I fixed the script, feel free to try it.
However, linux is strongly recomended. 
ESP32 DEVKIT DOIT board has no support for (PIE) Processor Instruction Extensions, so all ops run in c implementation. C implementation currently is not fully align with esp-ppq, which means there will be some errors between the pc simulation result and real deployed model.
esp32s3 and esp32p4 has PIE Instructions, model runs faster, and all supported ops are tested. 
you can find the supported ops here.
https://github.com/espressif/esp-dl/tree/master/esp-dl/dl/module/include
In addition to checking whether the model's ops are supported, the size of the model should also be considered. Now model stores in flash, so the model file should fit in the flash size. And FlOPs of model must be controlled. Model under 1FLOPs may work well on esp32s3 and esp32p4.

morenzoe, 2024-12-11:
Hi @100312dog, thank you very much for your quick answer and fix! The code works now with the mobilenet_v2 example.

When quantizing my own model, I got several output that I can't find the explanation anywhere.

```
Analysing Layerwise quantization error:: 100%|███████████████████████████████████████████████████████████████| 
Layer           | NOISE:SIGNAL POWER RATIO
/fc2/MatMul:    | ████████████████████ | 2.621%
/scale2/MatMul: | ██████████           | 1.251%
/fc1/MatMul:    |                      | 0.000%
/fc0/MatMul:    |                      | 0.000%
[ERROR][ESPDL][2024-12-11 17:43:47]:  Can not reset MatMul:/fc0/MatMul layout
[ERROR][ESPDL][2024-12-11 17:43:47]:  Can not reset MatMul:/fc1/MatMul layout
[ERROR][ESPDL][2024-12-11 17:43:47]:  Can not reset MatMul:/fc2/MatMul layout
[ERROR][ESPDL][2024-12-11 17:43:47]:  Can not reset MatMul:/scale2/MatMul layout
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[INFO][ESPDL][2024-12-11 17:43:47]:  skip not QuantableOperation
[WARNING][ESPDL][2024-12-11 17:43:47]:  /Gather_output_0 does not bind exponents parameter
[WARNING][ESPDL][2024-12-11 17:43:47]:  input does not bind exponents parameter
[WARNING][ESPDL][2024-12-11 17:43:47]:  /lif2/Mul_output_0 does not bind exponents parameter
[WARNING][ESPDL][2024-12-11 17:43:47]:  /lif0/Mul_output_0 does not bind exponents parameter
[WARNING][ESPDL][2024-12-11 17:43:47]:  /lif0/Clip_output_0 does not bind exponents parameter
[WARNING][ESPDL][2024-12-11 17:43:47]:  /lif1/Mul_output_0 does not bind exponents parameter
[WARNING][ESPDL][2024-12-11 17:43:47]:  /lif1/Greater_output_0 does not bind exponents parameter
[WARNING][ESPDL][2024-12-11 17:43:47]:  /lif1/Cast_output_0 does not bind exponents parameter     
[WARNING][ESPDL][2024-12-11 17:43:47]:  /lif0/Mul_1_output_0 does not bind exponents parameter    
[WARNING][ESPDL][2024-12-11 17:43:47]:  PPQ_Variable_22 does not bind exponents parameter        
[WARNING][ESPDL][2024-12-11 17:43:47]:  PPQ_Variable_25 does not bind exponents parameter
```

Can you help me understand the "Can not reset ... layout" error, "skip not QuantableOperation" info, and "... does not bind exponents parameter" warning? Thanks again

BlueSkyB, 2024-12-12:
Currently, matmul is not yet supported and is being added.
"skip not QuantableOperation" indicates that there are non-quantized operators in the model.

morenzoe, 2024-12-13:
Hi @BlueSkyB , thank you for your answer. I would like to follow up with several more questions if you don't mind. Is there any way to substitute matmul into simpler operations that are supported with the esp quantization? How do I know which non-quantized operator is skipped in the process? I do still get the .espdl file at the end of the program, can I still deploy that file to esp32s3 board? Thank you in advance

BlueSkyB, 2024-12-13:
The latest ESP-PPQ and ESP-DL already support MatMul; you can update them all.
For the quantization operators supported by ESP-PPQ, you can check the `quant_operation_types` function in the source code file "esp-ppq/ppq/quantization/quantizer/EspdlQuantizer.py".
However, the limitations of the operators are actually more about whether they are supported in the ESP-DL inference framework, which can be checked in the directory "esp_dl/esp-dl/dl/module/include" within ESP-DL.
We will likely have a page displaying the currently supported operators in the future, but for now, it can only be done by reviewing the source code.

BlueSkyB, 2024-12-13:
Additionally, even if you have an .espdl file, if the file contains operators not supported by ESP-DL, exceptions will be thrown during deployment.
If you want to view the model structure within the .espdl file, you can check the generated .info file. If this file is not produced, you can set the export_config parameter to True when you call the espdl_quantize_torch or espdl_quantize_onnx interface during the model quantization process to generate it.

morenzoe, 2024-12-18:
Hi @BlueSkyB thank you for the update. Now I can quantized my own model without error from MatMul. 

Recently I have acquired an esp32s3 devkit board to deploy my own model. I use the .\examples\mobilenet_v2 folder but change the .espdl, .info, .json, and .onnx file in the .\examples\mobilenet_v2\models folder. I have also renamed the CONFIG_MODEL_FILE_PATH="models/esp32s3/mobilenet_v2.espdl" in sdkconfig.defaults.esp32s3 file to my own model .espdl filename. However, the output from the board is still the same layers from mobilenet_v2 (Conv and Clip layers, which is not present in my own model at all). I have tried deleting everything inside the .\examples\mobilenet_v2\models folder but it can still build and flash and output the same layers info. I have checked the .info file and it has the correct layers of my own model. For every changes I made, I always clean the build first before deploying again. I have also tried making a new project in different directory.

So my question now is how do I declare explicitly which .espdl model file I want to deploy to the esp32s3 board? Thank you again for your help!

morenzoe, 2024-12-18:
Hi @BlueSkyB thank you for the update. Now I can quantized my own model without error from MatMul. 

Recently I have acquired an esp32s3 devkit board to deploy my own model. I use the ".\examples\mobilenet_v2" folder but change the .espdl, .info, .json, and .onnx file in the ".\examples\mobilenet_v2\models" folder. I have also renamed the `CONFIG_MODEL_FILE_PATH="models/esp32s3/mobilenet_v2.espdl"` in ".\examples\mobilenet_v2\sdkconfig.defaults.esp32s3" file to my own model .espdl filename. However, the output from the board is still the same layers from mobilenet_v2 (Conv and Clip layers, which is not present in my own model at all). I have tried deleting everything inside the .\examples\mobilenet_v2\models folder but it can still build and flash and output the same layers info. I have checked the .info file and it has the correct layers of my own model. For every changes I made, I always clean the build first before deploying again. I have also tried making a new project in different directory.

So my question now is how do I declare explicitly which .espdl model file I want to deploy to the esp32s3 board? Thank you again for your help!

100312dog, 2024-12-18:
[image]
The name of this option dispalyed may be a little confusing.
The configuration showing in menuconfig is controlled by Kconfig/Kconfig.projbuild file.
https://github.com/espressif/esp-dl/blob/master/examples/mobilenet_v2/main/Kconfig.projbuild
The mobilenetv2 example is designed to show you how to quantize a model, and use the test inputs and outputs to ensure the model inference result is correct.
The other examples is some models we want users to use it out of box without any modification, so we don't provide any configuration in them, just use fixed model path.
You can compare the cmakelists.txt below to see the difference.
https://github.com/espressif/esp-dl/blob/master/examples/mobilenet_v2/main/CMakeLists.txt
https://github.com/espressif/esp-dl/blob/master/models/human_face_detect/CMakeLists.txt

morenzoe, 2024-12-18:
Thank youu @100312dog! I didn't have the Kconfig/Kconfig.projbuild file in my new directory, is there any way to generate that file?

Now I am able to deploy my own model. However, there is a layer that is not supported by esp-dl, which is `%/Gather_output_0 = Gather[axis = 0, quant_type = 'F32'](%input, %/Constant_output_0)`. So, I am going to work on that now. Again, thank you very much @100312dog and @BlueSkyB!

100312dog, 2024-12-18:
@morenzoe 
The Kconfig is written by yourself. And Its only effect is defining some macros. 
If you want to use it to control the model path. You must use the macro to control logic in cmakelists.txt.
The cmakelists.txt is the key to determine the model path.
If you use flash rodata to store model, you must modify the embed_files in cmakelist.txt.
If you use flash partition to store model, you must modify the path in esptool_py_flash_to_partition()
https://github.com/espressif/esp-dl/blob/master/tutorial/how_to_load_model_en.md
pls refer to this doc

morenzoe, 2024-12-18:
Thank you very much for the answer @100312dog ! Looking forward to the gather operation support update. I am closing this issue for now :D

closed 2024-12-18 by morenzoe