espressif/esp-dl #186: face_recognizer模型输入rgb565 数据不能正常工作

ganfeng45, 2024-12-18:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

对于rgb565可以正常推理

### Actual behavior (suspected bug)

不工作

### Steps to reproduce the behavior

example中的示例
修改get_image函数 输出rgb565格式
运行
face_recognizer->recognize((uint16_t *)bill1, {300, 300, 3});
提示没有人脸

### Project release version

最新

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

win10

### Shell

ZSH

100312dog, 2024-12-18:
@ganfeng45 
please refer to 
https://github.com/espressif/esp-dl/blob/master/examples/human_face_recognition/main/app_main.cpp
```
dl::image::img_t img = {.data = (void *) your_rgb565_data,
                                      .width = 1000,
                                      .height = 1000,
                                      .pix_type = dl::image::DL_IMAGE_PIX_TYPE_RGB565
                                      };
face_recognizer->recognize(img, detect_result);
```

ganfeng45, 2024-12-18:
感谢您的回复，但是目前好像还是不能正常工作
这是我的测试代码
#include "esp_log.h"
#include "human_face_detect.hpp"
#include "human_face_recognition.hpp"

extern const uint8_t bill1_jpg_start[] asm("_binary_bill1_jpg_start");
extern const uint8_t bill1_jpg_end[] asm("_binary_bill1_jpg_end");
extern const uint8_t bill2_jpg_start[] asm("_binary_bill2_jpg_start");
extern const uint8_t bill2_jpg_end[] asm("_binary_bill2_jpg_end");
extern const uint8_t musk1_jpg_start[] asm("_binary_musk1_jpg_start");
extern const uint8_t musk1_jpg_end[] asm("_binary_musk1_jpg_end");
extern const uint8_t musk2_jpg_start[] asm("_binary_musk2_jpg_start");
extern const uint8_t musk2_jpg_end[] asm("_binary_musk2_jpg_end");
const char *TAG = "human_face_recognition";

extern "C" void app_main(void)
{
    dl::image::jpeg_img_t bill1_jpeg = {.data = (uint8_t *)bill1_jpg_start,
                                        .width = 300,
                                        .height = 300,
                                        .data_size = (uint32_t)(bill1_jpg_end - bill1_jpg_start)};
    dl::image::img_t bill1;
    bill1.pix_type = dl::image::DL_IMAGE_PIX_TYPE_RGB565;
    sw_decode_jpeg(bill1_jpeg, bill1);

   /*  dl::image::jpeg_img_t bill2_jpeg = {.data = (uint8_t *)bill2_jpg_start,
                                        .width = 300,
                                        .height = 300,
                                        .data_size = (uint32_t)(bill2_jpg_end - bill2_jpg_start)};
    dl::image::img_t bill2;
    bill2.pix_type = dl::image::DL_IMAGE_PIX_TYPE_RGB888;
    sw_decode_jpeg(bill2_jpeg, bill2);

    dl::image::jpeg_img_t musk1_jpeg = {.data = (uint8_t *)musk1_jpg_start,
                                        .width = 300,
                                        .height = 300,
                                        .data_size = (uint32_t)(musk1_jpg_end - musk1_jpg_start)};
    dl::image::img_t musk1;
    musk1.pix_type = dl::image::DL_IMAGE_PIX_TYPE_RGB888;
    sw_decode_jpeg(musk1_jpeg, musk1);

    dl::image::jpeg_img_t musk2_jpeg = {.data = (uint8_t *)musk2_jpg_start,
                                        .width = 300,
                                        .height = 300,
                                        .data_size = (uint32_t)(musk2_jpg_end - musk2_jpg_start)};
    dl::image::img_t musk2;
    musk2.pix_type = dl::image::DL_IMAGE_PIX_TYPE_RGB888;
    sw_decode_jpeg(musk2_jpeg, musk2); */

    auto human_face_detect = new HumanFaceDetect();
    auto human_face_recognizer = new HumanFaceRecognizer();
    human_face_recognizer->enroll(bill1, human_face_detect->run(bill1));
    // human_face_recognizer->enroll(bill2, human_face_detect->run(bill2));
    // human_face_recognizer->enroll(musk1, human_face_detect->run(musk1));

    // auto res = human_face_recognizer->recognize(musk2, human_face_detect->run(musk2));
    // for (const auto &k : res) {
    //     printf("id: %d, sim: %f\n", k.id, k.similarity);
    // }

    human_face_recognizer->clear_all_feats();
    delete human_face_recognizer;
    delete human_face_detect;
    heap_caps_free(bill1.data);
    /* heap_caps_free(bill2.data);
    heap_caps_free(musk1.data);
    heap_caps_free(musk2.data); */
}
这是对应的日志输出
 (1786) dl::Model: Conv_102: Conv
I (1826) MemoryManagerGreedy: Maximum mermory size: 840448

W (1896) HumanFaceRecognizer: Failed to enroll. No face detected.
I (2496) main_task: Returned from app_main()

ganfeng45, 2024-12-18:
感谢，解决了

closed 2024-12-18 by ganfeng45