espressif/esp-dl #187: Unable to export to espdl format

NeumoNeumo, 2024-12-24:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I expect the ONNX file to be exported into the espdl format.

### Actual behavior (suspected bug)

Instead, `espdl_quantize_onnx` successfully quantized the model but failed to export it to espdl format

### Error logs or terminal output

```shell
!  cse12111103@ailab01  ◰eye  ~/.x/c/t/test  python test.py        (eye) 1.8m  15:35:56

    ___________ ____        ____  ____  ____ 
   / ____/ ___// __ \      / __ \/ __ \/ __ \
  / __/  \__ \/ /_/ /_____/ /_/ / /_/ / / / /
 / /___ ___/ / ____/_____/ ____/ ____/ /_/ / 
/_____//____/_/         /_/   /_/    \___\_\ 

load random calibration dataset
[15:37:42] PPQ Quantization Fusion Pass Running ...       Finished.
[15:37:42] PPQ Quantize Simplify Pass Running ...         Finished.
[15:37:42] PPQ Parameter Quantization Pass Running ...    Finished.
Calibration Progress(Phase 1): 100%|███████████████████████████| 8/8 [00:07<00:00,  1.10it/s]
Calibration Progress(Phase 2): 100%|███████████████████████████| 8/8 [00:10<00:00,  1.29s/it]
Finished.
[15:38:02] PPQ Quantization Alignment Pass Running ...    Finished.
[15:38:02] PPQ Passive Parameter Quantization Running ... Finished.
--------- Network Snapshot ---------
Num of Op:                    [106]
Num of Quantized Op:          [103]
Num of Variable:              [229]
Num of Quantized Var:         [223]
------- Quantization Snapshot ------
Num of Quant Config:          [327]
ACTIVATED:                    [115]
OVERLAPPED:                   [138]
PASSIVE:                      [61]
FP32:                         [13]
Network Quantization Finished.
Analysing Graphwise Quantization Error(Phrase 1):: 100%|███████| 1/1 [00:00<00:00,  1.11it/s]
Analysing Graphwise Quantization Error(Phrase 2):: 100%|███████| 1/1 [00:02<00:00,  2.83s/it]
Layer    | NOISE:SIGNAL POWER RATIO 
Conv_13: | ████████████████████ | 184.897%
Conv_16: | ███████████████████  | 171.792%
Conv_10: | ████████████████     | 149.795%
Conv_19: | ████████████████     | 145.506%
Conv_3:  | █████████████        | 125.246%
Conv_65: | █████████████        | 118.245%
Conv_69: | ███████████          | 104.715%
Conv_20: | ███████████          | 104.423%
Conv_7:  | ███████████          | 100.389%
Conv_23: | █████████            | 83.162%
Conv_52: | █████████            | 82.598%
Conv_73: | █████████            | 82.564%
Conv_66: | ████████             | 75.942%
Conv_15: | ███████              | 70.589%
Conv_12: | ███████              | 69.057%
Conv_55: | ███████              | 64.928%
Conv_67: | ███████              | 62.850%
[... 256 lines cut ...]
3. run `python test.py` and you'll get the error log.

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 20.04.6 LTS (Focal Fossa)

### Shell

Fish

### Additional context

I successfully quantize `tools/quantization/models/torch/mobilenet_v2_relu.onnx` and export it to espdl using the `test.py`.

BlueSkyB, 2024-12-25:
Check the log; it indicates that the attribute type of an operator is not supported. Currently, esp-ppq supports the ONNX opset version v13. If you are using PyTorch's `torch.onnx.export` interface, you need to set the parameter `opset_version=13`. I suspect that the opset version of the ONNX file you exported is not compatible.

NeumoNeumo, 2025-01-04:
> Check the log; it indicates that the attribute type of an operator is not supported. Currently, esp-ppq supports the ONNX opset version v13. If you are using PyTorch's `torch.onnx.export` interface, [...]

I use this code to check the opset_version of the model.
```python
import onnx

# Load the ONNX model
model = onnx.load("fumo_n_640_640.onnx")

# Get the opset version
opset_version = model.opset_import[0].version
print(f"Opset version: {opset_version}")
```
The output is
```bash
Opset version: 11
```
ONNX guarantees backward compatibility according to [this page](https://onnxruntime.ai/docs/reference/compatibility.html). So the opset_version=11 should be OK.

BlueSkyB, 2025-01-06:
@NeumoNeumo  Could you please provide us with the ONNX model file that you quantized so we can debug the issue?

NeumoNeumo, 2025-01-06:
> @NeumoNeumo Could you please provide us with the ONNX model file that you quantized so we can debug the issue?

Actually, if you `ctrl+f` the keyword "model.zip" in this page, you should find a match at the top of this issue. Unzip the file and you'll get the ONNX model file named "fumo_n_640_640.onnx".

Copy-and-paste here for your convenience.  [model.zip](https://github.com/user-attachments/files/18237247/model.zip)

BlueSkyB, 2025-01-13:
Sorry, I overlooked that the model link has already been provided. We will debug the cause of the issue in the near future.

BlueSkyB, 2025-01-17:
@NeumoNeumo 
I've checked and found that the error occurs because the model contains three ConstantOfShape operators, whose attributes include types that are not supported. I see that these three ConstantOfShape operators are glue operators and have no practical use. Can you remove these three operators before exporting to ONNX? Because even if the issue with esp-ppq is fixed, esp-dl currently still does not support the ConstantOfShape operator.
As for the issue of esp-ppq exporting the ConstantOfShape operator abnormally, we will fix it, and esp-dl will also gradually cover the ConstantOfShape operator in the future.

BlueSkyB, 2025-01-26:
The exception caused by the operator's attributes has been resolved. Please also update esp-ppq and esp-dl.
The ConstantOfShape operator has not been added yet. However, in this particular issue, the ConstantOfShape operator in the model is invalid. It can be removed in advance using the onnx.utils.extract_model API.
The issue has been resolved.

closed 2025-01-26 by BlueSkyB