espressif/esp-dl #188: Do not support MatMul, please implement and register it first. (ESP-IDF Monitor )

JimLin0406, 2025-01-01:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

###  **IDF version.**
ESP-IDF v5.3.1 

### **Operating System used.**
Windows

### **How did you build your project?**
VS Code IDE

### **ESP32-SOC**
ESP32-S3R2

### **Descroption of encountered issue**
I originally planned to use the `espdl_quantize_torch` tool provided by ESP-DL to convert my trained PyTorch model into an ESP-DL model. However, I encountered numerous errors, including unsupported operators such as MatMul, AveragePool, and Transpose. As a result, I decided to troubleshoot each operator individually. Even when using a model with only the MatMul operator, I still encountered warnings about unsupported operators.

### **Expected behavior**
According to the [Operator Support State](https://github.com/espressif/esp-dl/blob/master/operator_support_state.md), the MatMul operator should be supported.

### **Error logs or terminal output**

```
I (268) main_task: Calling app_main()
I (268) MOBILENET_V2_EXAMPLE: get into app_main
I (278) MOBILENET_V2_EXAMPLE: Load feature model.....
I (278) FbsLoader: The storage free size is 30016 KB
I (288) FbsLoader: The partition size is 4000 KB
I (288) dl::Model: model:main_graph, version:0

I (298) dl::Model: /MatMul: MatMul
E (298) dl::Model: Do not support MatMul, please implement and register it first.
I (308) MOBILENET_V2_EXAMPLE: exit app_main
I (318) main_task: Returned from app_main()
```

### **Steps to Generate an ESP-DL Model**

```
C:\Users\WTMH\anaconda3\envs\onnx_py310_cpu\python.exe C:\Users\WTMH\Desktop\ESP-dl_onnx\main.py 

    ___________ ____        ____  ____  ____ 
   / ____/ ___// __ \      / __ \/ __ \/ __ \
  / __/  \__ \/ /_/ /_____/ /_/ / /_/ / / / /
 / /___ ___/ / ____/_____/ ____/ ____/ /_/ / 
/_____//____/_/         /_/   /_/    \___\_\ 

load random calibration dataset
[15:47:54] PPQ Quantization Fusion Pass Running ...       Finished.
[15:47:54] PPQ Quantize Simplify Pass Running ...         Finished.
[15:47:54] PPQ Parameter Quantization Pass Running ...    Finished.
[15:47:54] PPQ Runtime Calibration Pass Running ...       Finished.
Calibration Progress(Phase 1): 100%|██████████| 32/32 [00:00<00:00, 6421.59it/s]
[15:47:54] PPQ Quantization Alignment Pass Running ...    Finished.
[15:47:54] PPQ Passive Parameter Quantization Running ... Finished.
--------- Network Snapshot ---------
Num of Op:                    [1]
Num of Quantized Op:          [0]
Num of Variable:              [3]
Num of Quantized Var:         [0]
------- Quantization Snapshot ------
Num of Quant Config:          [0]
Network Quantization Finished.
Oops. you got nothing to analyse.
Oops. you got nothing to analyse.
[INFO][ESPDL][2025-01-01 15:47:54]:  skip not QuantableOperation
[WARNING][ESPDL][2025-01-01 15:47:54]:  output_gram does not bind exponents parameter
[WARNING][ESPDL][2025-01-01 15:47:54]:  input does not bind exponents parameter

Process finished with exit code 0
```

### **Additional issuse and content**

When using the ESP-DL conversion tool provided by ESP-DL, the warning `[WARNING][ESPDL][2025-01-01 15:47:54]: output_gram does not bind exponents parameter appears.` Does this indicate that the model cannot be converted, or can I safely ignore this warning?

sun-xiangyu, 2025-01-02:
Usually we can ignore this warning, the latest release v3.0 has support for MatMul, please try again.

JimLin0406, 2025-01-03:
Hi @sun-xiangyu, thanks for your reply. 
I tried removing `espdl v^3.0.0-rc.1` and installing `espdl v^3.0.0`, but the bug still persists.
I’ve check the idf_component.yml file, which indicates that the version of esp-dl being used is the latest.
```
## IDF Component Manager Manifest File
dependencies:
  espressif/esp-dl:
    version: "^3.0.0"
```
What parts do you think I should check for potential issues?
Thanks...

sun-xiangyu, 2025-01-03:
1, please make sure the MatMul has been registered: https://github.com/espressif/esp-dl/blob/0aa80ee4bba1ccd086f05bf501d1e50300bcb767/esp-dl/dl/module/include/dl_module_creator.hpp#L133
2. If you code is the the latest, please attach the model structure or .info file generated by the esp-ppq.

JimLin0406, 2025-01-03:
Hi @sun-xiangyu , followed by step 1, I check ` this->register_module("MatMul", MatMul::deserialize); ` is existing in my library, and it can be register sucessfully.

Here show my model structure (file generated by the esp-ppq):
```
graph main_graph (
  %input[FLOAT, 1x1x70x16], exponents: [0]
) initializers (
  %weight[FLOAT, 16x16]
) {
  %output_1 = MatMul[quant_type = 'F32'](%input, %weight)
  return %output_1
}

initializers value:
%weight, shape: [16, 16], exponents: [0], docString: layout ==> UNK, 
value: array([ 1.77986294e-01, -5.04638255e-01, -1.26055074e+00, -7.19045699e-01,
       -1.54329503e+00,  4.82517034e-02, -1.82153273e+00, -8.24052766e-02,
       -1.26030087e+00, -8.02605033e-01,  9.74059522e-01, -8.01942110e-01,
        8.26574638e-02,  7.74493933e-01, -4.81240004e-01, -2.74584472e-01,
       -2.88804603e+00, -2.54903615e-01,  1.07202828e+00,  7.55520165e-01,
       -2.26695085e+00, -1.02612484e+00,  2.41951275e+00,  2.88007796e-01,
        9.00623322e-01,  2.12164849e-01,  8.67453754e-01, -7.52089143e-01,
       -5.52865207e-01, -1.15949055e-02, -2.99839735e-01,  1.15165961e+00,
       -2.31438205e-01,  9.95340496e-02, -3.66042733e-01,  1.29580617e+00,
        6.99246645e-01,  1.11428738e+00, -5.32040238e-01, -7.49482751e-01,
       -1.11145914e+00, -1.87695324e+00, -7.39436924e-01,  1.72408178e-01,
        3.29371601e-01, -1.47264671e+00, -4.51147482e-02, -1.14229393e+00,
       -2.35117257e-01,  6.93222046e-01, -2.75364339e-01,  2.06477955e-01,
        7.97045767e-01, -4.33177263e-01,  3.87673438e-01,  2.39728308e+00,
       -1.76295424e+00,  9.12599683e-01,  9.72974598e-02, -1.47031999e+00,
        6.51563883e-01, -1.89064804e-03,  1.05887222e+00, -1.84918976e+00,
       -3.70621979e-01, -2.72348560e-02,  1.27256489e+00,  1.11228272e-01,
       -6.69979155e-01, -2.45173782e-01,  3.99963111e-01, -2.14491200e+00,
        4.53696489e-01, -8.19913268e-01, -1.19400692e+00, -1.22852361e+00,
        1.49104238e+00,  1.36343107e-01, -1.33302140e+00, -7.17977643e-01,
        3.58378887e-01, -8.23128879e-01, -1.40532970e+00,  9.05162513e-01,
       -1.27748661e-02,  2.34812036e-01, -1.44767702e+00,  6.74140096e-01,
        4.21245426e-01,  7.41991341e-01,  1.84674487e-01, -2.07005405e+00,
       -5.81103206e-01,  3.22411805e-01, -1.61026597e+00,  4.03050274e-01,
       -1.66807520e+00,  5.47350585e-01,  3.84826027e-02,  1.28340185e-01,
       -3.49296033e-01, -4.20106016e-02,  7.11596787e-01, -9.28181887e-01,
        4.81022149e-01, -6.79755151e-01, -1.73774898e+00,  2.06229353e+00,
       -6.00241959e-01, -2.05161881e+00, -3.98163527e-01,  1.41760182e+00,
       -2.43463516e-01, -1.23872511e-01, -2.13790014e-01,  2.19093347e+00,
       -2.94377178e-01,  7.88495898e-01, -1.01550269e+00, -1.03504336e+00,
        1.07539065e-01,  6.59962237e-01, -7.53275633e-01,  3.54138881e-01,
        6.27757430e-01,  1.34784949e+00, -6.85059309e-01,  1.83133736e-01,
        1.24997401e+00,  1.49437177e+00,  2.17294753e-01,  1.24444342e+00,
        2.58553386e-01,  5.57168722e-01,  1.57214427e+00, -6.51938617e-01,
        1.36581886e+00,  2.13974983e-01, -1.35544956e+00, -1.03702998e+00,
        5.51088154e-01, -1.42583644e+00,  1.28857100e+00, -9.84142661e-01,
       -1.48701680e+00, -7.07573712e-01, -9.95103568e-02,  1.28821373e+00,
        1.05300200e+00, -1.18875372e+00,  2.42371225e+00, -7.50279486e-01,
        1.62014067e+00,  2.16767812e+00, -1.57358909e+00,  1.55470335e+00,
        1.49580848e+00,  1.25984204e+00,  4.91510779e-01,  3.38736683e-01,
       -1.97497272e+00,  6.53145850e-01,  5.63487530e-01,  1.24312949e+00,
        7.71490633e-01,  1.64067477e-01, -3.18461239e-01,  1.14621326e-01,
        1.66842246e+00, -7.28038967e-01,  7.13071674e-02, -1.06996536e+00,
       -1.13153851e+00,  3.77819300e-01, -2.03376937e+00,  1.07297480e+00,
        9.29480195e-01, -6.04998171e-01, -5.36426425e-01, -1.57597393e-01,
        9.00002241e-01, -5.14593720e-01,  3.83004427e-01, -5.06395042e-01,
       -6.36868298e-01,  1.88657209e-01,  1.74443591e+00,  2.45143577e-01,
        1.95441842e+00,  1.34499118e-01,  3.37360725e-02, -8.99382055e-01,
       -4.39009309e-01,  4.16882634e-01, -1.67281735e+00, -1.33353746e+00,
       -1.65491962e+00,  4.48154300e-01, -8.27507794e-01,  1.11321342e+00,
       -4.33879763e-01, -2.37403542e-01, -1.79690135e+00, -1.06599212e-01,
        3.11821967e-01,  2.08485627e+00, -3.52461517e-01,  4.72599626e-01,
        6.04842901e-01,  8.24756742e-01, -1.20014834e+00, -4.55555320e-01,
       -1.21109021e+00,  1.80246174e-01,  6.54582024e-01,  4.91839707e-01,
        1.69945490e+00,  1.52496889e-01, -3.95976692e-01, -1.37135193e-01,
[... 558 lines cut ...]
        4.63170338e+00, -2.78938591e-01,  3.08476782e+00,  9.51015294e-01,
        4.46878999e-01,  2.14533493e-01, -4.95857894e-01,  4.38080043e-01,
       -3.21045607e-01,  2.00744823e-01, -1.24427712e+00, -3.18856239e+00,
        5.92270947e+00, -8.72532785e-01, -4.62458324e+00, -7.92324066e-01,
        1.10090864e+00,  2.30095601e+00, -5.60023928e+00, -3.57071590e+00,
        2.73678929e-01, -1.80422390e+00,  1.71367145e+00,  2.96440053e+00,
        1.85038865e+00, -2.29419917e-01,  7.23498702e-01,  4.25269336e-01,
        2.15718627e+00, -2.12667394e+00, -7.99902534e+00, -4.71432066e+00,
       -7.74218035e+00,  2.56692362e+00, -8.79756355e+00,  4.81187677e+00,
       -7.21370459e+00, -1.29993129e+00, -3.88048440e-01, -3.45062852e+00,
       -1.38981187e+00,  3.51744246e+00, -1.52434778e+00, -1.24741495e+00,
[... cut ...]
       -4.89974165e+00,  2.23120761e+00, -7.69842672e+00,  3.39755917e+00,
       -6.57956362e-01,  8.55622053e-01, -5.16608572e+00, -3.92353511e+00,
        2.46364641e+00,  1.17910471e+01, -2.94786263e+00,  1.14626861e+00,
       -5.73394012e+00,  3.43378758e+00,  3.07195634e-01,  1.01500368e+01,
       -2.46612430e+00,  5.35496140e+00, -2.91710281e+00,  2.22072572e-01,
        1.63887680e-01, -3.99512672e+00, -1.30261116e+01,  9.35228765e-01,
       -1.05056357e+00,  5.49318075e+00, -4.91392279e+00, -9.21217501e-01],
      dtype=float32)
```

sun-xiangyu, 2025-01-03:
` %output_1 = MatMul[quant_type = 'F32'](%input, %weight) `
The quant_type of your MatMul is 'F32'(float32). You need to quantize the model by int8 or int16

JimLin0406, 2025-01-03:
It works!
I think I need to first understand all the functionalities of esp-ppq. Then the next step will be to split the model and flash it into the ESP32.
Many thanks for your help!

closed 2025-01-03 by JimLin0406