espressif/esp-dl #190: Cannot create an esp-ppq installation where quantize_custom_torch_model.py works properly.

nicklasb, 2025-01-05:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Installing ESP-PPQ following the "pip install git+https://github.com/espressif/esp-ppq.git" instructions makes it possible to quantize my model using quantize_custom_torch_model.py (albeit providing my own model instead of mobilenet etc.). 

### Actual behavior (suspected bug)

ESP-PPQ depends on older versions of torch that ends up conflicting with torchvision. 

### Error logs or terminal output

```shell
You can end up in several situations, but this seems most relevant, as old torch (2.2.2) is forced:

  File "...\Projects\esp-dl\tools\quantization\quantize_custom_torch_model.py", line 6, in <module>
    import torchvision
  File "...\Projects\esp-dl\.venv\Lib\site-packages\torchvision\__init__.py", line 10, in <module>
    from torchvision import _meta_registrations, datasets, io, models, ops, transforms, utils  # usort:skip
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "...\Projects\esp-dl\.venv\Lib\site-packages\torchvision\_meta_registrations.py", line 163, in <module>
    @torch.library.register_fake("torchvision::nms")
     ^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'torch.library' has no attribute 'register_fake'
```

### Steps to reproduce the behavior

* Follow the esp-ppq installation on esp-dl
* Quantize a model.

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

PowerShell

sun-xiangyu, 2025-01-06:
Please [install torch>=2.4.0](https://github.com/huggingface/autotrain-advanced/issues/734) first and then reinstall esp-ppq.

nicklasb, 2025-01-12:
Thank you, but I am afraid that esp-ppq causes torch 2.2.2 to reinstall and then it fails again.
However, upgrading Torch and then leave esp-ppq untouched as is, works. 
I suppose this must imply that threre is a problem with dependencies?

sun-xiangyu, 2025-01-13:
Hi @nicklasb,
Thanks for your feedback, which version is your python? I will check it.

nicklasb, 2025-03-14:
Sorry for the delay, this was with 3.11. I have since made it work by *really* triangulating the exact versions. 
It would be great if things could be made a little bit more flexible.