espressif/esp-dl #191: 使用人脸检测的时候提示 fatal error: climits: No such file or directory

pangziandapple, 2025-01-07:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

正常编译通过

### Actual behavior (suspected bug)

编译时出错，提示找不到climits

### Error logs or terminal output

```shell
In file included from D:/human-face/managed_components/espressif__esp-dl/vision/detect/dl_detect_define.hpp:3,
                 from D:/human-face/managed_components/espressif__esp-dl/vision/detect/dl_detect_postprocessor.hpp:2,
                 from D:/human-face/managed_components/espressif__esp-dl/vision/detect/dl_detect_base.hpp:3,
                 from D:/human-face/managed_components/espressif__human_face_detect/human_face_detect.hpp:3,
                 from D:/human-face/main/app_main.c:15:
D:/human-face/managed_components/espressif__esp-dl/dl/dl_define.hpp:5:10: fatal error: climits: No such file or directory
    5 | #include <climits>
```

### Steps to reproduce the behavior

运行esp-dl中的demo，human_face_detect，在编译的时候提示找不到climits

### Project release version

3.0.0~1-rc.2

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows11

### Shell

ZSH

BlueSkyB, 2025-01-07:
climits 是C++标准库的头文件，可以检查下您的IDF编译环境，看编译工具链是否正确安装配置。

pangziandapple, 2025-01-08:
正常可以找到，只是在编译的时候过不去

BlueSkyB, 2025-01-13:
目前看climits头文件，并没有被使用，是代码遗留下来的。您可以去除该头文件的include操作，代码路径：esp-dl/dl/dl_define.hpp。去除之后也是正常的，我们后面也会先去除"#include <climits>".
如果后面还出现其它标准头文件找不到，还是建议再检查下编译环境，若环境没问题，欢迎再给我们提issue。

pangziandapple, 2025-01-15:
可能是我对这个编译环境不太熟悉，我在同事的电脑和我自己的三个电脑上都遇到这个问题，都是通过vscode进行创建项目的。建议配置尽量简单点，特别是对我们刚开始入门的人员不太友好。如果有需要特别配置的，建议在文档中标注出来。

closed 2025-01-16 by pangziandapple