espressif/esp-dl #195: Cmake Error from Example-Apps

israr34907, 2025-01-31:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

So I have been trying to use the example models provided in this repository on our device, Having **ESP-IDF version : V5.4** , and Device : **Esp32 DevBoard (s3)** . 

When I tried to tun mobilenet-v2 example using idf.py build, we got some issues regarding project's Cmake. Here are the detailed Logs:
```
Executing action: all (aliases: build)
Running cmake in directory /home/naveed/esp-dl/examples/human_face_detect/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/naveed/.espressif/python_env/idf5.4_py3.10_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/naveed/esp-dl/examples/human_face_detect"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/naveed/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/naveed/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/naveed/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
NOTICE: Dependencies lock doesn't exist, solving dependencies.
NOTICE: Skipping optional dependency: espressif/esp32_p4_function_ev_board_noglib
NOTICE: Skipping optional dependency: espressif/esp32_s3_eye_noglib
NOTICE: Skipping optional dependency: espressif/esp32_p4_function_ev_board_noglib
NOTICE: Skipping optional dependency: espressif/esp32_s3_eye_noglib
NOTICE: Skipping optional dependency: espressif/esp32_p4_function_ev_board_noglib
NOTICE: Skipping optional dependency: espressif/esp32_s3_eye_noglib
NOTICE: Using component placed at /home/naveed/esp-dl/models/human_face_detect for dependency "espressif/human_face_detect", specified in /home/naveed/esp-dl/examples/human_face_detect/main/idf_component.yml
.NOTICE: Skipping optional dependency: espressif/esp32_p4_function_ev_board_noglib
NOTICE: Skipping optional dependency: espressif/esp32_s3_eye_noglib
CMake Error at /home/naveed/esp/esp-idf/tools/cmake/build.cmake:610 (message):
  ERROR: Because project depends on espressif/esp-dl (^3.1.0) which doesn't
  match any versions, version solving failed.

Call Stack (most recent call first):
  /home/naveed/esp/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:7 (project)

-- Configuring incomplete, errors occurred!
See also "/home/naveed/esp-dl/examples/human_face_detect/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /home/naveed/esp-dl/examples/human_face_detect/build/log/idf_py_stderr_output_40225 and /home/naveed/esp-dl/examples/human_face_detect/build/log/idf_py_stdout_output_40225
```

Any suggestion about how to solve this issue will be appreciated !

nicklasb, 2025-03-06:
Please warn about this in the docs if you haven't added it. It just took a lot of time to realize this, all aren't versed in cmake internals, so this is kind of difficult to realize.

nicklasb, 2025-03-06:
@100312dog That didn't matter, I downloaded the zip, the make folder is not in the package.

closed 2025-06-06 by sun-xiangyu