espressif/esp-dl #197: Error and fail malloc with Picodet network in espdl

simoberny, 2025-02-11:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hi, I'm trying to use a custom-trained Picodet network for person detection with ESP-DL, but I'm encountering an error when loading the model. My goal is to obtain a network similar to the one used in the pedestrian detection example, as I was informed that it is a modified version of Picodet.
I trained a Picodet XS with some modifications, using ReLU6 instead of hardswish as the activation function, to avoid opset 14 in the onnx format. 

My setup version: 
- ESP-IDF 5.4.0
- ESP-DL 3.0.0

I successfully exported the .onnx model to .espdl using the provided tools. However, when I try to load the model using dl::Model, I encounter the following error:

```I (1368) dl::Model: model:PaddlePaddle Graph 0, version:0

I (1368) dl::Model: Conv.0: Conv
I (1368) dl::Model: Clip.0: Clip
I (1368) dl::Model: Conv.1: Conv
I (1368) dl::Model: Clip.2: Clip
I (1368) dl::Model: Conv.2: Conv
I (1378) dl::Model: Clip.4: Clip
I (1378) dl::Model: Conv.3: Conv
I (1378) dl::Model: Clip.6: Clip
I (1378) dl::Model: Conv.4: Conv
I (1388) dl::Model: Clip.8: Clip
I (1388) dl::Model: Conv.5: Conv
I (1388) dl::Model: Clip.10: Clip
I (1398) dl::Model: Conv.6: Conv
I (1398) dl::Model: Clip.12: Clip
I (1398) dl::Model: Conv.7: Conv
I (1408) dl::Model: Clip.14: Clip
I (1408) dl::Model: Conv.8: Conv
I (1408) dl::Model: Clip.16: Clip
I (1408) dl::Model: Conv.9: Conv
I (1418) dl::Model: Clip.18: Clip
I (1418) dl::Model: Conv.10: Conv
I (1418) dl::Model: Clip.20: Clip
I (1428) dl::Model: Conv.11: Conv
I (1428) dl::Model: Conv.31: Conv
I (1428) dl::Model: Clip.22: Clip
I (1438) dl::Model: Clip.54: Clip
I (1438) dl::Model: Conv.12: Conv
I (1438) dl::Model: Clip.24: Clip
I (1438) dl::Model: Conv.13: Conv
I (1448) dl::Model: Clip.26: Clip
I (1448) dl::Model: Conv.14: Conv
I (1448) dl::Model: Clip.28: Clip
I (1458) dl::Model: Conv.15: Conv
I (1458) dl::Model: Clip.30: Clip
I (1458) dl::Model: Conv.16: Conv
I (1468) dl::Model: Clip.32: Clip
I (1468) dl::Model: Conv.17: Conv
I (1468) dl::Model: Clip.34: Clip
I (1478) dl::Model: Conv.18: Conv
I (1478) dl::Model: Clip.36: Clip
I (1478) dl::Model: Conv.19: Conv
I (1478) dl::Model: Clip.38: Clip
I (1488) dl::Model: Conv.20: Conv
I (1488) dl::Model: Clip.40: Clip
I (1488) dl::Model: Conv.21: Conv
I (1498) dl::Model: Clip.42: Clip
I (1498) dl::Model: Conv.22: Conv
I (1498) dl::Model: Clip.44: Clip
I (1508) dl::Model: Conv.23: Conv
I (1508) dl::Model: Conv.32: Conv
I (1508) dl::Model: Clip.46: Clip
[... 196 lines cut ...]

Backtrace: 0x40379f8d:0x3fcaf8e0 0x403830ed:0x3fcaf900 0x4038ab8d:0x3fcaf920 0x420349a4:0x3fcafa40 0x42049ef3:0x3fcafa70 0x4204b046:0x3fcafc70 0x4203d62d:0x3fcafce0 0x4203dae7:0x3fcafd70 0x4200bbec:0x3fcafda0 0x42125e6b:0x3fcafe00
--- 0x40379f8d: panic_abort at /pathtoesp/v5.4/esp-idf/components/esp_system/panic.c:454
0x403830ed: esp_system_abort at /pathtoesp/v5.4/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x4038ab8d: __assert_func at /path/v5.4/esp-idf/components/newlib/assert.c:80
0x420349a4: dl::module::Concat::get_output_shape(std::vector<std::vector<int, std::allocator<int> >, std::allocator<std::vector<int, std::allocator<int> > > >&) at /project/managed_components/espressif__esp-dl/dl/module/include/dl_module_concat.hpp:58 (discriminator 1)
0x42049ef3: dl::memory::MemoryManagerGreedy::get_tensor_info_from_fbs(fbs::FbsModel*, std::vector<dl::module::Module*, std::allocator<dl::module::Module*> >, std::vector<dl::memory::TensorInfo*, std::allocator<dl::memory::TensorInfo*> >&) at /project/Ocio/managed_components/espressif__esp-dl/dl/model/src/dl_memory_manager_greedy.cpp:140
0x4204b046: dl::memory::MemoryManagerGreedy::alloc(fbs::FbsModel*, std::vector<dl::module::Module*, std::allocator<dl::module::Module*> >&) at /project/managed_components/espressif__esp-dl/dl/model/src/dl_memory_manager_greedy.cpp:41 (discriminator 1)
0x4203d62d: dl::Model::build(unsigned int, dl::memory_manager_t, bool) at /project/managed_components/espressif__esp-dl/dl/model/src/dl_model_base.cpp:171
0x4203dae7: dl::Model::Model(char const*, fbs::model_location_type_t, int, dl::memory_manager_t, unsigned char*, bool) at /project/managed_components/espressif__esp-dl/dl/model/src/dl_model_base.cpp:20
 (inlined by) dl::Model::Model(char const*, fbs::model_location_type_t, int, dl::memory_manager_t, unsigned char*, bool) at /project/managed_components/espressif__esp-dl/dl/model/src/dl_model_base.cpp:12
0x4200bbec: app_main at /project/main/main.cpp:222 (discriminator 1)
0x42125e6b: main_task at /pathtoesp/v5.4/esp-idf/components/freertos/app_startup.c:208
```

If I try to load the model using the Pico Model from the PedestrianDetection example instead, the system attempts to allocate 223,827,472 bytes, which inevitably causes a crash.

```
I (1378) dl::Model: Conv.0: Conv
W (1378) calloc_aligned: heap_caps_aligned_calloc failed, retry with MALLOC_CAP_8BIT
E (1388) calloc_aligned: Fail to malloc 223827472 bytes from DRAM(80235 bytyes) and PSRAM(4730728 bytes), PSRAM is on.
```

Thanks in advance in any help!
P.S: Is it possible to have some insight on the parameter and the procedure used to obtain the model from pedestrian example?

100312dog, 2025-02-12:
@simoberny 
The PedestrianDetection example in esp-dl should work. Update esp-dl to 3.1.0 and try. If the PedestrianDetection example works, export your own model with latest esp-ppq, and replace the model in the example with your own.
If the issue still exists, can you share your onnx file? So that we can determine why it happens and fix it.

simoberny, 2025-02-12:
> [@simoberny](https://github.com/simoberny) The PedestrianDetection example in esp-dl should work. Update esp-dl to 3.1.0 and try. If the PedestrianDetection example works, export your own model with [...]

Thank you for your prompt response.
I have updated PPQ to the latest commit and upgraded ESP-DL to version 3.1.0.

The pedestrian example runs perfectly, but when I use my model, I still encounter the same error as before:

```
assert failed: virtual std::vector<std::vector<int> > dl::module::Concat::get_output_shape(std::vector<std::vector<int> >&) dl_module_concat.hpp:58 (input_shapes[i][j] == input_shapes[i - 1][j]) 
```

Looking at the layers of the network seems really different from the pedestrian example tough. 
This is my onnx file: 
[picodet_xs_processed.zip](https://github.com/user-attachments/files/18765240/picodet_xs_processed.zip)

100312dog, 2025-02-14:
@simoberny Thank you for your info. I did a super quick test. I am able to run the exported model without the assert failure, although the model inference result is incorrect. We will try to reproduce your problem and dig deeper to find out why the inference result is incorrect.
What's your operating system? Our quantization tools seems to have some issues on windows.

simoberny, 2025-02-14:
@100312dog Yeah my bad, Windows was the problem, I was using a conda environment. 
I switched to a Linux machine, and now the exported model loads correctly.

Now it is panicking on inference call

simoberny, 2025-02-18:
I understand that the ESP-DL post-processing code is designed for a specific network configuration.

I tried replacing score0, bbox0 with the actual output tensors from my PicoDet XS model, which was trained on a custom dataset with a single "person" label:

```c++
    TensorBase *score0 = m_model->get_intermediate("save_infer_model/scale_0.tmp_0");
    TensorBase *bbox0 = m_model->get_intermediate("save_infer_model/scale_4.tmp_0");
    TensorBase *score1 = m_model->get_intermediate("save_infer_model/scale_1.tmp_0");
    TensorBase *bbox1 = m_model->get_intermediate("save_infer_model/scale_5.tmp_0");
    TensorBase *score2 = m_model->get_intermediate("save_infer_model/scale_2.tmp_0");
    TensorBase *bbox2 = m_model->get_intermediate("save_infer_model/scale_6.tmp_0");
```

I'm using three out of the four FPN scales (64, 32, 16), but the code gets stuck during parsing.

Could you help me understand if I am making a conceptual mistake? Or is it possibile to have the picodet configuration to replicate the pedestrian example network.

Thank you for your help

100312dog, 2025-02-19:
@simoberny 
We noticed that many customers have many problems when deploying their own network using esp-dl. 
So We have a plan to share one or two object detection architecture to make it easier. The tutorial will come in one or two months.
And i'll solve your offical picodetxs deploy problem ASAP.

closed 2025-05-12 by sun-xiangyu