espressif/esp-dl #198: Cant Recreate mobilenet_v2 onnx to espdl exmpale

danweil66, 2025-02-12:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hey all I tried to re-quantize the exact mobilenet_v2 given "https://github.com/espressif/esp-dl/tree/master/examples/mobilenet_v2/models/esp32s3" 
I run the  
https://github.com/espressif/esp-dl/blob/master/tools/quantization/quantize_onnx_model.py
script (all on master) 
The only change is :
target to esp32s3
and 
" quant_setting, ONNX_PATH = quant_setting_mobilenet_v2(
        ONNX_PATH, None
    )" 
. 
The scripts work but the yield a model that crush in run time (when I use the given espdl it works) 

[image]

Also, i see  weird numbers in the .info : 

[image]

I tried to update the ppq but API changes. 
Also my venv: 
aiohappyeyeballs==2.4.6
aiohttp==3.11.12
aiosignal==1.3.2
attrs==25.1.0
certifi==2025.1.31
cffi==1.17.1
charset-normalizer==3.4.1
colorama==0.4.6
coloredlogs==15.0.1
cryptography==44.0.0
datasets==3.2.0
dill==0.3.8
filelock==3.17.0
flatbuffers==25.1.24
frozenlist==1.5.0
fsspec==2024.9.0
huggingface-hub==0.28.1
humanfriendly==10.0
idna==3.10
Jinja2==3.1.5
markdown-it-py==3.0.0
MarkupSafe==3.0.2
mdurl==0.1.2
mpmath==1.3.0
multidict==6.1.0
multiprocess==0.70.16
networkx==3.4.2
numpy==1.26.4
onnx==1.17.0
onnxruntime==1.20.1
onnxsim==0.4.36
packaging==24.2
pandas==2.2.3
pillow==11.1.0
ppq @ git+https://github.com/espressif/esp-ppq.git@2a38cc8cc4ccc45d25256fa3ae3b077246031254
propcache==0.2.1
protobuf==3.20.2
pyarrow==19.0.0
pycparser==2.22
Pygments==2.19.1
pyreadline3==3.5.4
python-dateutil==2.9.0.post0
pytz==2025.1
PyYAML==6.0.2
requests==2.32.3
rich==13.9.4
six==1.17.0
sympy==1.13.1
torch==2.2.2
torchvision==0.21.0
tqdm==4.67.1
typing_extensions==4.12.2
tzdata==2025.1
urllib3==2.3.0
xxhash==3.5.0
yarl==1.18.3

Thanks for the support

danweil66, 2025-02-13:
Window 11 Arm64

sun-xiangyu, 2025-02-13:
> Window 11 Arm64

We haven't tested this system yet. From your log, the values of exponent and shape are clearly incorrect.  This might be due to the incorrect adaptation to the operating system.

BlueSkyB, 2025-02-14:
The issue with the abnormally large values is due to the changes made in esp-ppq that caused incompatibility with Windows. We have just submitted the fixes, and you can update esp-ppq again to give it a try. We apologize for the inconvenience caused.

danweil66, 2025-02-17:
Working in my side , thanks for the quick respond and fix, its a nice community hoping to contribute soon as well

closed 2025-02-17 by danweil66