espressif/esp-dl #201: 使用 esp-dl/tools/quantization/quantize_torch_model.py 转的espdl模型，在esp-dl/examples/mobilenet_v2中加载模型报错

WhiteDoveBuct, 2025-02-14:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

[image]

100312dog, 2025-02-18:
@WhiteDoveBuct 这个我这边没法复现呀

WhiteDoveBuct, 2025-02-18:
你先执行 esp-dl(v3.0.0)下的esp-dl/tools/quantization/quantize_torch_model.py 这个脚本，会生成 esp-dl/tools/quantization/models/torch/mobilenet_v2.espdl 这个文件，然后esp-dl/examples/mobilenet_v2 中的模型指向 esp-dl/tools/quantization/models/torch/mobilenet_v2.espdl，如下图：

[image]
，然后编译，烧录到 esp32-p4开发版上，在监视器中可以看到：

[image]

注意：使用的是 esp-dl v3.0.0

WhiteDoveBuct, 2025-02-18:
同样的，这个问题也是版本的问题，v3.0.0上可复现，master上无法复现，说明master已经修复这个问题了

closed 2025-02-18 by BlueSkyB

100312dog, 2025-02-18:
v3.1.0修改了模型的定义。
v3.1.0可以兼容老的模型，正确前向。
但是v3.0.0用新的模型，会有问题。
另外，v3.0.0中
https://github.com/espressif/esp-dl/blob/v3.0.0/models/imagenet_cls/models/p4/imagenet_cls_mobilenetv2_s8_v1.espdl 模型是正常使用Imagenet图像量化的
https://github.com/espressif/esp-dl/blob/v3.0.0/examples/mobilenet_v2/models/esp32p4/mobilenet_v2.espdl
这个模型应该是随机输入量化的，所以看起来推理结果有问题。