espressif/esp-dl #202: 在加载模型的过程中，在 AveragePool 算子抛出异常

WhiteDoveBuct, 2025-02-19:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

esp-dl: master
chip: esp32-p4
esp-idf: 5.4.0
模型和量化脚本：

[code.zip](https://github.com/user-attachments/files/18863828/code.zip)
加载模型：Model *model = new Model("model", fbs::MODEL_LOCATION_IN_FLASH_PARTITION);
报错日志：
```
I (33) boot: ESP-IDF v5.4-dirty 2nd stage bootloader
I (33) boot: compile time Feb 19 2025 15:08:50
I (34) boot: Multicore bootloader
I (35) boot: chip revision: v1.0
I (36) boot: efuse block revision: v0.1
I (40) qio_mode: Enabling default flash chip QIO
I (44) boot.esp32p4: SPI Speed      : 80MHz
I (48) boot.esp32p4: SPI Mode       : QIO
I (52) boot.esp32p4: SPI Flash Size : 16MB
I (56) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 factory          factory app      00 00 00010000 007d0000
I (76) boot:  1 model            Unknown data     01 82 007e0000 0041a000
I (83) boot: End of partition table
I (86) esp_image: segment 0: paddr=00010020 vaddr=480f0020 size=5af84h (372612) map
I (135) esp_image: segment 1: paddr=0006afac vaddr=30100000 size=0002ch (    44) load
I (137) esp_image: segment 2: paddr=0006afe0 vaddr=3010002c size=0003ch (    60) load
I (140) esp_image: segment 3: paddr=0006b024 vaddr=4ff00000 size=04ff4h ( 20468) load
I (150) esp_image: segment 4: paddr=00070020 vaddr=48000020 size=ed8ech (973036) map
I (264) esp_image: segment 5: paddr=0015d914 vaddr=4ff04ff4 size=0d4c4h ( 54468) load
I (272) esp_image: segment 6: paddr=0016ade0 vaddr=4ff12500 size=032f0h ( 13040) load
I (279) boot: Loaded app from partition at offset 0x10000
I (280) boot: Disabling RNG early entropy source...
I (290) hex_psram: vendor id    : 0x0d (AP)
I (291) hex_psram: Latency      : 0x01 (Fixed)
I (291) hex_psram: DriveStr.    : 0x00 (25 Ohm)
I (291) hex_psram: dev id       : 0x03 (generation 4)
I (296) hex_psram: density      : 0x07 (256 Mbit)
I (301) hex_psram: good-die     : 0x06 (Pass)
I (305) hex_psram: SRF          : 0x02 (Slow Refresh)
I (310) hex_psram: BurstType    : 0x00 ( Wrap)
I (314) hex_psram: BurstLen     : 0x03 (2048 Byte)
I (318) hex_psram: BitMode      : 0x01 (X16 Mode)
I (323) hex_psram: Readlatency  : 0x04 (14 cycles@Fixed)
I (328) hex_psram: DriveStrength: 0x00 (1/1)
I (332) MSPI DQS: tuning success, best phase id is 2
I (515) MSPI DQS: tuning success, best delayline id is 12
I esp_psram: Found 32MB PSRAM device
I esp_psram: Speed: 200MHz
I (533) mmu_psram: .rodata xip on psram
I (579) mmu_psram: .text xip on psram
I (579) hex_psram: psram CS IO is dedicated
I (580) cpu_start: Multicore app
I (1067) esp_psram: SPI SRAM memory test OK
I (1076) cpu_start: Pro cpu start user code
I (1076) cpu_start: cpu freq: 360000000 Hz
I (1076) app_init: Application information:
I (1076) app_init: Project name:     mobilenet_v2_example
I (1081) app_init: App version:      v3.0.0-65-g62684a6-dirty
I (1087) app_init: Compile time:     Feb 19 2025 15:07:20
I (1092) app_init: ELF file SHA256:  c806c1b00...
I (1096) app_init: ESP-IDF:          v5.4-dirty
[... 126 lines cut ...]
4ff1aa20: 0x6576412f 0x65676172 0x6c6f6f50 0x4803a400 0x4ff1aac8 0x4ff1aae8 0x0000000f 0x4ff1694c
--- 0x4803a400: dl::module::ModuleCreator::get_instance() at /home/lzm/esp-dl/esp-dl/dl/module/include/dl_module_creator.hpp:64
 (inlined by) dl::Model::load(fbs::FbsModel*) at /home/lzm/esp-dl/esp-dl/dl/model/src/dl_model_base.cpp:123

4ff1aa40: 0x4ff23fec 0x4ff29c04 0x0000000c 0x4803a5e8 0x00000000 0x00000000 0x00000000 0x00000000
--- 0x4803a5e8: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_data() const at /home/lzm/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/riscv32-esp-elf/include/c++/14.2.0/bits/basic_string.h:228
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_is_local() const at /home/lzm/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/riscv32-esp-elf/include/c++/14.2.0/bits/basic_string.h:269
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_dispose() at /home/lzm/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/riscv32-esp-elf/include/c++/14.2.0/bits/basic_string.h:287
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string() at /home/lzm/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/riscv32-esp-elf/include/c++/14.2.0/bits/basic_string.h:809
 (inlined by) dl::module::ModuleCreator::create(fbs::FbsModel*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) at /home/lzm/esp-dl/esp-dl/dl/module/include/dl_module_creator.hpp:92
 (inlined by) dl::Model::load(fbs::FbsModel*) at /home/lzm/esp-dl/esp-dl/dl/model/src/dl_model_base.cpp:135

4ff1aa60: 0x00000001 0x00000002 0x00000000 0x00000678 0x4ff24220 0x00000045 0x480f6000 0x0000000b
4ff1aa80: 0x00000001 0x00000004 0x00000004 0x4ff2a2e4 0x4ff2a9bc 0x4ff2aee4 0x4ff1aaa0 0x0000000c
4ff1aaa0: 0x6576412f 0x65676172 0x6c6f6f50 0x00365f00 0x4ff1aab8 0x0000000b 0x72657641 0x50656761
4ff1aac0: 0x006c6f6f 0x00007261 0x4ff24220 0x0000000c 0x6576412f 0x65676172 0x6c6f6f50 0x00365f00
4ff1aae0: 0x4ff1aae8 0x0000000c 0x6576412f 0x65676172 0x6c6f6f50 0x00365f00 0x4ff1ab00 0x00000000
4ff1ab00: 0x65764100 0x65676172 0x6c6f6f50 0x00365f00 0x00000001 0x00000003 0x4ff19dac 0x00000000
4ff1ab20: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000001 0x00000001
4ff1ab40: 0x00000000 0x4ff19dac 0x4ff23fec 0x4803b130 0x00000000 0x00000000 0x4ff23fec 0x00000000
--- 0x4803b130: dl::Model::Model(char const*, fbs::model_location_type_t, int, dl::memory_manager_t, unsigned char*, bool) at /home/lzm/esp-dl/esp-dl/dl/model/src/dl_model_base.cpp:19 (discriminator 1)

4ff1ab60: 0x00000000 0x00000000 0x480f4000 0x4800c9ee 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
--- 0x4800c9ee: app_main at /home/lzm/esp-dl/examples/audia_test/build/../main/app_main.cpp:120’‘’
```

sun-xiangyu, 2025-02-26:
我怀疑是因为 默认假设了 pads 参数是完整的，你的可能是不完整的

sun-xiangyu, 2025-03-20:
This bug has been fixed, please try again.

WhiteDoveBuct, 2025-03-20:
您好，我会尽快回复您的！谢谢！