espressif/esp-dl #204: tf.reverse support

carolineys, 2025-02-28:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

When I tried to export to esp-dl from onnx, I got this error:
RuntimeError: Op Execution Error: timnet__model_1/model/tf.reverse/ReverseV2_ReverseSequence__20(Type: ReverseSequence, Num of Input: 2, Num of Output: 1)
I think it is because tf.reverse is not included as a suported operator yet. Is there any work around you recommend? 
I tried to use tf.slice to replace my tf.reverse, but I was not able to get it work. 

### Use cases

export onnx model to espdl

sun-xiangyu, 2025-02-28:
Hi @carolineys , Which ONNX operator does tf.reverse in your model get exported to?

carolineys, 2025-03-02:
Hi @sun-xiangyu, Could you please provide an estimated timeline for when we can expect this operator to be supported?

sun-xiangyu, 2025-03-03:
About two weeks

sun-xiangyu, 2025-03-12:
@carolineys，ReverseSequence has been supported.

closed 2025-03-12 by sun-xiangyu