espressif/esp-dl #206: Using Cyclic Buffer Input

danweil66, 2025-03-03:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hey, can I use cyclic buffer data for inference?
This means that data is not stored in consecutive memory but rather divided into two chunks

sun-xiangyu, 2025-03-03:
I think that you can achieve this by creating two instances of the model. If you want to achieve this with a single model instance, you can inherit the Model class and add your own code.

danweil66, 2025-03-03:
I have seen "class TensorBase {
public:
    int size;                     ///< size of element including padding
    std::vector<int> shape;       ///< shape of Tensor
    dtype_t dtype;                ///< data type of element
    int exponent;                 ///<  exponent of element
    bool auto_free;               ///<  free element when object destroy
    std::vector<int> axis_offset; ///<  element offset of each axis
    void *data;                   ///<  data pointer
    void *cache;                  ///<  cache pointer， used for preload and do not need to free
    uint32_t caps;                ///<  flags indicating the type of memory
".
Can we use the   std::vector<int> axis_offset; ///<  element offset of each axis?

sun-xiangyu, 2025-03-03:
> I have seen "class TensorBase { public: int size; ///< size of element including padding std::vector shape; ///< shape of Tensor dtype_t dtype; ///< data type of element int exponent; ///< exponent of element bool auto_free; ///< free element when object destroy std::vector axis_offset; ///< element offset of each axis void *data; ///< data pointer void *cache; ///< cache pointer， used for preload and do not need to free uint32_t caps; ///< flags indicating the type of memory ". Can we use the std::vector axis_offset; ///< element offset of each axis?

The `axis_offset` is automatically updated based on the `shape` and is used for quick element indexing.

sun-xiangyu, 2025-03-03:
Is the purpose of using cyclic buffer data  to reuse the previously computed results?

danweil66, 2025-03-03:
Yes, I need to stack X frames for single input but some can be reuse in next iteration

sun-xiangyu, 2025-03-04:
Currently, esp-dl does not support data reuse. You need to develop your own memory management method based on MemoryManagerBase.

jfinkelstein83, 2025-03-04:
When model in runs, is the input data copied? Can you run without copying? Does running the model effect the original data?

sun-xiangyu, 2025-03-05:
The input data is copied to the shared memory within the model, so the original input data will not be modified.

jfinkelstein83, 2025-03-05:
How would we be able to implement that the data supplied to the model would not be copied and would run in place on the input buffer provided. What part of the process copies duplicates the data?

feizi, 2025-03-06:
The feature is not supported yet. Now the model's memory is a single, continuously reused block.  You can get the data pointer of the input tensors and then perform in-place operations on your own.

danweil66, 2025-03-09:
I didn't completely understand. @feizi, how can I get a pointer to Tensor input data? If I am working with that pointer in preprocess, can I avoid the copy? 
Thanks for the support