espressif/esp-dl #209: Concat operator not supported

carolineys, 2025-03-04:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

When I attempted to deploy my .espdl model on esp32s3, I encountered the following error:

I (1078) dl::Model: net__model_1/model/lambda/stack_Concat__726: Concat

E (1078) dl::Model: Concat is not supported. Please implement and register it first.

Since the concat operator is supported by esp-ppq, I assumed that the list of supported operators would be the same. Is there any workaround if this concat operator is indeed not supported if I need to deploy this model?
Where can I find the list of operators supported by esp-idf?

sun-xiangyu, 2025-03-05:
https://github.com/espressif/esp-dl/blob/master/operator_support_state.md

carolineys, 2025-03-05:
Thank you @sun-xiangyu !  Concat is in the list of supported operator, but when I tried to deploy the model following the example: https://github.com/espressif/esp-dl/blob/master/examples/mobilenet_v2/main/app_main.cpp I got the error : 
E (1078) dl::Model: Concat is not supported. Please implement and register it first.

sun-xiangyu, 2025-03-05:
I guess it's a quantization issue, because Concat currently only supports int8 and int16, while the data type of your Concat might be float32.

carolineys, 2025-03-05:
You mean during quantization (exporting the model to espdl format), I need to explicitly define the inputs to be int8 or int16 somewhere? Do you mind giving an example or some guidance on how to do it?

sun-xiangyu, 2025-03-05:
esp-dl documents: https://docs.espressif.com/projects/esp-dl/en/latest/index.html