espressif/esp-dl #211: 模型加载过程中，Guru Meditation Error

carolineys, 2025-03-06:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

I (315) spi_flash: flash io: qio
I (319) sleep: Configure to isolate all GPIO pins in sleep state
I (326) sleep: Enable automatic switching of GPIO sleep configuration
I (333) main_task: Started on CPU0
I (363) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (363) main_task: Calling app_main()
I (363) zc_demo: get into app_main
I (373) FbsLoader: The storage free size is 23808 KB
I (373) FbsLoader: The partition size is 3900 KB
I (393) dl::Model: model:tf2onnx, version:0

I (393) dl::Model: timnet__model_1/model/conv1d/Conv1D/ExpandDims: Unsqueeze
I (403) dl::Model: timnet__model_1/model/conv1d/Conv1D__9: Transpose
I (403) dl::Model: PPQ_Operation_0: QuantizeLinear
I (413) dl::Model: PPQ_Operation_37: Transpose
I (413) dl::Model: timnet__model_1/model/conv1d/Conv1D: Conv
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4202cfa7  PS      : 0x00060b30  A0      : 0x8201d70f  A1      : 0x3fca29b0  
--- 0x4202cfa7: dl::module::Conv2D::deserialize(fbs::FbsModel*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) at /media/lwt/extraDisk/llm/esp-dl/esp-dl/dl/module/include/dl_module_conv.hpp:211

A2      : 0x00000000  A3      : 0x3fca2a40  A4      : 0x3fcbf2b8  A5      : 0x3fca29b8  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3fca2990  
A10     : 0x3fcbf258  A11     : 0x3fca29b0  A12     : 0x00000002  A13     : 0x00000400  
A14     : 0x3fcbe5dc  A15     : 0x00000054  SAR     : 0x00000009  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  
--- 0x40056f5c: memcpy in ROM
0x40056f72: memcpy in ROM

Backtrace: 0x4202cfa4:0x3fca29b0 0x4201d70c:0x3fca2a40 0x4202e6da:0x3fca2a80 0x4202e942:0x3fca2b70 0x4200a30e:0x3fca2ba0 0x4209e797:0x3fca2c20 0x40380dd1:0x3fca2c40
--- 0x4202cfa4: std::vector<int, std::allocator<int> >::operator[](unsigned int) at /home/lwt/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/stl_vector.h:1126
 (inlined by) dl::module::Conv2D::deserialize(fbs::FbsModel*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) at /media/lwt/extraDisk/llm/esp-dl/esp-dl/dl/module/include/dl_module_conv.hpp:204
0x4201d70c: dl::module::Module* std::__invoke_impl<dl::module::Module*, dl::module::Module* (*&)(fbs::FbsModel*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), fbs::FbsModel*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::__invoke_other, dl::module::Module* (*&)(fbs::FbsModel*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), fbs::FbsModel*&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&) at /home/lwt/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/invoke.h:61 (discriminator 1)
 (inlined by) std::enable_if<is_invocable_r_v<dl::module::Module*, dl::module::Module* (*&)(fbs::FbsModel*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), fbs::FbsModel*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, dl::module::Module*>::type std::__invoke_r<dl::module::Module*, dl::module::Module* (*&)(fbs::FbsModel*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), fbs::FbsModel*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(dl::module::Module* (*&)(fbs::FbsModel*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), fbs::FbsModel*&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&) at /home/lwt/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/invoke.h:114 (discriminator 1)
 (inlined by) std::_Function_handler<dl::module::Module* (fbs::FbsModel*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), dl::module::Module* (*)(fbs::FbsModel*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)>::_M_invoke(std::_Any_data const&, fbs::FbsModel*&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&) at /home/lwt/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/std_function.h:290 (discriminator 1)
0x4202e6da: std::function<dl::module::Module* (fbs::FbsModel*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)>::operator()(fbs::FbsModel*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) const at /home/lwt/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/std_function.h:591
 (inlined by) dl::module::ModuleCreator::create(fbs::FbsModel*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) at /media/lwt/extraDisk/llm/esp-dl/esp-dl/dl/module/include/dl_module_creator.hpp:92
 (inlined by) dl::Model::load(fbs::FbsModel*) at /media/lwt/extraDisk/llm/esp-dl/esp-dl/dl/model/src/dl_model_base.cpp:135
0x4202e942: dl::Model::load(char const*, fbs::model_location_type_t, unsigned char*, bool) at /media/lwt/extraDisk/llm/esp-dl/esp-dl/dl/model/src/dl_model_base.cpp:81 (discriminator 1)
 (inlined by) dl::Model::Model(char const*, fbs::model_location_type_t, int, dl::memory_manager_t, unsigned char*, bool) at /media/lwt/extraDisk/llm/esp-dl/esp-dl/dl/model/src/dl_model_base.cpp:19 (discriminator 1)
0x4200a30e: app_main at /media/lwt/extraDisk/llm/esp-dl/examples/mobilenet_v2/build/../main/app_main.cpp:44 (discriminator 1)
0x4209e797: main_task at /home/lwt/adf_mpy/esp-idf/components/freertos/app_startup.c:208
0x40380dd1: vPortTaskWrapper at /home/lwt/adf_mpy/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

carolineys, 2025-03-06:
[net_test.onnx.zip](https://github.com/user-attachments/files/19104332/net_test.onnx.zip)
Attached is the onnx file, and here is the python quantization code I used:

import os
import subprocess
from typing import Iterable, Tuple, List, Tuple

import torch
from ppq import QuantizationSettingFactory, QuantizationSetting
from ppq.api import espdl_quantize_onnx, get_target_platform
from torch.utils.data import DataLoader
from torch.utils.data.dataset import Subset
import urllib.request
import zipfile
import random
import onnx

def quant_setting_mobilenet_v2(
    onnx_path: str,
    optim_quant_method: List[str] = None,
) -> Tuple[QuantizationSetting, str]:
    """Quantize onnx model with optim_quant_method.

    Args:
        optim_quant_method (List[str]): support 'MixedPrecision_quantization', 'LayerwiseEqualization_quantization'
        -'MixedPrecision_quantization': if some layers in model have larger errors in 8-bit quantization, dispathching
                                        the layers to 16-bit quantization. You can remove or add layers according to your
                                        needs.
        -'LayerwiseEqualization_quantization'： using weight equalization strategy, which is proposed by Markus Nagel.
                                                Refer to paper https://openaccess.thecvf.com/content_ICCV_2019/papers/Nagel_Data-Free_Quantization_Through_Weight_Equalization_and_Bias_Correction_ICCV_2019_paper.pdf for more information.
                                                Since ReLU6 exists in MobilenetV2, convert ReLU6 to ReLU for better precision.

    Returns:
        [tuple]: [QuantizationSetting, str]
    """
    quant_setting = QuantizationSettingFactory.espdl_setting()
    if optim_quant_method is not None:
        if "MixedPrecision_quantization" in optim_quant_method:
            # These layers have larger errors in 8-bit quantization, dispatching to 16-bit quantization.
            # You can remove or add layers according to your needs.
            quant_setting.dispatching_table.append(
                "/features/features.1/conv/conv.0/conv.0.0/Conv",
                get_target_platform(TARGET, 16),
            )
            quant_setting.dispatching_table.append(
                "/features/features.1/conv/conv.0/conv.0.2/Clip",
                get_target_platform(TARGET, 16),
            )
        elif "LayerwiseEqualization_quantization" in optim_quant_method:
            # layerwise equalization
            quant_setting.equalization = True
            quant_setting.equalization_setting.iterations = 4
            quant_setting.equalization_setting.value_threshold = 0.4
            quant_setting.equalization_setting.opt_level = 2
            quant_setting.equalization_setting.interested_layers = None
            # replace ReLU6 with ReLU
            onnx_path = onnx_path.replace("mobilenet_v2.onnx", "mobilenet_v2_relu.onnx")
        else:
            raise ValueError(
                "Please set optim_quant_method correctly. Support 'MixedPrecision_quantization', 'LayerwiseEqualization_quantization'"
            )
        
    print("finished")

    return quant_setting, onnx_path

def collate_fn1(x: Tuple) -> torch.Tensor:
    return torch.cat([sample[0].unsqueeze(0) for sample in x], dim=0)

def collate_fn2(batch: torch.Tensor) -> torch.Tensor:
    return batch.to(DEVICE)

def report_hook(blocknum, blocksize, total):
    downloaded = blocknum * blocksize
    percent = downloaded / total * 100
    print(f"\rDownloading calibration dataset: {percent:.2f}%", end="")

if __name__ == "__main__":
    BATCH_SIZE = 1
    N = 90
    INPUT_SHAPE = [N, 39]
    DEVICE = "cpu"  #  'cuda' or 'cpu', if you use cuda, please make sure that cuda is available
    TARGET = "esp32s3"
    NUM_OF_BITS = 8
    ONNX_PATH = "./net.onnx"
    ESPDL_MODLE_PATH = "net.espdl"

    model = onnx.load("net.onnx")
    opset_version = model.opset_import[0].version
    print(f"Opset version: {opset_version}")

    # define a dataloader here
    def load_random_calibration_dataset() -> Iterable:
        return [torch.rand(size=(N, 39)) for _ in range(BATCH_SIZE)]

        # Load training data for creating a calibration dataloader.
    dataloader = DataLoader(
        dataset=load_random_calibration_dataset(),
        batch_size=BATCH_SIZE,
        shuffle=False,
    )

    # -------------------------------------------
    # Quantize ONNX Model.
    # --------------------------------------------

    # create a setting for quantizing your network with ESPDL.
    # if you don't need to optimize quantization, set the input 1 of the quant_setting_mobilenet_v2 function None
    # Example: Using LayerwiseEqualization_quantization
    quant_setting, ONNX_PATH = quant_setting_mobilenet_v2(
        ONNX_PATH, ["LayerwiseEqualization_quantization"]
    )

    quant_ppq_graph = espdl_quantize_onnx(
        onnx_import_file=ONNX_PATH,
        espdl_export_file=ESPDL_MODLE_PATH,
        calib_dataloader=dataloader,
        calib_steps=8,
        input_shape=[1] + INPUT_SHAPE,
        target=TARGET,
        num_of_bits=NUM_OF_BITS,
        collate_fn=collate_fn2,
        setting=quant_setting,
        device=DEVICE,
        error_report=True,
        skip_export=False,
        export_test_values=False,
        verbose=1,
    )

BlueSkyB, 2025-05-09:
你提供的onnx，与你量化时的input shape不匹配，onnx输入形状是：[unk__1236,90,39,1]，还有个动态batch维度。但是你量化时，只构建了三维输入。
我用esp-ppq量化你的onnx模型，会报错，前向会出现异常。
另外3月6号之前，我们还不支持ReverseSequence算子，所以那时部署也会遇到算子问题。