espressif/esp-dl #218: dl::Model: Assign input failed

1a1a1a1a1a1a1a1, 2025-03-17:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I want to input the RGB565 image captured by the camera into a deep - learning model. The captured image is 240*320 in RGB565 format. I plan to convert it to a 128*128*3 RGB888 image and then input it into a tensor for inference in the deep - learning model. Here is my code.
#include "who_human_face_detection.hpp"
#include "esp_log.h"
#include "esp_camera.h"
#include "dl_image.hpp"
#include "esp_camera.h"
#include "esp32_s3_szp.h"
#include "dl_model_base.hpp"
#include "esp_log.h"
#include "esp_timer.h"
#include "fbs_loader.hpp"
#include <type_traits>
#include "esp_heap_caps.h"
#include "string.h"

using namespace dl;

static const char *TAG = "flowerrecognition";

static QueueHandle_t xQueueLCDFrame = NULL;
static QueueHandle_t xQueueAIFrame = NULL;

static bool gEvent = true;
static bool gReturnFB = true;

static uint8_t ai_buffer[128 * 128 * 3];

// void scale_RGB565_to_CHW(uint8_t *src_buf, int src_width, int src_height,
//                          uint8_t *dst_buf, int dst_width, int dst_height)
// {
//     uint16_t *src = (uint16_t *)src_buf;
//     int chw_size = dst_width * dst_height; // 128*128

//     for (int y_dst = 0; y_dst < dst_height; y_dst++)
//     {
//         int y_src = (y_dst * src_height) / dst_height; // 计算源图像y坐标（整数运算）
//         for (int x_dst = 0; x_dst < dst_width; x_dst++)
//         {
//             int x_src = (x_dst * src_width) / dst_width; // 计算源图像x坐标

//             // 提取RGB565像素值
//             uint16_t rgb565 = src[y_src * src_width + x_src];

//             // 转换为RGB888
//             uint8_t r = (rgb565 >> 11) & 0x1F;
//             uint8_t g = (rgb565 >> 5) & 0x3F;
//             uint8_t b = rgb565 & 0x1F;
//             r = (r << 3) | (r >> 2); // 5位转8位
//             g = (g << 2) | (g >> 4); // 6位转8位
//             b = (b << 3) | (b >> 2); // 5位转8位

//             // 计算目标缓冲区的位置
//             int chw_r = y_dst * dst_width + x_dst; // 127*128+127=127*129
//             int chw_g = chw_size + chw_r;          // 128*128+127*129
//             int chw_b = 2 * chw_size + chw_r;      // 2*128*128+127*129

//             // 将RGB值存储到目标缓冲区（确保值范围在0-255之间）
//             dst_buf[chw_r] = r;
[... 679 lines cut ...]
[0;32mI (1081) dl::Model: /features/features.17/conv/conv.0/conv.0.2/Clip: Clip[0m
[0;32mI (1081) dl::Model: /features/features.17/conv/conv.1/conv.1.0/Conv: Conv[0m
[0;32mI (1081) dl::Model: /features/features.17/conv/conv.1/conv.1.2/Clip: Clip[0m
[0;32mI (1091) dl::Model: /features/features.17/conv/conv.2/Conv: Conv[0m
[0;32mI (1121) dl::Model: /features/features.18/features.18.0/Conv: Conv[0m
[0;32mI (1151) dl::Model: /features/features.18/features.18.2/Clip: Clip[0m
[0;32mI (1151) dl::Model: /avgpool/GlobalAveragePool: GlobalAveragePool[0m
[0;32mI (1151) dl::Model: PPQ_Operation_5: Transpose[0m
[0;32mI (1161) dl::Model: /Flatten: Flatten[0m
[0;32mI (1161) dl::Model: /classifier/classifier.1/Gemm: Gemm[0m
[0;32mI (1191) MemoryManagerGreedy: Maximum mermory size: 557056
[0m
[0;32mI (1211) MOBILENET_V2_EXAMPLE: Model loaded in 828 ms[0m
[0;32mI (1211) MOBILENET_V2_EXAMPLE: Running inference...[0m
[0;32mI (1731) MOBILENET_V2_EXAMPLE: Running end[0m
 1.26391e-20
 2.94273e-43
 0
 0
 4.4081e-39
max_index: 0

daisy: 0
[0;32mI (1741) MOBILENET_V2_EXAMPLE: exit  [0m
[0;32mI (1741) main_task: Returned from app_main()[0m

100312dog, 2025-03-18:
@1a1a1a1a1a1a1a1 https://github.com/espressif/esp-dl/blob/master/esp-dl/dl/tensor/src/dl_tensor_base.cpp#L187
报错是在这边出现的，你可以自己debug一下为什么这个函数会返回false。我怀疑是model的input尺寸并不是128*128*3。
验证的方式， model->get_inputs().begin()->second->print();

1a1a1a1a1a1a1a1, 2025-03-18:
> [@1a1a1a1a1a1a1a1](https://github.com/1a1a1a1a1a1a1a1) https://github.com/espressif/esp-dl/blob/master/esp-dl/dl/tensor/src/dl_tensor_base.cpp#L187 报错是在这边出现的，你可以自己debug一下为什么这个函数会返回false。我怀疑是model的input尺寸并不是128x128x3。 验证的方式， model->get_inputs().begin()->second->print();

谢谢你的建议，我会试一下这个方法，但是这个模型的输入的确是1*128*128*3我存储在.hpp文件中的IMAGE_ELEMENT[]就是128*128*3的矩阵，这个可以成功输入，不管怎样，感谢你的建议，我再试试你的方案打印一下。其次我自己排查到一个错误就是我在推理结束后释放了model,这个造成了程序崩溃

1a1a1a1a1a1a1a1, 2025-03-18:
> [@1a1a1a1a1a1a1a1](https://github.com/1a1a1a1a1a1a1a1) https://github.com/espressif/esp-dl/blob/master/esp-dl/dl/tensor/src/dl_tensor_base.cpp#L187 报错是在这边出现的，你可以自己debug一下为什么这个函数会返回false。我怀疑是model的input尺寸并不是128x128x3。 验证的方式， model->get_inputs().begin()->second->print();

以下是我生成输入.hpp的python程序，这个生成的矩阵的确会是128*128*3
import cv2
import numpy as np

if __name__ == '__main__':
    input_path = r"data\flower7595\flowers\daisy\5547758_eea9edfd54_n.jpg"
    output_path = "rgb_image.hpp"

    # 读取图像并处理为 RGB 格式
    image = cv2.imread(input_path)
    img = cv2.cvtColor(image, cv2.COLOR_BGR2RGB)  # 转换为 RGB 格式
    img = cv2.resize(img, (128, 128))  # 调整尺寸为 128x128

    # 转换为通道优先的格式（1, 3, 128, 128）
    img = np.transpose(img, (2, 0, 1))  # 转换为 (channels, height, width)
    img = np.expand_dims(img, axis=0)  # 添加批量维度，变为 (1, 3, 128, 128)

    c, h, w = img.shape[1:]  # 获取通道、高度和宽度

    with open(output_path, 'w') as file:
        file.write('#pragma once\n')
        file.write('#include <stdint.h>\n\n')
        file.write(f'#define IMAGE_BATCH 1\n')
        file.write(f'#define IMAGE_CHANNELS {c}\n')
        file.write(f'#define IMAGE_HEIGHT {h}\n')
        file.write(f'#define IMAGE_WIDTH {w}\n\n')
        file.write('const static uint8_t IMAGE_ELEMENT[] = {\n')

        # 将图像展平为一维数组
        image_flat = img.flatten()
        for i, element in enumerate(image_flat):
            if i == 0:
                file.write('    ')
            file.write(f'{element}, ')
            if (i + 1) % 32 == 0 or i == len(image_flat) - 1:
                file.write('\n    ') if i != len(image_flat) - 1 else file.write('\n')
        file.write('};\n')

    print(f"头文件已生成: {output_path}")

1a1a1a1a1a1a1a1, 2025-03-18:
> 谢谢你的建议，我会试一下这个方法，但是这个模型的输入的确是1_128_128_3我存储在.hpp文件中的IMAGE_ELEMENT[]就是128_128*3的矩阵，这个可以成功输入，不管怎样，感谢你的建议，我再试试你的方案打印一下。其次我自己排查到一个错误就是我在推理结束后释放了model,这个造成了程序崩溃

真的太感谢你了，我的问题解决了,不知道为什么，他没有加载我之前训练好的模型，我重新替换了一下，现在它可以正常运行花卉分类了。我把我的代码贴出来供后来者参考一下。#include "dl_model_base.hpp"
#include "esp_log.h"
#include "esp_timer.h"
#include "fbs_loader.hpp"
#include <type_traits>
#include "rgb_image.hpp"
 
 
static const char *TAG = "MOBILENET_V2_EXAMPLE";
 
// using namespace fbs;
using namespace dl;
int input_height = 128;
int input_width = 128;
int input_channel = 3;
int input_exponent = -7;
 
 
 
extern "C" void app_main(void)
{
	ESP_LOGI(TAG, "Starting gesture recognition...");
	
	    int64_t start_time = esp_timer_get_time();  // 模型加载开始时间
	
	    // 创建模型实例
	    Model* model = new Model("model", fbs::MODEL_LOCATION_IN_FLASH_PARTITION);
	    if (!model) {
	        ESP_LOGE(TAG, "Failed to create model");
	        return;
	    }
	
	    int64_t load_time = esp_timer_get_time();   // 模型加载结束时间
	    ESP_LOGI(TAG, "Model loaded in %lld ms", (load_time - start_time) / 1000);
	
	    // 准备输入数据
	    TensorBase* input_tensor = new TensorBase(
		        {1, 3, 128, 128},              // 输入形状 [batch, channel, height, width]
		        IMAGE_ELEMENT,             // 来自 test_image.hpp 的数据指针
		        -7,                          // exponent，需要根据实际量化参数调整
		        dl::DATA_TYPE_INT8,          // 数据类型
		        false,                       // 是否需要拷贝数据
		        MALLOC_CAP_SPIRAM           // 内存分配标志
		    ); 
	     
	
	    ESP_LOGI(TAG, "Running inference...");
		model->run(input_tensor,RUNTIME_MODE_AUTO);

		model->get_inputs().begin()->second->print();
		ESP_LOGI(TAG, "Running end");
		
		auto outputs = model->get_outputs();
		auto output_iter = outputs.begin();
		dl::TensorBase* tensor = output_iter->second;
		float max_value = 0;
		size_t max_index = 0;
		if (outputs.empty()) {
		        ESP_LOGE(TAG, "No outputs from model");
		    }  {
					float* output_data = static_cast<float*>(tensor->get_element_ptr());
			         
			        for (size_t i = 0; i <5; ++i) {
			        	std::cout << " " << output_data[i] << std::endl;
						            if (output_data[i] > max_value) {
						            max_value = output_data[i];
						            max_index = i;
						                    }
						                }
				   std::cout << "max_index: " << max_index << std::endl;
													 
											}
											
		printf("\n");
		switch (max_index)
		                {
		                    case 0:
		                    printf("daisy: 0");
		                    break;
		                    case 1:
		                    printf("dandelion: 1");
		                    break;
		                    case 2:
		                    printf("rose: 2");
		                    break;
		                    case 3:
		                    printf("sunflower: 3");
		                    break;
		                    case 4:
		                    printf("tulip: 4");
		                    break;
		                    default:
		                    printf("No result");
		
		                }
		                printf("\n");
											
	
	
	delete input_tensor;
    delete model;
    ESP_LOGI(TAG, "exit  ");
}

1a1a1a1a1a1a1a1, 2025-03-18:
> [@1a1a1a1a1a1a1a1](https://github.com/1a1a1a1a1a1a1a1) https://github.com/espressif/esp-dl/blob/master/esp-dl/dl/tensor/src/dl_tensor_base.cpp#L187 报错是在这边出现的，你可以自己debug一下为什么这个函数会返回false。我怀疑是model的input尺寸并不是128x128x3。 验证的方式， model->get_inputs().begin()->second->print();

真的太感谢你了，我的问题解决了,不知道为什么，他没有加载我之前训练好的模型，我重新替换了一下，现在它可以正常运行花卉分类了。我把我的代码贴出来供后来者参考一下。#include "who_human_face_detection.hpp"
#include "esp_log.h"
#include "esp_camera.h"
#include "dl_image.hpp"
#include "esp_camera.h"
#include "esp32_s3_szp.h"
#include "dl_model_base.hpp"
#include "esp_log.h"
#include "esp_timer.h"
#include "fbs_loader.hpp"
#include <type_traits>
#include "esp_heap_caps.h"

using namespace dl;
static const char *TAG = "flowerrecognition";

static QueueHandle_t xQueueLCDFrame = NULL;
static QueueHandle_t xQueueAIFrame = NULL;

static bool gEvent = true;
static bool gReturnFB = true;

void scale_RGB565_to_CHW(uint8_t *src_buf, int src_width, int src_height, 
    uint8_t *dst_buf, int dst_width, int dst_height) 
{
    uint16_t *src = (uint16_t *)src_buf;
    int chw_size = dst_width * dst_height;

    for (int y_dst = 0; y_dst < dst_height; y_dst++) {
        int y_src = (y_dst * src_height) / dst_height; // 计算源图像y坐标（整数运算）
        for (int x_dst = 0; x_dst < dst_width; x_dst++) {
            int x_src = (x_dst * src_width) / dst_width; // 计算源图像x坐标

            // 提取RGB565像素值
            uint16_t rgb565 = src[y_src * src_width + x_src];

            // 转换为RGB888
            uint8_t r = (rgb565 >> 11) & 0x1F;
            uint8_t g = (rgb565 >> 5)  & 0x3F;
            uint8_t b = rgb565 & 0x1F;
            r = (r << 3) | (r >> 2);  // 5位转8位
            g = (g << 2) | (g >> 4);  // 6位转8位
            b = (b << 3) | (b >> 2);  // 5位转8位

            // 计算目标缓冲区的位置
            int chw_r = y_dst * dst_width + x_dst;
            int chw_g = chw_size + chw_r;
            int chw_b = 2 * chw_size + chw_r;

            // 将RGB值存储到目标缓冲区（确保值范围在0-255之间）
            dst_buf[chw_r] = r;
            dst_buf[chw_g] = g;
            dst_buf[chw_b] = b;
        }
    }
}

uint8_t ai_buffer[128*128*3];

// AI处理任务
static void task_process_ai(void *arg)
{
    camera_fb_t *frame = NULL;
    //uint8_t *ai_buffer = (uint8_t *)heap_caps_malloc(128*128*3, MALLOC_CAP_SPIRAM);

 
    ESP_LOGI(TAG, "Starting flower recognition...");

[... 481 lines cut ...]
[0;31mE (5836) lcd_panel.io.spi: panel_io_spi_tx_color(390): spi transmit (queue) color failed[0m
[0;31mE (5896) lcd_panel.st7789: panel_st7789_draw_bitmap(223): io tx color failed[0m
[0;32mI (6436) flowerrecognition: Running end[0m
 -8.41532e+09
 2.0331e-39
 0
 2.51188e-15
 2.81241e-23
max_index: 3

sunflower: 3
[0;32mI (6436) print: shape: [1, 128, 128, 3], dtype: int8, exponent: -7, auto_free: 0[0m
[0;32mI (6446) flowerrecognition: Input data 0: 41[0m
[0;32mI (6446) flowerrecognition: Input data 1: 41[0m
[0;32mI (6446) flowerrecognition: Input data 2: 74[0m
[0;32mI (6456) flowerrecognition: Input data 3: 107[0m
[0;32mI (6456) flowerrecognition: Input data 4: 140[0m
[0;32mI (6466) flowerrecognition: Input data 5: 173[0m
[0;32mI (6466) flowerrecognition: Input data 6: 247[0m
[0;32mI (6476) flowerrecognition: Input data 7: 8[0m
[0;32mI (6476) flowerrecognition: Input data 8: 82[0m
[0;32mI (6486) flowerrecognition: Input data 9: 82[0m
[0;32mI (6486) flowerrecognition: Running inference...[0m
[0;31mE (6456) lcd_panel.io.spi: panel_io_spi_tx_color(390): spi transmit (queue) color failed[0m
[0;31mE (6506) lcd_panel.st7789: panel_st7789_draw_bitmap(223): io tx color failed[0m