espressif/esp-dl #219: Cannot run yolo11n_qat.py on macOS (Apple M2 Pro)

MacWyznawca, 2025-03-17:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

When I run `yolo11n_qat.py` I have error with loading lib `onnxsim_cpp2py_export.cpython-312-darwin.so`

### Actual behavior (suspected bug)

Error `import onnxsim.onnxsim_cpp2py_export as C`

### Error logs or terminal output

```shell
ImportError: dlopen(/Users/jkk/miniconda3/lib/python3.12/site-packages/onnxsim/onnxsim_cpp2py_export.cpython-312-darwin.so, 0x0002): tried: '/Users/jkk/miniconda3/lib/python3.12/site-packages/onnxsim/onnxsim_cpp2py_export.cpython-312-darwin.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/Users/jkk/miniconda3/lib/python3.12/site-packages/onnxsim/onnxsim_cpp2py_export.cpython-312-darwin.so' (no such file), '/Users/jkk/miniconda3/lib/python3.12/site-packages/onnxsim/onnxsim_cpp2py_export.cpython-312-darwin.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64'))
```

### Steps to reproduce the behavior

Run python yolo11n_qat.py on Mac with Apple Silicone
Before I install I required packages

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

macOS

### Shell

ZSH

Ginosko-mia, 2025-03-18:
From the logs, the **onnxsim_cpp2py_export.cpython-312-darwin.so** file in onnxsim you installed was compiled for the x86_64 architecture and is incompatible with your system's ARM architecture. To solve it,  
1. python3.8~3.11 is recommended.
2. reinstall onnxsim. Choose one according to your architecture(ARM64 for Apple M1/M2 ) and python version from [here](https://pypi.org/project/onnxsim/0.4.36/#files).

MacWyznawca, 2025-03-18:
Thanks @Ginosko-mia!