espressif/esp-dl #221: ValueError: Device type mps not understand [macOS]

MacWyznawca, 2025-03-18:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I would like to use the Apple Metal acceleration in PyTorch. 
Unfortunately the `mps` device is not recognised.

### Actual behavior (suspected bug)

I habe: ValueError: Device type mps not understand.

### Error logs or terminal output

```shell
___________ ____        ____  ____  ____ 
   / ____/ ___// __ \      / __ \/ __ \/ __ \
  / __/  \__ \/ /_/ /_____/ /_/ / /_/ / / / /
 / /___ ___/ / ____/_____/ ____/ ____/ /_/ / 
/_____//____/_/         /_/   /_/    \___\_\ 

Traceback (most recent call last):
  File "yolo11n_qat.py", line 125, in <module>
    qat_graph = qat()
  File "yolo11n_qat.py", line 73, in qat
    executor = TorchExecutor(graph=graph, device=device)
  File "/Users/jkk/miniconda3/envs/esp-ml/lib/python3.8/site-packages/ppq/executor/torch.py", line 287, in __init__
    super().__init__(graph)
  File "/Users/jkk/miniconda3/envs/esp-ml/lib/python3.8/site-packages/ppq/executor/base.py", line 117, in __init__
    self.load_graph(graph=graph)
  File "/Users/jkk/miniconda3/envs/esp-ml/lib/python3.8/site-packages/ppq/executor/torch.py", line 626, in load_graph
    self._runnable_graph(GraphDeployCommand(device=self._device))
  File "/Users/jkk/miniconda3/envs/esp-ml/lib/python3.8/site-packages/ppq/IR/base/command.py", line 132, in __init__
    raise ValueError(f'Device type {device} not understand.')
ValueError: Device type mps not understand.
```

### Steps to reproduce the behavior

Ex.: change `cpu` to `mps`  in:
`device = "cuda" if torch.cuda.is_available() else "cpu"`

### Project release version

latest

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

MacOS

### Operating system version

15.3.2 (24D81)

### Shell

ZSH

Ginosko-mia, 2025-03-19:
hi, @MacWyznawca. ESP-PPQ don't support "mps" for graph inference. You can use "cpu" first.

MacWyznawca, 2025-03-19:
Hi @Ginosko-mia it's just that in my case (M2 Pro) the `mps` is 60 to 100 times faster than the `cpu`.
When are you going to introduce `mps` support?

BlueSkyB, 2025-03-20:
We can't find any Mac computers with MPS here for the time being. You can download the ESP-PPQ source code and try modifying it yourself. Method:
1.Add the definition "DEPLOY_TO_MPS = 4" in the file "esp-ppq/ppq/IR/base/command.py".
[image]

2.Add the corresponding initialization of DEPLOY_TO_MPS in the GraphDeployCommand class in "esp-ppq/ppq/IR/base/command.py".
[image]

3.Add the corresponding DEPLOY_TO_MPS handling in the “process” and "_acceptable_command_types" function of RunnableGraph class in "esp-ppq/ppq/IR/deploy.py".
[image]

[image]

Finally, reinstall esp-ppq.

MacWyznawca, 2025-03-20:
Thanks, @BlueSkyB works! Some other problems (buffer size etc.) but this is other part ;-)

closed 2025-03-20 by MacWyznawca