espressif/esp-dl #226: Face recognition documentation doesn't match reality

cnadler86, 2025-04-03:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

The documentation provided in the face recognition model is not correct.
I measure the following latencies of detecting and recognizing the example pictures with an ESP32S2 as follow:
MFN:  4200 ms
MBF: 730 ms

For me, it seems that the MBF model is in reality the MFN and the MFN model is something else?
The latency of 730ms could be the addition of the detection and recognition models (seems at least plausible for me comparing this to the examples and documentation provided).
but the latency on the MFN model should be actually lower than the latency of the MBF

100312dog, 2025-04-03:
@cnadler86 In your description you said you used ESP32S2, the results in the document are tested on ESP32S3.
Now the master branch does not fully support ESP32S2.

cnadler86, 2025-04-03:
Sorry that's a typo, used a s3 of course

100312dog, 2025-04-03:
@cnadler86 Can you provide your test code? I just tested it, and the results are consistent with the doc.

cnadler86, 2025-04-03:
I am implementing a micrpython binding, so a lot of function in here are from the micropython framework. I hope it is understandable. Here the relevant code snippets:

**Object definition**
```
struct MP_FaceRecognizer {
    mp_obj_base_t base;
    std::shared_ptr<HumanFaceDetect> FaceDetector;
    std::shared_ptr<HumanFaceFeat> FaceFeat;
    std::shared_ptr<HumanFaceRecognizer> FaceRecognizer;
    dl::image::img_t img;
    bool return_features;   
    char db_path[64];
    bool validate_enroll;
};
```

**Constructor**
```
static mp_obj_t face_recognizer_make_new(const mp_obj_type_t *type, size_t n_args, size_t n_kw, const mp_obj_t *args) {
    enum { ARG_width, ARG_height, ARG_features, ARG_db_path, ARG_pixelformat, ARG_validate_enroll, ARG_model };
    static const mp_arg_t allowed_args[] = {
        { MP_QSTR_width, MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = 320} },
        { MP_QSTR_height, MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = 240} },
        { MP_QSTR_features, MP_ARG_KW_ONLY | MP_ARG_BOOL, {.u_bool = true} },
        { MP_QSTR_db_path, MP_ARG_KW_ONLY | MP_ARG_OBJ, {.u_obj = mp_const_none} },
        { MP_QSTR_pixelformat, MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = dl::image::DL_IMAGE_PIX_TYPE_RGB888} },
        { MP_QSTR_validate_enroll, MP_ARG_KW_ONLY | MP_ARG_BOOL, {.u_bool = false} },
        { MP_QSTR_model, MP_ARG_KW_ONLY | MP_ARG_OBJ, {.u_obj = mp_const_none} },
    };

    mp_arg_val_t parsed_args[MP_ARRAY_SIZE(allowed_args)];
    mp_arg_parse_all_kw_array(n_args, n_kw, args, MP_ARRAY_SIZE(allowed_args), allowed_args, parsed_args);

    MP_FaceRecognizer *self = mp_obj_malloc_with_finaliser(MP_FaceRecognizer, &mp_face_recognizer_type);

    strncpy(self->db_path, "/face.db", sizeof(self->db_path));
    if (parsed_args[ARG_db_path].u_obj != mp_const_none) {
        snprintf(self->db_path, sizeof(self->db_path), "/%s", mp_obj_str_get_str(parsed_args[ARG_db_path].u_obj));
    }
    
    self->FaceDetector = std::make_shared<HumanFaceDetect>();

    if (parsed_args[ARG_model].u_obj == mp_const_none) {
        self->FaceFeat = std::make_shared<HumanFaceFeat>();
    } else {
        const char *model = mp_obj_str_get_str(parsed_args[ARG_model].u_obj);

        if (strcmp(model, "MBF")) {
            self->FaceFeat = std::make_shared<HumanFaceFeat>(HumanFaceFeat::MBF_S8_V1);
        } else if (strcmp(model, "MFN")) {
            self->FaceFeat = std::make_shared<HumanFaceFeat>(HumanFaceFeat::MFN_S8_V1);
        }
        else {
            mp_printf(&mp_plat_print, "Model %s invalid. Using default feature model\n", model);
            self->FaceFeat = std::make_shared<HumanFaceFeat>();
        }
    }
    
    self->FaceRecognizer = std::make_shared<HumanFaceRecognizer>(self->FaceFeat.get(), self->db_path);

    if ((!self->FaceDetector) || (!self->FaceFeat) || (!self->FaceRecognizer)) {
        mp_raise_msg(&mp_type_RuntimeError, "Failed to create model instances");
    }

    mp_esp_dl::initialize_img(self->img, parsed_args[ARG_width].u_int, parsed_args[ARG_height].u_int, 
                             static_cast<dl::image::pix_type_t>(parsed_args[ARG_pixelformat].u_int));

    self->return_features = parsed_args[ARG_features].u_bool;

    return MP_OBJ_FROM_PTR(self);
}
```

**Recognize method**
```
static mp_obj_t face_recognizer_recognize(mp_obj_t self_in, mp_obj_t framebuffer_obj) {
    MP_FaceRecognizer *self = mp_esp_dl::get_and_validate_framebuffer<MP_FaceRecognizer>(self_in, framebuffer_obj); //This gives me an object structure back. The framebuffer_obj is the Image in RGB888 format

    auto &detect_results = self->FaceDetector->run(self->img);

    if (detect_results.size() == 0) {
        return mp_const_none;
    }

    mp_obj_t list = mp_obj_new_list(0, NULL);
    for (const auto &res : detect_results) {
        mp_obj_list_append(list, mp_obj_new_float(res.score));
        mp_obj_t tuple[4];
        for (int i = 0; i < 4; ++i) {
            tuple[i] = mp_obj_new_int(res.box[i]);
        }
        mp_obj_list_append(list, mp_obj_new_tuple(4, tuple));
        
        std::list<dl::detect::result_t> single_result_list = { res };
        auto recon_results = self->FaceRecognizer->recognize(self->img, single_result_list);

        if(recon_results.size() == 0) {
            mp_obj_list_append(list, mp_const_none);
        } else {
            for (const auto &recon : recon_results) {
                mp_obj_t tuple[2];
                tuple[0] = mp_obj_new_int(recon.id);
                tuple[1] = mp_obj_new_float(recon.similarity);
                mp_obj_list_append(list, mp_obj_new_tuple(2, tuple));
            }
        }

        if (self->return_features) {
            mp_obj_t features[10];
            for (int i = 0; i < 10; ++i) {
                features[i] = mp_obj_new_int(res.keypoint[i]);
            }
            mp_obj_list_append(list, mp_obj_new_tuple(10, features));
        }
    }
    return list;
}
```

I also has to do some minor adaptations to the dl__recognition::DataBase in order to "communicate" to the micropython file system, but i don't Think this has in impact in the latency since the features are in memory. 

What bothers me is the fact that the small model is a lot slower than the big one with the exact same code and input data. I can choose at runtime which model to use. 

Since you could not reproduce my issue I assume the poor performance is somewhere in my application. 
Do you have any clue where the poor performance can come from? I also needed to activate RTTI as compile option (had some linker problems), but this should affect both models, i think. I have also think about configuration (didn't use the full configuration provided by the example in the assumption this has no high impact on the performance).

100312dog, 2025-04-03:
@cnadler86 
https://github.com/espressif/esp-dl/blob/master/esp-dl/dl/dl_define.hpp#L11 
Opening this macro will print the time taken by the model. I am not sure if it will work in your case because I am not familiar with micropython.

cnadler86, 2025-04-03:
Yes, this should work. Thank you ver muchö! Will active all of then and test it in the next days. 
I will close this issue here and in case I find the issue to be in the model or pre/post processing open an issue then.

cnadler86, 2025-04-05:
I could improve performance of my application by changing the configuration to the ones given in the examples. Thanks once again.
Just for information: By the small face recon model I am still far away from the benchmark (the model still takes around 1s). I assume it is because I don't activated `CONFIG_SPIRAM_XIP_FROM_PSRAM`. Sadly I could not find so much information about this. With this, my application doesn't start. Perhaps you could add just some information to the documentation (or a comment in the sdk-Config files) that this config might cause problems and may have an impact on models performance

cnadler86, 2025-04-05:
I managed to compile and start my application with `CONFIG_SPIRAM_XIP_FROM_PSRAM` (was using IDF 5.4.0, but there was a bug corrected at 5.4.1). 

I get the following latencies:
**MFN-Model**
```
I (379564) detect: pre: 10161 us
I (379594) detect: model: 32805 us
I (379594) detect: post: 446 us
I (379604) detect: pre: 1193 us
I (379604) detect: model: 5639 us
I (379604) detect: post: 117 us
I (379614) feat: pre: 8354 us
I (380694) feat: model: **1071924** us
I (380694) feat: post: 96 us
```

**MBF-Model:**
```
I (426764) detect: pre: 10172 us
I (426804) detect: model: 32785 us
I (426804) detect: post: 442 us
I (426814) detect: pre: 1189 us
I (426814) detect: model: 5641 us
I (426814) detect: post: 118 us
I (426824) feat: pre: 8346 us
I (427084) feat: model: **256085** us
I (427084) feat: post: 98 us
```

Taking the latencies from the readme:
mfn_s8_v1_s3: 248803
mbf_s8_v1_s3: 1072427

I would say they are flipped in the readme. Can you please recheck?

100312dog, 2025-04-07:
@cnadler86 I have no idea why you met this issue. I tested with the example https://github.com/espressif/esp-dl/tree/master/examples/human_face_recognition, everything seems to be OK. The bigger mbf model runs slower and the smaller model mfn runs faster.

cnadler86, 2025-04-08:
then the problem is definitely at my side. Will try with some other configs then. Thanks for your help!

closed 2025-04-08 by cnadler86