espressif/esp-dl #234: Quantization Error with YOLOv11n (192x192) - Invalid Reshape during espdl_quantize_onnx

gassechen, 2025-04-23:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hi ESP-DL Team,

I trained a YOLOv11n model at 192x192 resolution and attempted to quantize it using PPQ with espdl_quantize_onnx. However, I encountered a runtime error during the quantization process related to a Reshape operation.

`python quantize_onnx_model.py --onnx final_adjusted_yarn192.onnx --espdl yarn.espdl --size 192`

`RuntimeError: shape '[-1, 2, 128, 400]' is invalid for input of size 9216
During handling of the above exception, another exception occurred:
RuntimeError: Op Execution Error: /model.10/m/m.0/attn/Reshape(Type: Reshape, Num of Input: 2, Num of Output: 1)`

Observations
The original ONNX model runs inference correctly in PyTorch.

This appears to be caused by a hardcoded Reshape expecting a resolution different from 192x192, possibly 320x320.

The issue occurs during quantization with PPQ, not during training or ONNX export.

I’m using the calibration dataset provided by Espressif, resized to 192x192.

Questions
Does espdl fully support 192x192 for YOLOv11n?

Are there internal shape assumptions that break when downscaling below 320x320?

Is there a recommended way to avoid fixed-shape errors like this during quantization?

Any help is appreciated. Below is the full script I used for quantization.

Script (quantize_onnx_model.py)
```
import os
import argparse
from ppq import QuantizationSettingFactory
from ppq.api import espdl_quantize_onnx, get_target_platform
from torch.utils.data import DataLoader, Dataset
from torchvision import transforms
from PIL import Image
from onnxsim import simplify
import onnx
import zipfile
import urllib.request
from tqdm import tqdm

class CaliDataset(Dataset):
    def __init__(self, path, size):
        self.size = size
        self.transform = transforms.Compose([
            transforms.ToTensor(),
            transforms.Resize((self.size, self.size)),
            transforms.Normalize(mean=[0, 0, 0], std=[1, 1, 1]),
        ])
        self.imgs_path = [os.path.join(path, img_name) for img_name in os.listdir(path)]

    def __len__(self):
        return len(self.imgs_path)

    def __getitem__(self, idx):
        img = Image.open(self.imgs_path[idx])
        img = self.transform(img)
        return img

def report_hook(blocknum, blocksize, total):
    downloaded = blocknum * blocksize
    percent = downloaded / total * 100
    print(f"\rDownloading calibration dataset: {percent:.2f}%", end="")

def resize_images(source_dir, dest_dir, target_size):
    os.makedirs(dest_dir, exist_ok=True)
    for filename in tqdm(os.listdir(source_dir), desc="Redimensionando imágenes"):
        src_path = os.path.join(source_dir, filename)
        dst_path = os.path.join(dest_dir, filename)
        try:
            img = Image.open(src_path).convert("RGB")
            img = img.resize((target_size, target_size))
            img.save(dst_path)
        except Exception as e:
            print(f"Error with {filename}: {e}")

def quant_yolo11n(onnx_path, espdl_path, size):
    BATCH_SIZE = 1
    INPUT_SHAPE = [3, size, size]
    DEVICE = "cpu"
    TARGET = "esp32s3"
    NUM_OF_BITS = 8
    script_dir = os.path.dirname(os.path.abspath(__file__))
    ONNX_PATH = os.path.join(script_dir, "model.onnx")
    ESPDL_MODLE_PATH = os.path.join(script_dir, "model.espdl")

    yolo11n_caib_url = "https://dl.espressif.com/public/calib_yolo11n.zip"
    CALIB_DIR = "calib_yolo11n"
    RESIZED_CALIB_DIR = f"{CALIB_DIR}_{size}"

    urllib.request.urlretrieve(yolo11n_caib_url, "calib_yolo11n.zip", reporthook=report_hook)
    with zipfile.ZipFile("calib_yolo11n.zip", "r") as zip_file:
        zip_file.extractall("./")

    resize_images(CALIB_DIR, RESIZED_CALIB_DIR, size)

    model = onnx.load(ONNX_PATH)
    model, check = simplify(model)
    assert check, "Simplified ONNX model could not be validated"
    onnx.save(onnx.shape_inference.infer_shapes(model), ONNX_PATH)

    calibration_dataset = CaliDataset(RESIZED_CALIB_DIR, size)
    dataloader = DataLoader(dataset=calibration_dataset, batch_size=BATCH_SIZE, shuffle=False)

    def collate_fn(batch):
        print(f"Batch shape before quantization: {batch.shape}")
        return batch.to(DEVICE)

    quant_setting = QuantizationSettingFactory.espdl_setting()
    quant_setting.dispatching_table.append(
        operation='/model.*/conv/Conv',
        platform=get_target_platform(TARGET, NUM_OF_BITS)
    )

    quant_ppq_graph = espdl_quantize_onnx(
        onnx_import_file=ONNX_PATH,
        espdl_export_file=ESPDL_MODLE_PATH,
        calib_dataloader=dataloader,
        calib_steps=32,
        input_shape=[1] + INPUT_SHAPE,
        target=TARGET,
        num_of_bits=NUM_OF_BITS,
        collate_fn=collate_fn,
        setting=quant_setting,
        device=DEVICE,
        error_report=True,
        skip_export=False,
        export_test_values=False,
        verbose=0,
        inputs=None,
    )
    return quant_ppq_graph

if __name__ == "__main__":
    parser = argparse.ArgumentParser(description="Quantize ONNX model to ESP-DL format")
    parser.add_argument("--onnx", type=str, required=True, help="Input ONNX model path")
    parser.add_argument("--espdl", type=str, required=True, help="Output ESP-DL model path")
    parser.add_argument("--size", type=str, required=True, help="Input resolution, e.g. 640")
    args = parser.parse_args()
    quant_yolo11n(args.onnx, args.espdl, int(args.size))
```

Thanks for your support and great work!

Ginosko-mia, 2025-04-24:
Hi, @gassechen . I exported yolo11n with 192x192 input and quantized it successfully with your quantize script. There is something wrong with your .onnx file. We offer a script which can help you export onnx effortlessly. Please use  [export_onnx.py](https://github.com/espressif/esp-dl/blob/master/models/coco_detect/models/export_onnx.py) for export.

Ginosko-mia, 2025-04-24:
As for your questions, 
1. ESP-DL fully supports 192x192 input for YOLOv11n. 
2. Not exactly. When the input resolution is 96x96 or smaller, there are some issues with the MaxPool layers inside the SPPF block of YOLOv11n. We are aware of this and plan to fix it in the coming weeks. In the meantime, if you're using an input smaller than 96x96, you can safely remove the MaxPool layers from the SPPF block — the results will be the same, as the pooling has no effect at that scale.
 3. We recommend using  [export_onnx.py](https://github.com/espressif/esp-dl/blob/master/models/coco_detect/models/export_onnx.py) when exporting the model to .onnx file.

gassechen, 2025-04-24:
If it worked, then the problem was indeed with the onnx.

closed 2025-04-24 by gassechen