espressif/esp-dl #236: hw_decode_jpeg defaulting to BGR in rgb_order?

nicklasb, 2025-05-04:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I expect rgb_order to default to RGB. 

### Actual behavior (suspected bug)

Instead rgb_order  , it defaults to BGR, which almost drove me nuts until i figured it out.

So this looks wrong in dl_image_jpeg.cpp :    

jpeg_decode_cfg_t decode_cfg = {.output_format = convert_pix_type_to_dec_output_fmt(decoded_img.pix_type),
                                    **.rgb_order = swap_color_bytes ? JPEG_DEC_RGB_ELEMENT_ORDER_RGB
                                                                  : JPEG_DEC_RGB_ELEMENT_ORDER_BGR,**
                                    .conv_std = JPEG_YUV_RGB_CONV_STD_BT601};
    uint32_t out_size = 0;

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

Use ESL-DL and decode a jpg.

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

N/A

### Shell

Bash

100312dog, 2025-05-06:
@nicklasb Yes, you're right. hw_decode_jpeg gives bgr image in default if it is decoded into RGB888. And if the image is decoded to RGB565, it is a rgb image, and the swap_color_bytes only controls the rgb image in big endian or little endian.
It is a little bit strange indeed, i'll change the default behavior and add some comments to the function.

nicklasb, 2025-05-08:
Ok, good stuff!
It will break my code in a hard-to-understand way so please update here when it is merged.

100312dog, 2025-05-09:
@nicklasb It's merged. Please try.

nicklasb, 2025-05-12:
Sorry, forgot this, I just tried with this commit, and it worked great, thank you!

(That said, I do that there are a lot of breaking changes in non-major versions, while I understand that it is early in the development of ESP-DL, please align more to the semantic versioning principles down the line)

closed 2025-05-12 by nicklasb