espressif/esp-dl #237: 完成量化后，在esp32s3上运行model->test()时报错

kuiandhui, 2025-05-07:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

我参考文档中的quantize_onnx_model.py完成了自己模型的量化，在将模型部署到esp32 s3运行时报错，报错信息完整的粘贴在下方。在我的模型输入尺寸为[1,1,96,96]，在调用espdl_quantize_onnx()时我将input shape设置为了该值，在esp32s3上，使用model->get_inputs()并且访问model_input的shape对象，得到的尺寸是[1,,96,96,1]。

奇怪的事情是，当我调用espdl_quantize_onnx()时如果将输入尺寸设置为[1,1,100,100]，运行model->test()时会提示验Testing output 420 Test Pass! 
我将这两个espidf文件附上，除了导出时选择的验证输入的尺寸，没有做其他改动

[model_.zip](https://github.com/user-attachments/files/20085717/model_.zip)

我的ppq版本是根据文档安装的0.1.2，组件版本使用的最新版espressif/esp-dl: "^3.1.2"，espidf版本5.4.1

>  I (3020) tof task: Starting NN inference...
> 
> Guru Meditation Error: Core  / panic'ed (Cache disabled but cached memory region accessed). 
> MMU entry fault error occurred while accessing the address 0x3c900000 (invalid mmu entry)
> 
> 
> Core  0 register dump:
> PC      : 0x4037b146  PS      : 0x00060a34  A0      : 0x82003215  A1      : 0x3fca92a0  
> --- 0x4037b146: esp_cpu_wait_for_intr at /home/kui/esp/v5.4.1/esp-idf/components/esp_hw_support/cpu.c:64
> 
> A2      : 0x00000000  A3      : 0xffffffff  A4      : 0x803770f8  A5      : 0x00060023  
> A6      : 0x3fca8770  A7      : 0x0000cdcd  A8      : 0x8206c166  A9      : 0x3fca9260  
> A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00060020  A13     : 0x3fca8b70  
> A14     : 0x00000000  A15     : 0x3fca946c  SAR     : 0x0000001e  EXCCAUSE: 0x00000007  
> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
> 
> 
> Backtrace: 0x4037b143:0x3fca92a0 0x42003212:0x3fca92c0 0x403805a9:0x3fca92e0 0x4037f67d:0x3fca9300
> --- 0x4037b143: xt_utils_wait_for_intr at /home/kui/esp/v5.4.1/esp-idf/components/xtensa/include/xt_utils.h:82
> ---  (inlined by) esp_cpu_wait_for_intr at /home/kui/esp/v5.4.1/esp-idf/components/esp_hw_support/cpu.c:55
> --- 0x42003212: esp_vApplicationIdleHook at /home/kui/esp/v5.4.1/esp-idf/components/esp_system/freertos_hooks.c:58
> --- 0x403805a9: prvIdleTask at /home/kui/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4353 (discriminator 1)
> --- 0x4037f67d: vPortTaskWrapper at /home/kui/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
> 
> 
> 
> Core  1 register dump:
> PC      : 0x42066c5f  PS      : 0x00060634  A0      : 0x820819ce  A1      : 0x3fcb3f30  
> --- 0x42066c5f: tie728_s8_depthwise_conv2d_per_layer_hwc1_bias_loop at /home/kui/esp/proj/maincontrol-esp32/build/../managed_components/espressif__esp-dl/dl/base/isa/tie728/dl_tie728_s8_depthwise_conv2d.S:999
> 
> A2      : 0x3c8ff9f0  A3      : 0x3c8ffed0  A4      : 0x3fcb5ebc  A5      : 0x3c10a1f0  
> A6      : 0x00000020  A7      : 0xffffffe0  A8      : 0xfffffed0  A9      : 0x00000001  
> A10     : 0x00000003  A11     : 0x00000001  A12     : 0x00000005  A13     : 0x00000001  
> A14     : 0x3c10a000  A15     : 0x00000010  SAR     : 0x00000010  EXCCAUSE: 0x00000007  
> EXCVADDR: 0x00000000  LBEG    : 0x42066c40  LEND    : 0x42066c4e  LCOUNT  : 0x00000000  
> --- 0x42066c40: tie728_s8_depthwise_conv2d_per_layer_hwc1_bias_loop at /home/kui/esp/proj/maincontrol-esp32/build/../managed_components/espressif__esp-dl/dl/base/isa/tie728/dl_tie728_s8_depthwise_conv2d.S:999
> --- 0x42066c4e: tie728_s8_depthwise_conv2d_per_layer_hwc1_bias_loop at /home/kui/esp/proj/maincontrol-esp32/build/../managed_components/espressif__esp-dl/dl/base/isa/tie728/dl_tie728_s8_depthwise_conv2d.S:999
> 
> 
> 
> Backtrace: 0x42066c5c:0x3fcb3f30 0x420819cb:0x3fcb3fb0 0x42038ed7:0x3fcb3fd0 0x42039e04:0x3fcb4140 0x420144b7:0x3fcb41a0 0x420220e2:0x3fcb41f0 0x4200cbf8:0x3fcb42b0 0x4037f67d:0x3fcb4320
> --- 0x42066c5c: tie728_s8_depthwise_conv2d_per_layer_hwc1_bias_loop at /home/kui/esp/proj/maincontrol-esp32/build/../managed_components/espressif__esp-dl/dl/base/isa/tie728/dl_tie728_s8_depthwise_conv2d.S:999
> --- 0x420819cb: void std::__invoke_impl<void, void (*&)(signed char*, signed char*, void*), signed char*, signed char*, void*>(std::__invoke_other, void (*&)(signed char*, signed char*, void*), signed char*&&, signed char*&&, void*&&) at /home/kui/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/invoke.h:61
> ---  (inlined by) std::enable_if<is_invocable_r_v<void, void (*&)(signed char*, signed char*, void*), signed char*, signed char*, void*>, void>::type std::__invoke_r<void, void (*&)(signed char*, signed char*, void*), signed char*, signed char*, void*>(void (*&)(signed char*, signed char*, void*), signed char*&&, signed char*&&, void*&&) at /home/kui/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/invoke.h:111
> ---  (inlined by) std::_Function_handler<void (signed char*, signed char*, void*), void (*)(signed char*, signed char*, void*)>::_M_invoke(std::_Any_data const&, signed char*&&, signed char*&&, void*&&) at /home/kui/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/std_function.h:290
> --- 0x42038ed7: std::function<void (signed char*, signed char*, void*)>::operator()(signed char*, signed char*, void*) const at /home/kui/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/std_function.h:591
> ---  (inlined by) void dl::base::dwconv_operation_shell<signed char, long>(dl::base::ArgsType<signed char>&, std::function<void (signed char*, signed char*, void*)>, std::function<void (signed char*, signed char*, void*)>, void (*)(long*, signed char*, dl::base::ArgsType<signed char> const&), void (*)(long*, signed char*, dl::base::ArgsType<signed char> const&), void (*)(signed char*, long*, dl::base::ArgsType<signed char> const&)) at /home/kui/esp/proj/maincontrol-esp32/build/../managed_components/espressif__esp-dl/dl/base/dl_base.hpp:1307
> --- 0x42039e04: void dl::base::depthwise_conv2d<signed char, long, long>(void*) at /home/kui/esp/proj/maincontrol-esp32/build/../managed_components/espressif__esp-dl/dl/base/dl_base_depthwise_conv2d.cpp:814 (discriminator 2)
> --- 0x420144b7: dl::module::Conv2D::forward_args(void*) at /home/kui/esp/proj/maincontrol-esp32/build/../managed_components/espressif__esp-dl/dl/module/include/dl_module_conv.hpp:120
> ---  (inlined by) void dl::module::Conv2D::forward_template<signed char>(dl::ModelContext*, dl::runtime_mode_t) at /home/kui/esp/proj/maincontrol-esp32/build/../managed_components/espressif__esp-dl/dl/module/include/dl_module_conv.hpp:165
> ---  (inlined by) dl::module::Conv2D::forward(dl::ModelContext*, dl::runtime_mode_t) at /home/kui/esp/proj/maincontrol-esp32/build/../managed_components/espressif__esp-dl/dl/module/include/dl_module_conv.hpp:132
> --- 0x420220e2: dl::Model::test() at /home/kui/esp/proj/maincontrol-esp32/build/../managed_components/espressif__esp-dl/dl/model/src/dl_model_base.cpp:428
> --- 0x4200cbf8: tof_task(void*) at /home/kui/esp/proj/maincontrol-esp32/build/../main/task/tof_task.cpp:150
> --- 0x4037f67d: vPortTaskWrapper at /home/kui/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
> 
> 
> 
> 
> 
> ELF file SHA256: 02097e816
> 
> Rebooting...

kuiandhui, 2025-05-08:
量化使用的代码如下：
```
import torch
import numpy as np
from torch.utils.data import Dataset, DataLoader
from ppq.api import espdl_quantize_onnx

BATCHSIZE = 256
INPUT_SHAPE = [1, 1, 96, 96]  # 输入shape
TARGET = "esp32s3"
DEVICE = "cpu"
ONNX_PATH = "./model/onnx/ptq_test_m0.25.onnx"
ESPDL_MODEL_PATH = "./model/espdl/ptq_test.espdl"
NUM_OF_BITS = 16  # 量化位数

class tofDataset(Dataset):
    def __init__(self,
                 image_file,
                 transform=None):

        self.images  = np.load(image_file)
        self.transform = transform
    def __getitem__(self, index):
        image = self.images[index]
        image = torch.from_numpy(image)
        
        if self.transform is not None:
            image = self.transform(image)

        return image

    def __len__(self):
        return self.images.shape[0]
    
test_dataset  = tofDataset('./dataset/img_test.npy')
dataloader = DataLoader(test_dataset, batch_size=BATCHSIZE, shuffle=False)

def collate_fn(batch):
    img = batch
    return img.to(DEVICE)

quant_ppq_graph = espdl_quantize_onnx(
    onnx_import_file=ONNX_PATH,
    espdl_export_file=ESPDL_MODEL_PATH,
    calib_dataloader=dataloader,
    calib_steps=32,  # 校准的步数
    input_shape=INPUT_SHAPE,  # 输入形状，批次为 1
    inputs=None, # [test_dataset[0][0].unsqueeze(0).to(DEVICE)],
    target=TARGET,  # 目标量化类型
    num_of_bits=NUM_OF_BITS,  # 量化位数
    collate_fn=collate_fn,
    dispatching_override=None,
    device=DEVICE,
    # setting=quant_setting,
    error_report=True,
    skip_export=False,
    export_test_values=True,
    verbose=1,  # 输出详细日志信息
)
```
同时附上我的onnx文件

[ptq_test_m0.25.zip](https://github.com/user-attachments/files/20100550/ptq_test_m0.25.zip)

BlueSkyB, 2025-05-09:
感谢您提供的onnx文件，我们能复现该问题，将会开始着手分析该问题。

BlueSkyB, 2025-05-16:
该问题我们已经修复，commit id: 3eb0016
 可以更新下esp-dl再尝试一下。

kuiandhui, 2025-05-16:
> 该问题我们已经修复，commit id: [3eb0016](https://github.com/espressif/esp-dl/commit/3eb00161387be7cbbcaa7169c116e6498c2ba7bb) 可以更新下esp-dl再尝试一下。

我拉取了新版的代码，然后在idf_component.yml中修改了依赖的路径，使用我之前上传的.espdl模型在运行model->test()时确实能正常运行。
但是我重新运行量化脚本后，生成的新.espld文件后部署到esp32上运行时报错
```
E (1109) compare_elements: Inconsistent values, ground truth: 16360.0000000000, infer: 16352.0000000000, epsilon:1.0000100136
E (1113) compare_elements: The position is: [0, 0, 1, 1]
E (1118) dl::Model: Test output 420 does not match
```
误差似乎在我可以接受的范围内，但还是报错还是让我觉得有些困惑。附上报错对应的.espdl文件

[ptq_test.zip](https://github.com/user-attachments/files/20241886/ptq_test.zip)

BlueSkyB, 2025-05-16:
因为你将整个模型按16bit量化了，16bit量化精度较高，在PC端进行量化时，float运算可能损失精度。
你如果要将整个模型按16bit量化，又需要精度一致，可以将espdl_quantize_onnx接口的入参hi_precision，设置为True。
提醒一下，16bit部署模型，会比8bit慢不少

closed 2025-05-27 by BlueSkyB