espressif/esp-dl #245: ImportError: cannot import name 'YOLO11n' from 'yolo11n_eval'

ceoifung, 2025-06-09:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

我参考了教程https://docs.espressif.com/projects/esp-dl/zh_CN/latest/tutorials/how_to_quantize_model.html 搭建好了esp-ppq的相关环境。然后根据https://github.com/espressif/esp-dl/blob/578ca7ada61dadc2ff769c46a3e7b689cc37e3ab/examples/tutorial/how_to_quantize_model/quantize_yolo11n/trainer.py 想训练yolov11n的模型，然后量化模型为espdl。

### Actual behavior (suspected bug)

实际上执行程序无法执行，报如下错误
```shell
root@3dd5e519213c:/workspace/espdl-yolo-train# python trainer.py

    ___________ ____        ____  ____  ____
   / ____/ ___// __ \      / __ \/ __ \/ __ \
  / __/  \__ \/ /_/ /_____/ /_/ / /_/ / / / /
 / /___ ___/ / ____/_____/ ____/ ____/ /_/ /
/_____//____/_/         /_/   /_/    \___\_\

WARNING ⚠️ user config directory '/root/.config/Ultralytics' is not writeable, defaulting to '/tmp' or CWD.Alternatively you can define a YOLO_CONFIG_DIR environment variable for this path.
Traceback (most recent call last):
  File "/workspace/espdl-yolo-train/trainer.py", line 20, in <module>
    from yolo11n_eval import QuantDetectionValidator, YOLO11n
ImportError: cannot import name 'YOLO11n' from 'yolo11n_eval' 

```

yolo11n_eval.py中不存在YOLO11n

### Error logs or terminal output

```shell
root@3dd5e519213c:/workspace/espdl-yolo-train# python trainer.py

    ___________ ____        ____  ____  ____
   / ____/ ___// __ \      / __ \/ __ \/ __ \
  / __/  \__ \/ /_/ /_____/ /_/ / /_/ / / / /
 / /___ ___/ / ____/_____/ ____/ ____/ /_/ /
/_____//____/_/         /_/   /_/    \___\_\

WARNING ⚠️ user config directory '/root/.config/Ultralytics' is not writeable, defaulting to '/tmp' or CWD.Alternatively you can define a YOLO_CONFIG_DIR environment variable for this path.
Traceback (most recent call last):
  File "/workspace/espdl-yolo-train/trainer.py", line 20, in <module>
    from yolo11n_eval import QuantDetectionValidator, YOLO11n
ImportError: cannot import name 'YOLO11n' from 'yolo11n_eval' (/workspace/espdl-yolo-train/yolo11n_eval.py)
```

### Steps to reproduce the behavior

1. 下载quantize_yolo11n的代码
2. docker部署esp-ppq环境
3. 运行quantize_yolo11n的trainer.py
4. 报错，没有YOLO11n

### Project release version

master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 22.04.4 LTS

### Shell

Bash

Ginosko-mia, 2025-06-09:
这个是个bug, trainer匹配之前的commit中的yolo11n_eval.py.

ceoifung, 2025-06-09:
> 这个是个bug, trainer匹配之前的commit中的yolo11n_eval.py.

请问下，是哪一个commit版本，用官方的YOLO库能够替代不

Ginosko-mia, 2025-06-09:
可以，不过trainer eval整体有点bug, 我今天下午尽快fix提交后可以使用，着急的话先关掉eval， 改成：
    for epoch in range(EPOCH):
        trainer.epoch(training_dataloader)
        if not os.path.exists(str(epoch)):
            os.mkdir(str(epoch))
        qat_graph = trainer.save(
            os.path.join(str(epoch), "coco_detect_yolo11n_s8_v3.espdl"),
            os.path.join(str(epoch), "coco_detect_yolo11n_s8_v3.native"),
        )
    return qat_graph

ceoifung, 2025-06-10:
> 可以，不过trainer eval整体有点bug, 我今天下午尽快fix提交后可以使用，着急的话先关掉eval， 改成：
>
> ```
> for epoch in range(EPOCH):
>         trainer.epoch(training_dataloader)
>         if not os.path.exists(str(epoch)): [...]

你好，请问这个解决了吗，我按照你这个写法运行yolo11n_qat.py依然报错

```shell
    ___________ ____        ____  ____  ____
   / ____/ ___// __ \      / __ \/ __ \/ __ \
  / __/  \__ \/ /_/ /_____/ /_/ / /_/ / / / /
 / /___ ___/ / ____/_____/ ____/ ____/ /_/ /
/_____//____/_/         /_/   /_/    \___\_\

WARNING ⚠️ user config directory '/root/.config/Ultralytics' is not writeable, defaulting to '/tmp' or CWD.Alternatively you can define a YOLO_CONFIG_DIR environment variable for this path.
[Warning] Unexpected input value of operation /model.11/Resize, recieving "None" at its input 1
[Warning] Unexpected input value of operation /model.14/Resize, recieving "None" at its input 1
Traceback (most recent call last):
  File "/workspace/espdl-yolo-train/yolo11n_qat.py", line 134, in <module>
    qat_graph = qat()
  File "/workspace/espdl-yolo-train/yolo11n_qat.py", line 77, in qat
    training_dataloader = DataLoader(train_set, batch_size=CFG_BATCHSIZE, shuffle=True)
  File "/opt/conda/lib/python3.10/site-packages/torch/utils/data/dataloader.py", line 350, in __init__
    sampler = RandomSampler(dataset, generator=generator)  # type: ignore[arg-type]
  File "/opt/conda/lib/python3.10/site-packages/torch/utils/data/sampler.py", line 143, in __init__
    raise ValueError(f"num_samples should be a positive integer value, but got num_samples={self.num_samples}")
ValueError: num_samples should be a positive integer value, but got num_samples=0
```

请问一下，这是怎么回事呢？

Ginosko-mia, 2025-06-10:
eval的bug解决了，正在mr流程中。你这个应该是训练集加载的问题，你改CFG_TRAIN_DIR了吗？

Ginosko-mia, 2025-06-10:
新的代码已经上传，你可以更新master分支的代码试试看。yolo11n folder下的代码是KD QAT，如果想用数据集标签作为ground-truth进行QAT训练, 可以参考yolo11n-pose下的代码进行修改。

ceoifung, 2025-06-11:
> 新的代码已经上传，你可以更新master分支的代码试试看。yolo11n folder下的代码是KD QAT，如果想用数据集标签作为ground-truth进行QAT训练, 可以参考yolo11n-pose下的代码进行修改。

我这边使用的yolov11训练的出来的pt模型（用我自己的电脑训练的，也尝试用过docker的yolo进行训练），然后使用espdl的export_onnx.py，导出成onnx模型，使用的校准数据集是我自己的采集的数据集，图片大小是640x480，然后运行 python yolo11n_eval.py（里面的代码加载的onnx模型，pt文件还有校准集也改了），espdl模型也量化出来了，但是运行到后面会报错误。没有办法进行推理，这是什么问题导致的呢？是我的数据集有问题吗？还是什么。板子采用的是esp32s3 imgSz=640

```shell
root@c3ea966cd7f2:/workspace# python yolo11n_eval.py

    ___________ ____        ____  ____  ____
   / ____/ ___// __ \      / __ \/ __ \/ __ \
  / __/  \__ \/ /_/ /_____/ /_/ / /_/ / / / /
 / /___ ___/ / ____/_____/ ____/ ____/ /_/ /
/_____//____/_/         /_/   /_/    \___\_\

[Warning] Unexpected input value of operation /model.11/Resize, recieving "None" at its input 1
[Warning] Unexpected input value of operation /model.14/Resize, recieving "None" at its input 1
[06:05:21] PPQ Quantization Fusion Pass Running ...       Finished.
[06:05:21] PPQ Quantize Simplify Pass Running ...         Finished.
[06:05:21] PPQ Parameter Quantization Pass Running ...    Finished.
Calibration Progress(Phase 1): 100%|████████████████████████████████████████████████████| 32/32 [00:37<00:00,  1.18s/it]
Calibration Progress(Phase 2): 100%|████████████████████████████████████████████████████| 32/32 [04:18<00:00,  8.07s/it]
Finished.
[06:10:19] PPQ Quantization Alignment Pass Running ...    Finished.
[06:10:19] PPQ Passive Parameter Quantization Running ... Finished.
--------- Network Snapshot ---------
Num of Op:                    [296]
Num of Quantized Op:          [296]
Num of Variable:              [499]
Num of Quantized Var:         [499]
------- Quantization Snapshot ------
Num of Quant Config:          [915]
ACTIVATED:                    [334]
OVERLAPPED:                   [369]
PASSIVE:                      [194]
FP32:                         [18]
Network Quantization Finished.
Analysing Graphwise Quantization Error(Phrase 1):: 100%|██████████████████████████████████| 1/1 [00:03<00:00,  3.13s/it]
Analysing Graphwise Quantization Error(Phrase 2):: 100%|██████████████████████████████████| 1/1 [00:06<00:00,  6.00s/it]
Layer                                        | NOISE:SIGNAL POWER RATIO
/model.23/cv3.2/cv3.2.1/cv3.2.1.0/conv/Conv: | ████████████████████ | 29.155%
/model.23/cv3.2/cv3.2.1/cv3.2.1.1/conv/Conv: | ██████████████████   | 26.612%
/model.22/m.0/cv2/conv/Conv:                 | ██████████████████   | 25.910%
/model.10/m/m.0/attn/proj/conv/Conv:         | ██████████████████   | 25.649%
/model.23/cv3.2/cv3.2.0/cv3.2.0.1/conv/Conv: | █████████████████    | 25.028%
/model.10/m/m.0/ffn/ffn.1/conv/Conv:         | ████████████████     | 23.891%
/model.10/m/m.0/attn/MatMul_1:               | ███████████████      | 21.934%
/model.22/m.0/m/m.1/cv2/conv/Conv:           | █████████████        | 19.032%
/model.10/m/m.0/attn/MatMul:                 | █████████████        | 18.308%
/model.22/m.0/cv3/conv/Conv:                 | ████████████         | 18.132%
/model.10/m/m.0/attn/pe/conv/Conv:           | ████████████         | 17.906%
/model.23/cv3.2/cv3.2.0/cv3.2.0.0/conv/Conv: | ████████████         | 17.507%
/model.23/cv3.0/cv3.0.1/cv3.0.1.0/conv/Conv: | ████████████         | 17.132%
/model.23/cv3.0/cv3.0.1/cv3.0.1.1/conv/Conv: | ████████████         | 17.025%
/model.10/m/m.0/attn/qkv/conv/Conv:          | ████████████         | 16.942%
/model.10/cv1/conv/Conv:                     | ████████████         | 16.788%
/model.22/cv2/conv/Conv:                     | ███████████          | 16.745%
/model.23/cv2.2/cv2.2.0/conv/Conv:           | ███████████          | 15.694%
/model.8/m.0/m/m.1/cv2/conv/Conv:            | ██████████           | 14.689%
/model.8/m.0/m/m.1/cv1/conv/Conv:            | ██████████           | 14.308%
/model.8/m.0/cv2/conv/Conv:                  | ██████████           | 14.274%
/model.23/cv3.1/cv3.1.1/cv3.1.1.0/conv/Conv: | █████████            | 13.660%
/model.23/cv2.2/cv2.2.1/conv/Conv:           | █████████            | 13.612%
/model.8/m.0/cv3/conv/Conv:                  | █████████            | 13.608%
/model.22/m.0/m/m.1/cv1/conv/Conv:           | █████████            | 12.636%
/model.23/cv3.1/cv3.1.1/cv3.1.1.1/conv/Conv: | █████████            | 12.458%
/model.8/m.0/m/m.0/cv2/conv/Conv:            | ████████             | 12.231%
/model.23/cv3.0/cv3.0.0/cv3.0.0.1/conv/Conv: | ████████             | 12.194%
/model.22/m.0/m/m.0/cv2/conv/Conv:           | ████████             | 11.915%
/model.22/m.0/m/m.0/cv1/conv/Conv:           | ████████             | 11.595%
/model.23/cv3.1/cv3.1.0/cv3.1.0.1/conv/Conv: | ███████              | 10.283%
/model.8/cv2/conv/Conv:                      | ███████              | 9.716%
[... 248 lines cut ...]
Ultralytics 8.3.152 🚀 Python-3.10.14 torch-2.2.2 CPU (AMD Ryzen 9 7940HX with Radeon Graphics)
YOLO11n summary (fused): 100 layers, 2,582,347 parameters, 0 gradients, 6.3 GFLOPs
val: Fast image access ✅ (ping: 2.0±0.1 ms, read: 30.3±2.8 MB/s, size: 93.2 KB)
val: Scanning /workspace/datasets/onlymeter/labels/val... 9 images, 0 backgrounds, 0 corrupt: 100%|██████████| 9/9 [00:0
val: New cache created: /workspace/datasets/onlymeter/labels/val.cache
                 Class     Images  Instances      Box(P          R      mAP50  mAP50-95):   0%|          | 0/1 [00:01<?,
Traceback (most recent call last):
  File "/workspace/yolo11n_eval.py", line 235, in <module>
    results = model.val(
[... cut ...]
  File "/workspace/yolo11n_eval.py", line 212, in ppq_graph_inference
    y = detect_model._inference(x)
  File "/opt/conda/lib/python3.10/site-packages/ultralytics/nn/modules/head.py", line 162, in _inference
    x_cat = torch.cat([xi.view(shape[0], self.no, -1) for xi in x], 2)
  File "/opt/conda/lib/python3.10/site-packages/ultralytics/nn/modules/head.py", line 162, in <listcomp>
    x_cat = torch.cat([xi.view(shape[0], self.no, -1) for xi in x], 2)
RuntimeError: shape '[9, 144, -1]' is invalid for input of size 3744000
```

ceoifung, 2025-06-11:
> 维度不匹配导致的issue, 你训练的pt的检测对象是coco（nc=80）吗？

我训练的只有一个只有一个类别（nc=1），基于yolo11n.pt作为预训练权重训练的

Ginosko-mia, 2025-06-11:
yolo11n_eval.py中这行对应修改为nc=1
```
detect_model = Detect(nc=80, ch=[32, 64, 128])
```